Ysurac/openmptcprouter #1782: problème de Port forwarding

Pokpok-sbo, 2021-04-25:
Bonjour à tous,

Voilà, je suis enfin passé sur Openmptcprouter pour mon entreprise!!!

J'ai aucun probléme en local, tout fonctionne bien (acces à internet, vlan, dhcp, dns...)
Par contre je n'arrive pas à régler le Port Forwarding pour acceder à mes serveurs depuis l'extérieur.

J'ai réglé dans mon routeur le port forwarding vers l'interface correspondante dans l'Openmptcprouteur
[image]
voici le réglage de l'interface:
[image]
[image]
Voici les details du firewall
[image]
[image]

Merci d'avance pour votre temps !

Ysurac, 2021-04-25:
Le but est d'accéder depuis l'IP d'un wan ou depuis l'IP du VPS ?

Pokpok-sbo, 2021-04-26:
J'ai 4 connexions et pour l'instant le but serait d'accéder directement à partir du wan pour diminuer le ping.

Pokpok-sbo, 2021-04-27:
Aujourd'hui j'ai essayé d'insister en essayant sur d'autre connexion, en ajoutant des règles... mais rien ne marche :(
Quelqu'un aurait-il une piste pour me débloquer :)

Ysurac, 2021-04-27:
Tu as essayé de voir si quelque chose entre avec tcpdump ?

Pokpok-sbo, 2021-04-27:
Je ne connais pas cet outil, je vais essayer de voir si je peux trouver d'où vient le problème.
merci pour la piste !!!

Pokpok-sbo, 2021-04-28:
Belle journée à tous !

Après quelques pages de lecture, je pense avoir avancé. En faisant plusieurs tests, j'en ai conclu que le problème se situe au niveau de la réponse. Openmtcp n'envoie pas les paquets vers l'interface du modem.

J'ai lancé tcpdump sur l'interface (eth1) entre le modem et Openmptcp sur le port 5000 celui que j'utilise pour accéder à mon serveur voila ce qui c'est passé :
[image]
On voit bien que le paquet est bien transmit vers Openmptcp mais il n'y a jamais de réponse.

Ensuite j'ai lancé tcpdump sur l'interface (eth0.5) entre openmptcp et mon serveur :
[image]
Là, on voit bien que mon serveur (hozuki) reçoit bien le paquet et il y répond aussi

Maintenant je vois bien où se situe le problème, mais je ne vois pas quel paramètre changer dans openmptcp pour le résoudre étant donné que dans les règles de trafics :
[image]
et dans les port forwarding :
[image]

Déjà un grand merci, pour m'avoir orienté vers tcpdump que je ne connaissais pas. Ça à l'air d'être vraiment un outil super utile.

Quelqu'un saurait-il ce qui bloque la sortie de mes paquets ?

Belle journée à tous.

Ysurac, 2021-04-30:
J'ai le même soucis.
J'ai pensé que c'était un soucis dans les régles pour le proxy mais en le désactivant ça change rien. Pareil en désactivant le VPN. Pareil en virant les régles omr-bypass.

Pokpok-sbo, 2021-05-01:
D’un côté, je suis rassuré si même à toi ça te fait pareil, je commençais à penser que j’étais plus capable de faire du port forwarding.

Penses-tu que le problème est résoluble ?
Avec une version plus ancienne de Openmptcp le problème existé déjà ?

Par contre, je suis arrivé à faire mes redirections en passant par le vpn et mon adresse IP d’ovh. Déjà avec ça j’arrive à atteindre mes serveurs à distance. Mais étant à 10 000 km du serveur ovh (ce n’est pas une manière de parler mdr) mon ping est hyper long... avec l’aller-retour hahaha donc si je pouvais passer directement par mes modems ça me changerait la vie :)

Ysurac, 2021-05-04:
Le soucis semble être lié au double NAT, car pas de soucis en interrogeant directement l'IP du WAN.

Pokpok-sbo, 2021-05-08:
Bonjour, j'ai continué à chercher :)
j'ai eu l'idée de désactivé le VPN du le serveur OVH. Et la redirection de port fonctionne et j'arrive à me connecter à mes serveuses via les interfaces Wan.
Ensuite j'ai utilisé tcpdump et j'ai vu que mes redirections rentre bien par les interfaces Wan et va  jusqu'au server mais le routeur Openmptcp envoie la réponse vers le Tun0 au lieu de l'envoyer sur l'interface qui a servie à la redirection de port.

Ça serait à quel niveau que je pourrais modifier ça et dire au router Openmptcp qu'il doit renvoyer les réponses sur l'interface sur laquelle il l'a reçu?

Merci d'avance

cayenne17, 2021-06-02:
Bonjour,
C'est le même problème qu'ici : https://github.com/Ysurac/openmptcprouter/issues/1307
moi aussi je rencontre le même problème.

pour contourner le problème, j'ai mis une règles OMR-Bypass sur le serveur en plus de la règles NAT/PAT

closed 2021-09-06 by github-actions[bot]