Ysurac/openmptcprouter #1784: Tuto pour création Vlan

guillaumed24, 2021-04-25:
Bonjour,

Est-il possible d'avoir un coup de main pour créer un vlan avec son propre dhcp ?

Merci.

Ysurac, 2021-04-26:
Bonjour,
Pour créer les VLANs, il faut définir une nouvelle interfaces, dans Réseau->Interfaces, et utiliser en interface physique "eth0.200" pour le vlan 200 sur l'interface physique eth0.

guillaumed24, 2021-04-26:
Ok et eth1.200 pour l'interface eth1 ?
J'utilise une ancienne overthebox. Comment je peux identifier les ports eth1.2.3.4..... ?

Ysurac, 2021-04-26:
Oui exactement.
Pour identifier les ports, possible en branchant un câble sur chaque port un à un et en regardant les logs pour voir qui devient up.

guillaumed24, 2021-04-26:
Les logs > états : journaux système ou un autre ?

Ysurac, 2021-04-26:
Oui.
Il y a aussi via SSH sur le routeur, ```ethtool eth2``` avec l'information "Link detected".

guillaumed24, 2021-04-26:
Parce que dans les logs quelque soit le port j'ai ca : 

Mon Apr 26 10:56:39 2021 daemon.info dnsmasq-dhcp[21729]: DHCPDISCOVER(eth0) 192.168.40.140 84:78:ac:a5:****
Mon Apr 26 10:56:39 2021 daemon.info dnsmasq-dhcp[21729]: DHCPOFFER(eth0) 192.168.40.140 84:78:ac:a5:****
Mon Apr 26 10:56:39 2021 daemon.info dnsmasq-dhcp[21729]: DHCPREQUEST(eth0) 192.168.40.140 84:78:ac:a5:****
Mon Apr 26 10:56:39 2021 daemon.info dnsmasq-dhcp[21729]: DHCPACK(eth0) 192.168.40.140 84:78:ac:a5:****

Ysurac, 2021-04-26:
Si l'interface eth0 a le DHCP d'activer, normal. Le serveur DHCP est activé par défaut sur l'interface LAN.

closed 2021-07-31 by github-actions[bot]