Ysurac/openmptcprouter #1785: Can't Contact Server Admin Script

ccmks, 2021-04-26:
## Expected Behavior
Everything is working with all green check instead of yellow exclamation mark on OMR Status page

## Current Behavior
[image]

## Possible Solution
No idea

## Steps to Reproduce the Problem
It was working fine for several days until we get this

## Context (Environment)
N/A

## Specifications

  - OpenMPTCProuter version: penmptcprouter v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-MPTCP
  - OpenMPTCProuter platform: x86_64 Hyper-V

Ysurac, 2021-04-26:
What is this 27.112.xxx IP from ?
Check in State->System log why it can't contact VPS API.

ccmks, 2021-04-26:
What is this 27.112.xxx IP from ? - Indonesian VPS

It is working now by itself so I am not sure what happened. But I do know that it is happening intermittently. At least it has been doing it 3 times a week, so not sure if go to system log will be relevant?

Ysurac, 2021-04-26:
Yes you should check in system log if a VPS problem is detected when there is the problem.
Possible that the latency between connections and VPS is sometimes too high or some packets lost.

ccmks, 2021-04-26:
Here is the system log:

Mon Apr 26 21:08:34 2021 daemon.notice openvpn(wan3)[7347]: OpenVPN 2.5.0 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Apr 26 21:08:34 2021 daemon.notice openvpn(wan3)[7347]: library versions: OpenSSL 1.1.1i  8 Dec 2020
Mon Apr 26 21:08:34 2021 daemon.warn openvpn(wan3)[7347]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mon Apr 26 21:08:34 2021 daemon.warn openvpn(wan3)[7347]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mon Apr 26 21:08:34 2021 daemon.notice openvpn(wan3)[7347]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.78.109:65301
Mon Apr 26 21:08:34 2021 daemon.err openvpn(wan3)[7347]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Mon Apr 26 21:08:34 2021 daemon.notice openvpn(wan3)[7347]: Exiting due to fatal error
Mon Apr 26 21:08:39 2021 daemon.warn openvpn(wan3)[7967]: DEPRECATED OPTION: ncp-disable. Disabling cipher negotiation is a deprecated debug feature that will be removed in OpenVPN 2.6
Mon Apr 26 21:08:39 2021 daemon.warn openvpn(wan3)[7967]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM). Future OpenVPN version will ignore --cipher for cipher negotiations. Add 'AES-256-CBC' to --data-ciphers or change --cipher 'AES-256-CBC' to --data-ciphers-fallback 'AES-256-CBC' to silence this warning.
Mon Apr 26 21:08:39 2021 daemon.notice openvpn(wan3)[7967]: OpenVPN 2.5.0 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Apr 26 21:08:39 2021 daemon.notice openvpn(wan3)[7967]: library versions: OpenSSL 1.1.1i  8 Dec 2020
Mon Apr 26 21:08:39 2021 daemon.warn openvpn(wan3)[7967]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mon Apr 26 21:08:39 2021 daemon.warn openvpn(wan3)[7967]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mon Apr 26 21:08:39 2021 daemon.notice openvpn(wan3)[7967]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.78.109:65301
Mon Apr 26 21:08:39 2021 daemon.err openvpn(wan3)[7967]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Mon Apr 26 21:08:39 2021 daemon.notice openvpn(wan3)[7967]: Exiting due to fatal error
Mon Apr 26 21:08:44 2021 daemon.warn openvpn(wan3)[9275]: DEPRECATED OPTION: ncp-disable. Disabling cipher negotiation is a deprecated debug feature that will be removed in OpenVPN 2.6
Mon Apr 26 21:08:44 2021 daemon.warn openvpn(wan3)[9275]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM). Future OpenVPN version will ignore --cipher for cipher negotiations. Add 'AES-256-CBC' to --data-ciphers or change --cipher 'AES-256-CBC' to --data-ciphers-fallback 'AES-256-CBC' to silence this warning.
Mon Apr 26 21:08:44 2021 daemon.notice openvpn(wan3)[9275]: OpenVPN 2.5.0 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Apr 26 21:08:44 2021 daemon.notice openvpn(wan3)[9275]: library versions: OpenSSL 1.1.1i  8 Dec 2020
Mon Apr 26 21:08:44 2021 daemon.warn openvpn(wan3)[9275]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mon Apr 26 21:08:44 2021 daemon.warn openvpn(wan3)[9275]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mon Apr 26 21:08:44 2021 daemon.notice openvpn(wan3)[9275]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.78.109:65301
Mon Apr 26 21:08:44 2021 daemon.err openvpn(wan3)[9275]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Mon Apr 26 21:08:44 2021 daemon.notice openvpn(wan3)[9275]: Exiting due to fatal error
Mon Apr 26 21:08:45 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon Apr 26 21:08:49 2021 daemon.warn openvpn(wan3)[10927]: DEPRECATED OPTION: ncp-disable. Disabling cipher negotiation is a deprecated debug feature that will be removed in OpenVPN 2.6
Mon Apr 26 21:08:49 2021 daemon.warn openvpn(wan3)[10927]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM). Future OpenVPN version will ignore --cipher for cipher negotiations. Add 'AES-256-CBC' to --data-ciphers or change --cipher 'AES-256-CBC' to --data-ciphers-fallback 'AES-256-CBC' to silence this warning.
Mon Apr 26 21:08:49 2021 daemon.notice openvpn(wan3)[10927]: OpenVPN 2.5.0 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Apr 26 21:08:49 2021 daemon.notice openvpn(wan3)[10927]: library versions: OpenSSL 1.1.1i  8 Dec 2020
Mon Apr 26 21:08:49 2021 daemon.warn openvpn(wan3)[10927]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mon Apr 26 21:08:49 2021 daemon.warn openvpn(wan3)[10927]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mon Apr 26 21:08:49 2021 daemon.notice openvpn(wan3)[10927]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.78.109:65301
[... cut ...]
Mon Apr 26 21:10:37 2021 daemon.warn openvpn(wan3)[4789]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mon Apr 26 21:10:37 2021 daemon.notice openvpn(wan3)[4789]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.78.109:65301
Mon Apr 26 21:10:37 2021 daemon.err openvpn(wan3)[4789]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Mon Apr 26 21:10:37 2021 daemon.notice openvpn(wan3)[4789]: Exiting due to fatal error

ccmks, 2021-04-29:
Got same error again. But in this case the WAN1 is down due to outage. Although the bonding is not working, at least the user can still getting internet connection from single WAN (The user has 2 WAN only and WAN1 is currently down). In this case, if the WAN1 is down, the user can't access internet

Thu Apr 29 13:08:13 2021 daemon.notice openvpn(wan3)[27619]: Exiting due to fatal error
Thu Apr 29 13:08:18 2021 daemon.warn openvpn(wan3)[27745]: DEPRECATED OPTION: ncp-disable. Disabling cipher negotiation is a deprecated debug feature that will be removed in OpenVPN 2.6
Thu Apr 29 13:08:18 2021 daemon.warn openvpn(wan3)[27745]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM). Future OpenVPN version will ignore --cipher for cipher negotiations. Add 'AES-256-CBC' to --data-ciphers or change --cipher 'AES-256-CBC' to --data-ciphers-fallback 'AES-256-CBC' to silence this warning.
Thu Apr 29 13:08:18 2021 daemon.notice openvpn(wan3)[27745]: OpenVPN 2.5.0 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Apr 29 13:08:18 2021 daemon.notice openvpn(wan3)[27745]: library versions: OpenSSL 1.1.1i  8 Dec 2020
Thu Apr 29 13:08:18 2021 daemon.warn openvpn(wan3)[27745]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Apr 29 13:08:18 2021 daemon.warn openvpn(wan3)[27745]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Apr 29 13:08:18 2021 daemon.notice openvpn(wan3)[27745]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.X.X:65301
Thu Apr 29 13:08:18 2021 daemon.err openvpn(wan3)[27745]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Thu Apr 29 13:08:18 2021 daemon.notice openvpn(wan3)[27745]: Exiting due to fatal error
Thu Apr 29 13:08:23 2021 daemon.warn openvpn(wan3)[27854]: DEPRECATED OPTION: ncp-disable. Disabling cipher negotiation is a deprecated debug feature that will be removed in OpenVPN 2.6
Thu Apr 29 13:08:23 2021 daemon.warn openvpn(wan3)[27854]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM). Future OpenVPN version will ignore --cipher for cipher negotiations. Add 'AES-256-CBC' to --data-ciphers or change --cipher 'AES-256-CBC' to --data-ciphers-fallback 'AES-256-CBC' to silence this warning.
Thu Apr 29 13:08:23 2021 daemon.notice openvpn(wan3)[27854]: OpenVPN 2.5.0 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Apr 29 13:08:23 2021 daemon.notice openvpn(wan3)[27854]: library versions: OpenSSL 1.1.1i  8 Dec 2020
Thu Apr 29 13:08:23 2021 daemon.warn openvpn(wan3)[27854]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Apr 29 13:08:23 2021 daemon.warn openvpn(wan3)[27854]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Apr 29 13:08:23 2021 daemon.notice openvpn(wan3)[27854]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.X.X:65301
Thu Apr 29 13:08:23 2021 daemon.err openvpn(wan3)[27854]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Thu Apr 29 13:08:23 2021 daemon.notice openvpn(wan3)[27854]: Exiting due to fatal error
Thu Apr 29 13:08:26 2021 user.notice post-tracking-post-tracking: Check API configuration...
Thu Apr 29 13:08:28 2021 daemon.warn openvpn(wan3)[28051]: DEPRECATED OPTION: ncp-disable. Disabling cipher negotiation is a deprecated debug feature that will be removed in OpenVPN 2.6
Thu Apr 29 13:08:28 2021 daemon.warn openvpn(wan3)[28051]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM). Future OpenVPN version will ignore --cipher for cipher negotiations. Add 'AES-256-CBC' to --data-ciphers or change --cipher 'AES-256-CBC' to --data-ciphers-fallback 'AES-256-CBC' to silence this warning.
Thu Apr 29 13:08:28 2021 daemon.notice openvpn(wan3)[28051]: OpenVPN 2.5.0 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Apr 29 13:08:28 2021 daemon.notice openvpn(wan3)[28051]: library versions: OpenSSL 1.1.1i  8 Dec 2020
Thu Apr 29 13:08:28 2021 daemon.warn openvpn(wan3)[28051]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Apr 29 13:08:28 2021 daemon.warn openvpn(wan3)[28051]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Apr 29 13:08:28 2021 daemon.notice openvpn(wan3)[28051]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.X.X:65301
Thu Apr 29 13:08:28 2021 daemon.err openvpn(wan3)[28051]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Thu Apr 29 13:08:28 2021 daemon.notice openvpn(wan3)[28051]: Exiting due to fatal error
Thu Apr 29 13:08:33 2021 daemon.warn openvpn(wan3)[28332]: DEPRECATED OPTION: ncp-disable. Disabling cipher negotiation is a deprecated debug feature that will be removed in OpenVPN 2.6
[... cut ...]
Thu Apr 29 13:11:23 2021 daemon.warn openvpn(wan3)[6121]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Apr 29 13:11:23 2021 daemon.notice openvpn(wan3)[6121]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.X.X:65301
Thu Apr 29 13:11:23 2021 daemon.err openvpn(wan3)[6121]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Thu Apr 29 13:11:23 2021 daemon.notice openvpn(wan3)[6121]: Exiting due to fatal error

ccmks, 2021-04-29:
Here is the settings of my OMR

Dashboard
Status
Overview
Firewall
Routes
System Log
Kernel Log
Processes
Realtime Graphs
vnStat Traffic Monitor
System
OpenMPTCProuter
System
Administration
ACL Settings
Software
Startup
Scheduled Tasks
Mount Points
Backup / Flash Firmware
Reboot
Shutdown
Services
OMR-Tracker
iPerf
Proxy DNS Over HTTPS
Shadowsocks-libev
OMR-Bypass
Packet Capture
E-Mail
Wake on LAN
Terminal
uHTTPd
UPnP
Recursive DNS
Network
MPTCP
Interfaces
DHCP and DNS
Hostnames
Static Routes
SQM QoS
Firewall
Diagnostics
OMR-DSCP
Quota
SNMPd
VPN
MLVPN
Glorytun TCP
Glorytun UDP
DSVPN
OpenVPN
Logout
Settings Wizard
Status
Advanced Settings
Backup on server
Show all settings
All router settings
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
[... 1673 lines cut ...]
v2ray.omrtest.s_socks_auth='noauth'
v2ray.omrtest.s_socks_udp='1'
v2ray.omrtest.s_socks_ip='127.0.0.1'
v2ray.omrtest.s_socks_userlevel='0'
v2ray.main_reverse=reverse
v2ray.main_reverse.enabled='1'
v2ray.main_reverse.bridges='omrbridge|omr.lan'
v2ray.omrexit=outbound
v2ray.omrexit.protocol='freedom'
v2ray.omrexit.tag='out'
v2ray.omrrouting=routing_rule
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2' 'eth1' 'eth2' 'eth3'
Powered by openmptcprouter v0.57.3 r0+15225-bfc433efd4

Ysurac, 2021-04-29:
You seems to be using MPTCP over VPN option. Disable it.

ccmks, 2021-04-29:
Already disabled, still no luck

ccmks, 2021-04-29:
Older version used to work OK in this scenario when it was on 0.55 or 0.56

Ysurac, 2021-04-29:
MPTCP over VPN is not disabled in log and config.
You tried to reboot VPS and router ?

ccmks, 2021-04-29:
I rebooted both OMR and VPS, still like this, no luck

ccmks, 2021-04-29:
Here is the latest log

Thu Apr 29 14:33:09 2021 daemon.notice openvpn(wan3)[20846]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.X.X:65301
Thu Apr 29 14:33:09 2021 daemon.err openvpn(wan3)[20846]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Thu Apr 29 14:33:09 2021 daemon.notice openvpn(wan3)[20846]: Exiting due to fatal error
Thu Apr 29 14:33:09 2021 daemon.notice procd: /etc/rc.d/S95done: Failed to initialize libusb: LIBUSB_ERROR_OTHER
Thu Apr 29 14:33:10 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Apr 29 14:33:10 2021 user.notice firewall: Reloading firewall due to ifup of wan3 (eth3)
Thu Apr 29 14:33:11 2021 user.notice omr-bypass: Starting OMR-ByPass...
Thu Apr 29 14:33:12 2021 daemon.notice procd: /etc/rc.d/S95done: Failed to initialize libusb: LIBUSB_ERROR_OTHER
Thu Apr 29 14:33:13 2021 user.err vnstat_backup: cannot restore, backup file does not exist (yet)
Thu Apr 29 14:33:13 2021 user.notice adb-enablemodem: INFO: waiting for device
Thu Apr 29 14:33:14 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1)
Thu Apr 29 14:33:14 2021 user.notice Shadowsocks: Rules DOWN
Thu Apr 29 14:33:14 2021 daemon.warn openvpn(wan3)[23362]: DEPRECATED OPTION: ncp-disable. Disabling cipher negotiation is a deprecated debug feature that will be removed in OpenVPN 2.6
Thu Apr 29 14:33:14 2021 daemon.warn openvpn(wan3)[23362]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM). Future OpenVPN version will ignore --cipher for cipher negotiations. Add 'AES-256-CBC' to --data-ciphers or change --cipher 'AES-256-CBC' to --data-ciphers-fallback 'AES-256-CBC' to silence this warning.
Thu Apr 29 14:33:14 2021 daemon.notice openvpn(wan3)[23362]: OpenVPN 2.5.0 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Thu Apr 29 14:33:14 2021 daemon.notice openvpn(wan3)[23362]: library versions: OpenSSL 1.1.1i  8 Dec 2020
Thu Apr 29 14:33:14 2021 daemon.warn openvpn(wan3)[23362]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Thu Apr 29 14:33:14 2021 daemon.warn openvpn(wan3)[23362]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Apr 29 14:33:14 2021 daemon.notice openvpn(wan3)[23362]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.X.X:65301
Thu Apr 29 14:33:14 2021 daemon.err openvpn(wan3)[23362]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Thu Apr 29 14:33:14 2021 daemon.notice openvpn(wan3)[23362]: Exiting due to fatal error
Thu Apr 29 14:10:01 2021 user.notice omr-bypass: Restart dnsmasq...
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[6685]: exiting on receipt of SIGTERM
Thu Apr 29 14:10:02 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Apr 29 14:10:02 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Apr 29 14:10:02 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: started, version 2.82 cachesize 150
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: DNS service limited to local subnets
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq-dhcp[23666]: DHCP, IP range 192.168.120.100 -- 192.168.120.249, lease time 12h
Thu Apr 29 14:10:02 2021 user.notice omr-bypass: OMR-ByPass is running
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using only locally-known addresses for domain test
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using only locally-known addresses for domain onion
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using only locally-known addresses for domain localhost
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using only locally-known addresses for domain local
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using only locally-known addresses for domain invalid
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using only locally-known addresses for domain bind
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using only locally-known addresses for domain use-application-dns.net
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using only locally-known addresses for domain lan
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using nameserver 127.0.0.1#5353
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: using 1 more local addresses
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: read /etc/hosts - 4 addresses
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq[23666]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu Apr 29 14:10:02 2021 daemon.info dnsmasq-dhcp[23666]: read /etc/ethers - 0 addresses
Thu Apr 29 14:10:02 2021 user.notice omr-bypass: Starting OMR-ByPass...
Thu Apr 29 14:10:02 2021 user.notice Shadowsocks: Rules UP
Thu Apr 29 14:10:03 2021 user.notice Shadowsocks: Reload omr-bypass rules
Thu Apr 29 14:10:03 2021 user.notice omr-bypass: Starting OMR-ByPass...
[... cut ...]
Thu Apr 29 14:11:55 2021 daemon.warn openvpn(wan3)[13447]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Thu Apr 29 14:11:55 2021 daemon.notice openvpn(wan3)[13447]: TCP/UDP: Preserving recently used remote address: [AF_INET]27.112.X.X:65301
Thu Apr 29 14:11:55 2021 daemon.err openvpn(wan3)[13447]: TCP/UDP: Socket bind failed on local address [AF_INET]192.168.18.120:65301: Address not available (errno=99)
Thu Apr 29 14:11:55 2021 daemon.notice openvpn(wan3)[13447]: Exiting due to fatal error

ccmks, 2021-04-29:
Should I re-install the VPS?

Ysurac, 2021-04-29:
No.
Your still have MPTCP over VPN enabled for WAN3.

ccmks, 2021-04-29:
Hmmm, that's is strange, I disabled it earlier. MPTCP over VPN still being checked regardless I have unchecked it on Wizard

ccmks, 2021-04-29:
I can't disable MPTCP over VPN, it keeps turn on

ccmks, 2021-04-29:
I ended up factory reset

ccmks, 2021-04-29:
I think there is a bug in MPTCP over VPN. Once it is checked, it can't be unchecked no matter what you do. Also, how come MPTCP over VPN affect this issue?

Ysurac, 2021-04-29:
Maybe. There is not this MPTCP over VPN issue in 0.58rc1.
This affect this issue because MPTCP over VPN was not working, OpenVPN crashed.

ccmks, 2021-04-29:
Also, how do I run iperf from either of WAN to my VPS, just to make sure that I am getting the expected throughput by testing the iperf

closed 2021-08-03 by github-actions[bot]