Ysurac/openmptcprouter #1792: Question about Perf from 2 different arch

vinceducat, 2021-04-27:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

**Describe why you are asking for this support?**

**Additional context**

Ysurac, 2021-04-28:
Yes it's strange. Maybe the higher frequency by core of arch2 help.

vinceducat, 2021-04-28:
not a problem of ethernet card for you ?

vinceducat, 2021-04-28:
other question on the arch1 i have 2 eth port can i put on the second (eth1] my adsl gateway in place of "macvlan" on the eth0

Ysurac, 2021-04-29:
Yes you can use eth1 for any connection directly.

vinceducat, 2021-04-29:
Just need to change in the wizard from MacVlan eth0 to normal Eth1... or need to creation interface first ?

Ysurac, 2021-04-29:
Just need to change yes.

vinceducat, 2021-04-29:
Thanks wiil try ;-)

vinceducat, 2021-04-29:
seems to work but i have [   17.468749] r8169 0000:02:00.0 eth1: No native access to PCI extended config space, falling back to CSI

closed 2021-08-03 by github-actions[bot]