Ysurac/openmptcprouter #1794: add serial to grub.cfg to allow boot from devices which only have a serial console

nikwest, 2021-04-29:
In order to boot from a pc engines apu2 device, which only have a serial console I had to add serial to the grub.cfg in order to see the boot process. See attached diff for grub.cfg[grub.cfg.txt](https://github.com/Ysurac/openmptcprouter/files/6401605/grub.cfg.txt)
Not sure where this should be applied in the repository.

Ysurac, 2021-05-04:
In fact there is an option to set a serial port device at image compilation, I can try to set it by default.

nikwest, 2021-05-04:
yes, I was also investigating. I am currently trying to build from source. I created a new OMR_TARGET="apu2", added it to the build script to actually have x86_64 as real target. I then configured some options with make menuconfig, which openwrt recommended:

`CONFIG_PACKAGE_kmod-pcengines-apuv2=y
CONFIG_PACKAGE_amd64-microcode=y
CONFIG_PACKAGE_kmod-crypto-hw-ccp=y
CONFIG_PACKAGE_kmod-crypto-rsa=y
CONFIG_PACKAGE_kmod-leds-apu=y
CONFIG_PACKAGE_kmod-leds-gpio=y
CONFIG_PACKAGE_kmod-gpio-nct5104d=y
CONFIG_PACKAGE_kmod-sp5100-tco=y
CONFIG_PACKAGE_kmod-sound-core=y
CONFIG_PACKAGE_kmod-pcspkr=y
CONFIG_PACKAGE_kmod-usb-ohci=y
`

then also added the serial options:

`CONFIG_GRUB_SERIAL="ttyS0"`

to the config-apu2.

Unfortunately it didn't seem to create the images correctly.

So I build it in the apu2/source sub directory with make. This way it seems to work. I have no experience with openwrt and also the build scripts you created. So looks like I'm still missing something.

However the current master branch doesn't really work for me (e.g. dhcp option seems to always that IP address from eth0 ... ) Haven't done any detailed analysis yet though.

I guess it's probably better to try to build the 0.57 version from source to not have to many changes at a time.

closed 2021-08-07 by github-actions[bot]

saswatasarkar13, 2023-10-06:
You don't need to make a custom image.

saswatasarkar13, 2023-10-06:
You don't need to make custom-builds. Just add ```rootwait console=tty0 console=ttyS0,115200n8 noinitrd```  
where you see the pattern and change the 115200 to your device's baud rate in the /boot/grub/grub.cfg.