Ysurac/openmptcprouter #1795: same speed without vps

ciasco92, 2021-04-29:
hi, I have the openmptcprouter version v0.57.3 and two 4g modems, individually they download at 60 mbps, I have correctly installed the vps and my raspberry pi 4, I have correctly set the ip of my two 4g modems, I get my ip address of the vps, but when I do a speed test I get exactly the same speed as a single wan (60mbps) but I noticed that both modems use data (30mbps wan1 and 30mbps wan2).
both 4g connections are the same, same ping (30ms) and same speed

`omr-test-speest = 
100  188M    0  188M    0     0  7544k      0 --:--:--  0:00:25 --:--:-- 8193k
`

`omr-test-speed wan1 =
running test:
ipset v7.6: Syntax error: cannot parse 195.181.175.51195.181.175.49195.181.175.54: resolving to IPv4 address failed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:07 --:--:--     0
`

`omr-test-speed wan1 =
100  312M    0  312M    0     0  7625k      0 --:--:--  0:00:41 --:--:-- 8007k
`

[image]

ciasco92, 2021-04-30:
i forgot to mention that one connection has public ip, while the second one has nat, i don't know if this is important, but i have enabled "mptcp on vpn" on the second connection and now it seems to work

hairy-dog, 2021-05-01:
If each of the WAN modems can transfer data at 60mb/sec, if the system was to work at their full speed, there would be 120mb/sec going out through the eth0 port to the wan connections as well as 120mb/sec eth0 port from the lan.
I don't think the pi4's ethernet port can cope well with that much traffic.
I suggest that you buy two usb to ethernet adapters (they are not expensive), so that you can have the three connections on three different ethernet ports. 
You will probably see a massive speed improvement. I did!

ciasco92, 2021-05-01:
hello, thanks for the reply, in theory the rpi4 port is gigabit, also I know people who have a Raspberry Pi 4 with opennmptcp and aggregate two 4g connections reaching 330 mbps

hairy-dog, 2021-05-01:
I wonder what speeds they would have seen with two ports?
Even though the average throughput of a 120mb/sec connection might only be 240mb/sec, there will be times when it is twice that, and other times when it is lower. The peak data flows could easily be in both directions. I'm not sure how well the theory says it should work, but I suspect that it is a bottleneck.
And of course, a gigabit port is still connected to a cable, a switch, another cable. Any one of these might not really be good for high rates.
Adding a usb3 to ethernet port is very cheap and very easy. It made a significant difference for me.

ciasco92, 2021-05-04:
I think I have solved it, using band 7 on one modem and band 3 in another I get aggregation, if I use the same band for both I don't get aggregation, I have to use different bands, very curious