Ysurac/openmptcprouter #1798: OMR as a WAN

PxxBxx, 2021-05-01:
Hello OMR community (and Ycarus).
I want to setup OMR so that I only use its aggregation feature.
Today, I've a Unifi USG Gateway that is the only client (.100.2/24) of the OMR.
OMR is then a "WAN" for the gateway, and I've got my LAN in (.1.0/24), USG manage all this.
-> double-NAT IPV4 !! not optimal :-/ It works great, but...
(note : I don't use OMRBypass, so it's no big deal)

I would love to go IPV6 (thanks Milkywan).
Do you think there could be a way to easily "passthough" all IPV6 through USG+OMR ? and benefit from aggregation ?
Example :
* activate IPV6 on OMR
* setup my ::/48 prefix delegation on both OMR and USG
* setup Router Advertisement only on the USG

I'm a newb regarding IPV6 btw.... So If you think that's feasible, I'd love to be guided and have things explained. (what ? it's not christmas yet ?)

Thanks a lot.

Ysurac, 2021-05-03:
You need to have lan interface in the /48 prefix, then use it as gateway for USG. Should work.

closed 2021-08-06 by github-actions[bot]

BenHarris, 2022-02-16:
@PxxBxx did you manage to get this working?