Ysurac/openmptcprouter #1802: Haute Disponibilité entre 2 routeurs

coulicoulaser, 2021-05-02:
Bonjour à tous,

J'aurais souhaité savoir si il était possible de mettre en place de la HA entre 2 routeurs omptcp ?

J'ai vu que c'était possible sur OpenWRT, mais est-ce que cette HA serait compatible avec la surcouche omptcp ?

Ce serait intéressant notamment dans le cas où la machine hébergeant le routeur plante, d'avoir une porte de secours pour que internet ne soit pas coupé.

Merci !

Ysurac, 2021-05-03:
Normalement oui avec keepalived. Par contre un seul routeur peut se connecter au VPS.

coulicoulaser, 2021-05-05:
Merci pour la réponse ! Je vais installer mon nouveau serveur ce week-end, j'essaierai de voir si je peux faire un POC pour mettre ça en place.

coulicoulaser, 2021-05-08:
Bonjour,

Existe-t-il une commande permettant de couper et remettre la connexion au VPS manuellement ?

Commande que je pourrais essayer d'interfacer avec keepalived, soit par un script externe soit dans keepalived même (il faut encore que je me renseigne sur les possibilités précises de keepalived).

Merci !