Ysurac/openmptcprouter #1809: Three 4G connections, but the equivalent of the speed of a single ...

Gaou3, 2021-05-06:
Hello,

I have been using OpenMPTCProuten aggregation for a few weeks with a 4G Free mobile and an ADSL Free.

We had a very good connection thanks to 4G.

We recently added a 4G Orange, as well as a 4G SFR, and I am surprised to see that the difference is not huge ...

However we checked that the other two operators were well established in our area, and correctly directed our antennas ...

But when I disconnect the 4G Free (the good one) and leave the other 2 4G + ADSL, I have the connection equivalent to ADSL alone.

An idea ? Thank you

## Expected Behavior
A better upload test for each 4G operator

## Current Behavior
13 Mbps upload with free, and 1 Mbps for each others (SFR and Orange)

## Specifications

  - OpenMPTCProuter version: v0.57.3
  - OpenMPTCProuter VPS version: v0.57.3
  - OpenMPTCProuter platform: Raspberry 4 Pi

Ysurac, 2021-05-06:
The problem is only with upload ? No problem in upload ?

Gaou3, 2021-05-06:
The problem is for upload and download.

When my 4G free is connected with the 2 others and ADSL , I have 50 Mbps download and 12 Mpbs upload.
When 4G Free is disconnected and the 2 others+ ADSL connected, I have 12 Mpbs download and 1 Mpbs upload.
And when I only have ADSLO connected, I have 12 Mpbs download and 1 Mbps download.

Ysurac, 2021-05-06:
All is green in status page ?
You can try via ssh with ```omr-test-speed```, ```omr-test-speed wan1```... (Ctrl+C to stop after at least 2 minutes)

Gaou3, 2021-05-06:
All is green, but I have regularly disconnections with vps (OVH), OpenMPTCProuter, and the 2 routers from the 2 new 4G.

Sorry, but what is ssh, where could I find it to put the commands ? (Is it linked to the vps ? with Putty ?) 
Sorry for my ignorance

Ysurac, 2021-05-06:
You can use putty on OpenMPTCProuter IP.

For the disconnection you should check in State->System log what is detected.

Gaou3, 2021-05-06:
Ok, for Putty, it ask me a password, but I didn't set a pasword in OpenMPTCProuter ...  Access denied ... I don't understand.

For the system log, here it is:
Thu May  6 10:19:00 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Thu May  6 10:19:01 2021 user.notice omr-tracker: Launching...
Thu May  6 10:19:03 2021 user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...
Thu May  6 10:19:07 2021 user.notice omr-tracker: Launched
Thu May  6 10:19:07 2021 user.notice omr-tracker: Launching...
Thu May  6 10:19:07 2021 user.notice omr-bypass: Starting OMR-ByPass...
Thu May  6 10:19:08 2021 user.notice omr-bypass: Restart dnsmasq...
Thu May  6 10:19:08 2021 daemon.info dnsmasq[23565]: exiting on receipt of SIGTERM
Thu May  6 10:19:09 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu May  6 10:19:09 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu May  6 10:19:09 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: started, version 2.82 cachesize 150
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: DNS service limited to local subnets
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu May  6 10:19:09 2021 daemon.info dnsmasq-dhcp[3181]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using only locally-known addresses for domain test
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using only locally-known addresses for domain onion
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using only locally-known addresses for domain localhost
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using only locally-known addresses for domain local
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using only locally-known addresses for domain invalid
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using only locally-known addresses for domain bind
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using only locally-known addresses for domain use-application-dns.net
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using only locally-known addresses for domain lan
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using nameserver 127.0.0.1#5353
Thu May  6 10:19:09 2021 user.notice omr-bypass: OMR-ByPass is running
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: using 1 more local addresses
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: read /etc/hosts - 4 addresses
Thu May  6 10:19:09 2021 daemon.info dnsmasq[3181]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu May  6 10:19:09 2021 daemon.info dnsmasq-dhcp[3181]: read /etc/ethers - 0 addresses
Thu May  6 10:19:12 2021 user.notice omr-tracker: Launched
Thu May  6 10:19:13 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Thu May  6 10:19:14 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.0.0.1)
Thu May  6 10:19:16 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Thu May  6 10:19:18 2021 user.notice omr-tracker: Launching...
Thu May  6 10:19:23 2021 user.notice omr-tracker: Launched
Thu May  6 10:19:23 2021 user.notice omr-bypass: Starting OMR-ByPass...
Thu May  6 10:19:24 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.0.0.1)
Thu May  6 10:19:24 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Thu May  6 10:19:26 2021 user.notice omr-bypass: Restart dnsmasq...
Thu May  6 10:19:26 2021 daemon.info dnsmasq[3181]: exiting on receipt of SIGTERM
Thu May  6 10:19:26 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu May  6 10:19:26 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu May  6 10:19:26 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: started, version 2.82 cachesize 150
Thu May  6 10:19:26 2021 user.notice omr-bypass: OMR-ByPass is running
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: DNS service limited to local subnets
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu May  6 10:19:26 2021 daemon.info dnsmasq-dhcp[8351]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: using only locally-known addresses for domain test
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: using only locally-known addresses for domain onion
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: using only locally-known addresses for domain localhost
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: using only locally-known addresses for domain local
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: using only locally-known addresses for domain invalid
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: using only locally-known addresses for domain bind
Thu May  6 10:19:26 2021 daemon.info dnsmasq[8351]: using only locally-known addresses for domain use-application-dns.net
[... cut ...]
Thu May  6 10:51:46 2021 authpriv.info dropbear[32692]: Exit before auth from <192.168.100.21:60157>: Max auth tries reached - user 'is invalid'
Thu May  6 10:52:22 2021 authpriv.info dropbear[28303]: Child connection from 192.168.100.21:60176
Thu May  6 10:52:24 2021 authpriv.warn dropbear[28303]: Login attempt for nonexistent user
Thu May  6 10:53:12 2021 authpriv.info dropbear[28303]: Exit before auth from <192.168.100.21:60176>: Exited normally
Thu May  6 10:53:23 2021 authpriv.info dropbear[10779]: Child connection from 192.168.100.21:60192
Thu May  6 10:53:26 2021 authpriv.warn dropbear[10779]: Login attempt for nonexistent user

Gaou3, 2021-05-06:
Ok, sorry, I did it with the OpenMPTCProuter in my vps (sorry for my ignorance)

Here is the result:
[image]

Ysurac, 2021-05-06:
omr-test-speed need to be run on router side. User root and no password by default.

Many problems in log... WANs are ethernet router or USB dongles ?

Gaou3, 2021-05-06:
Ok, here is the result:

[image]

WANs are ethernet router

Gaou3, 2021-05-06:
If that can help to detect the problem, I notice that the disconnections of wan 3 and 4 are often done when I run a speed test on the internet to check the speed.
I made a lot of speed test today, and notice during the test that the State windows show me Wan 3 or 4 (or both) disconnected. And reconnect after the speed test.

Ok, I forget to show you the others tests:
omr-test-speed wan1
[image]

omr-test-speed wan2
[image]

omr-test-speed wan3
[image]

omr-test-speed wan4
[image]

Ysurac, 2021-05-06:
For wan2 & wan3, it's a bug in omr-test-speed, fixed in v0.58rc2
I see 11.9M on wan1, what is the VPS max speed ? what is the load average on status page of router and VPS ?
Who is the VPS provider ?

You can try to set SQM on wan2 & wan3.

Ysurac, 2021-05-06:
And what is wan4 ? This seems really slow in speed test.

pbapba512, 2021-05-06:
If you using USB sticks and they are disconnect when you do a speedtest measure, its worth to check the USB ports power output. RPI4 can "drive" 2 USB stick in paralel, but 3 pcs is too much. In this case you have to use external powered USB3 HUB!

Gaou3, 2021-05-07:
Hello Ysurac, thank's for your support !

Is there an easy way to upgrade VPS and Raspberry to OpenMPTCProoter 0.58rc2 ? Or do I have to reinstall all the system in VPS and in Raspberry ?

About my VPS, the VPS max speed is 250 Mbit/s , and the VPS provider is OVH (server located in Graveline, in my country: France)
[image]

Where could I find the load average of router and VPS ? 
Is that right ?

[image]

Wan4 is the SFR 4G router. It's very surprising, because we haven't used up all our data on this subscription ...
It would be more normal on the Wan 3 corresponding to our Orange subscription, because we used all the data.

For pbapba512, my 4 Wan are ethernet ports, and no USB.
But I tried yesterday to make a USB tethering with my smartphone, and it's run very well ! This gives me a good workaround in case the Wans don't work (but I'd rather not have to do this fallback and my Ethernet routers work). I have a video transmission to do tomorrow, with video duplex, and everything should work.

I still wanted to add a second smartphone in USB, but when I configure a second USB interface, I have the impression that the 2 share the same IP address. I have the impression that it does not work as it should, and the bandwith is not consistent.

Gaou3, 2021-05-10:
Ok, what other hosts do you recommend?

busybox11, 2021-05-15:
Hey,
Je recommande chaudement ONETSOLUTIONS pour des serveurs dédiés / VPS pas cher et performants.
Je suis client depuis deux ans et je n'ai rien à redire sur les performances et le débit, tu seras peut être limité par les 150Mbps de throughput mais il me semble qu'il n'y a pas de data cap.

adamxp12, 2021-08-23:
I too am having this problem
3x 4g connections which on their own can get like 20/30 down 10 up are getting 20 down about 10 up with OpenMPTCPRouter
In fact one router on its own is often faster than this.

Like I know its not magic and assumed with 2 routers it was not able to benefit but adding a 3rd router made zero difference despite them all showing activity when the server is running.

I am using a Vultr VPS which I tested the speed on and its getting gigabit speeds to the internet. Decent CPU/RAM usage (same on my local VM which is running in Hyper-V) I dont really understand why its not working as it should. the is literally zero benefits for me using this with 3x connections :| hoping its some bug or config issue as the idea is really solid

Ysurac, 2021-08-23:
@adamxp12 make same tests. Check that all connections are not on same antenna with a limited bandwidth.

adamxp12, 2021-08-23:
@Ysurac Using 3x different providers. 2x are on the same mast mind you. but the other one is on its own mast so cant see why I cant get beyond the speed of one router. I aint expecting a 3x speed boost of course but its not making a noticeable difference aside from maybe more stable? hard to tell all I know is the speed aint going up which is what I need in my rural location :(

Ysurac, 2021-08-24:
@delferius You are using a RPI3, don't expect too much from it. CPU is quite slow and ethernet port too. Try to run "omr-test-speed" without argument to check aggregated speed from the router.

@adamxp12 Did you run ```omr-test-speed``` tests ?

delferius, 2021-08-24:
@Ysurac thanks for your help, you expect the RPI3 is not powerful enough to aggregate connection taken over 100 mbit? Do you think it would be a solution to replace my RPI3 by a RPI4 to fix my problem?

[image]

Ysurac, 2021-08-24:
@delferius You can try to disable encryption in System->OpenMPTCProuter, "advanced settings" checkbox. I would need to test again on my RPI3, but this seems possible. You can always test speed via a virtualbox/vmware/... VM on your computer if it's better or not.

adamxp12, 2021-08-24:
@Ysurac Yes the speed tests on OMR are identical to that of my client. Each connection is testing fine its just not working great bonded.

Have disabled encryption where possible as I dont know if my Hyper-V VM supports hardware encryption or not tbh. but still only performing as good as one link. My providers are using CGNAT dont know if that breaks this idea or not fell like this idea works better with wired network connections but I would need like 10x DSL lines to get the same performance as a single 4g router lol. and I do need better than a single 4g router provides

Ysurac, 2021-08-24:
@adamxp12 would be better to have result pasted here.
All is green in status page ? You checked MPTCP support of each WAN in Network->MPTCP ? What is load average of both VPS and router part ? What is the latency of each WANs ?

hairy-dog, 2021-08-24:
I can assure you that using three 4g connections can get almost the sum of the three using a pi4. Not so sure with a pi3. My three 4g connections are to different base stations (two of them to different networks). 
However, the results are very variable. At some times, the downlink speeds fall off dramatically. I think this is simply congestion in the mobile networks, because the uplink speeds remain high all the time. The slowdowns are not always at predictable times.
I found that adding a usb3 network adapter speeded things up a lot, but in retrospect I suspect that this was improved by bypassing a rather slow network switch.

adamxp12, 2021-08-24:
Have attached a pic of the status page. shows the latencys there

All 3 are showing TCPOptionMPTCPCapable  instead of -TCPOptionMPTCPCapable so assume they support MPTCP?
I dont see anything under the MPTCP established connections though

as for speeds via OMR-speedtest
WAN1 3570k
WAN2 300k (this is much much slower than it really is?)
WAN3 1137k (again much slower than if I test outside of OMR)
Overall bonded speed: 900k thats 3x the slowest link which makes sense

So I removed WAN2 so should be 2000k on the test but still only getting 800-900k despite the "slow" link being removed?

Maybe I am doing something wrong. or maybe these 3x connections cant be bonded as I can see WAN1 is much faster. but still it dont seem to be doing much bonding when I can get faster with just WAN3 

[image]

Ysurac, 2021-08-24:
You should get some data in Network->MPTCP, MPTCP established connection tab.
Check also fullmesh tab.
Try another connection as master.

Ysurac, 2021-08-24:
Check also that promiscuous mode is enabled for interface in your Hyper-V or try to create a dedicated interface for each wan if you are use macvlan.

adamxp12, 2021-08-24:
@Ysurac Changing the master has made an entry show in the established connections.

Dont see an option for promiscuous mode? googling how to enable that on hyper-v showed mostly stuff about port mirroring?

closed 2021-11-28 by github-actions[bot]