Ysurac/openmptcprouter #1810: Shorewall rules replication when source IP is specified

gnosoz, 2021-05-07:
## Current Behavior
It appears that there is constant replication of rules where in there is a specific IP to accept traffic from an IP address
DNAT		net:<IP ADDR>		vpn:$OMR_ADDR	tcp	PORT	-		# OMR openmptcprouter redirect router xxx port tcp from 51.xxx.xxx.xxx
DNAT		net:<IP ADDR>		vpn:$OMR_ADDR	udp	PORT	-		# OMR openmptcprouter redirect router xxx port tcp from 51.xxx.xxx.xxx

This is replicated in some cases up to hundreds of times per rule which I then need to manually remove via cron job using awk as follows
awk '!seen[$0]++' /etc/shorewall/rules |tee /etc/shorewall/rules.uniq
mv /etc/shorewall/rules.uniq /etc/shorewall/rules
shorewall restart

## Possible Solution
Maybe is possible to check with a sed if a rule is already in the rules file prior to add it again
**Please note that this is issue it appears to be  happening only when specific IP of source are added and that for standard fìport forwarding instead is working like a charm with no duplication whatsoever**

## Steps to Reproduce the Problem

  1. create in the router via luci a new firewall port forward rule specifying in advanced setting the source IP address from which it needs to accept traffic from
  2. verify syslogs of the server how many times it runs the shorewall reload
  3. check /etc/shorewall/rules to verify that there are multiple lines with same exact details

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: VPS 0.1024
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-05-07:
Should be fixed in snapshot VPS script and v0.58rc1.

gnosoz, 2021-05-07:
fantastique :)