Ysurac/openmptcprouter #1811: Change Ports

homicida2000, 2021-05-07:
Hi,
where can I change the Network-Ports ? on router and vps ?

Ysurac, 2021-05-07:
You can't change them easily for now.

homicida2000, 2021-05-13:
Initial situation: IONOS root server (one IP) with proxmox and several OpenMPTCProuter VPS instances.

Every instance has a normal debian setup with debian10-x86_64.sh from git.

ORM at several locations normally set up.

Problem:
1. Root server has 10.255.255.1 as PTP gateway.
2. Glorytun-tcp has 10.255.255.1 as the host IP.
3. Ports in "VPS" and "ORM" are permanently stored.

Solution:

SERVER:
`
#!/bin/bash

###CHANGE VPN-IP-PREFIX for host like IONOS with Gateway 10.255.255.1
IP_CHANGE="10.255./172.17."

##CHANGE PORTS OldPort/NewPort
PORTS_CHANGE="65001/64001 65101/64101 65222/64222 65228/64228 65400/64400 65500/64500 65000/64000 "

IP_FILES="
/etc/glorytun-tcp/post.sh
/etc/glorytun-udp/post.sh
/usr/local/bin/omr-admin.py
/usr/local/bin/omr-service
"

PORT_FILES="
/etc/glorytun-tcp/tun0
/etc/glorytun-udp/tun0
/etc/openmptcprouter-vps-admin/omr-admin-config.json
/etc/shadowsocks-libev/manager.json
/etc/shorewall/rules
/etc/v2ray/v2ray-server.json
/etc/ssh/sshd_config
/usr/local/bin/omr-admin.py
/usr/local/bin/omr-service
/usr/share/shadowsocks-libev/manager.json
/var/lib/shorewall6/.start
/var/lib/shorewall6/.reload
/var/lib/shorewall6/.ip6tables-restore-input
/var/lib/shorewall6/firewall
/var/lib/shorewall/.start
/var/lib/shorewall/firewall
/var/lib/shorewall/.iptables-restore-input
/var/lib/shorewall/.reload
"

for f in $IP_FILES
do
        echo "Ändere IP in $f"
        sed -i "s/${IP_CHANGE}/g" $f
done

for f in $PORT_FILES
do
        echo "Ändere Ports in  $f"
        for p in $PORTS_CHANGE
        do
        sed -i "s/${p}/g" $f
        done
done
echo "Done."
`

ROUTER:
`#!/bin/sh

PORTS="65001/64001 65101/64101 65222/65222 65228/64228 65500/64500 65400/64400 65201/64201 65301/64301 65011/64011"

FILES="
/etc/config/dscp
/etc/config/dsvpn
/etc/config/glorytun
/etc/config/glorytun-udp
/etc/config/mlvpn
/etc/config/iperf
/etc/config/openmptcprouter
/etc/config/shadowsocks-libev
/etc/config/v2ray
/etc/init.d/mptcpovervpn
/etc/init.d/openmptcprouter-vps
/etc/init.d/mptcpovervpn
/etc/init.d/openmptcprouter-vps
/usr/lib/lua/luci/controller/openmptcprouter.lua
/usr/libexec/rpcd/openmptcprouter
/usr/share/ModemManager/mm-foxconn-t77w968-carrier-mapping.conf
/usr/libexec/rpcd/openmptcprouter
/usr/share/tracebox/omr-mptcp-trace.lua
/usr/share/omr-update/3010-omr-v2ray
/usr/share/omr-update/2020-omr-vpn
/usr/share/omr-update/openmptcprouter
/usr/share/tracebox/omr-mptcp-trace.lua
/var/etc/shadowsocks-libev/ss_redir.hi2.json
/var/etc/shadowsocks-libev/ss_redir.hi.json
/var/etc/shadowsocks-libev/ss_local.tracker.json
/var/luci-modulecache/6C7563692E636F6E74726F6C6C65722E6F70656E6D70746370726F75746572
"

for f in $FILES;
do
        echo "FILE : $f"

                for p in $PORTS;
                  do
                  sed -i "s/${p}/g" $f
                  done
done
`

According to this scheme, you can set up as many vpn as you like.
routing / port forwarding must be done in the Proxmox gateway.

Bumpalert, 2021-05-13:
@homicida2000 thts awesome. I was also looking to merge into single one as cpu load is low as streaming load comes once in a while.

closed 2021-08-17 by github-actions[bot]