Ysurac/openmptcprouter #1812: Charge Processeur Pi 4B 2Go à plus de 4 => obligé de reboot tous les 2 jours

JeedHome44, 2021-05-08:
## Expected Behavior

Bonjour,
Je n'ai plus de problème de WAN maintenant mais je me rends compte qu'au bout de 2 jours, générelement, la charge du processeur monte doucement durant la seconde journée. Le soir, tout rame et ensuite plus d'internet ni de LAN dispo.

Quand je veux consulter OMR sur mon PC fixe, ca rame à mort pour ouvrir le 192.168.100.1 et la charge a dépassé les 4 pour la minute, les 5 minutes et les 15 minutes.

## Current Behavior

J'ai regardé les logs mais je ne vois rien... j'y connais rien non plus... est-ce parce que j'ai trop d'appareils, trop de by-pass... que ca foire comme ça? Devrais-je changer le Pi 4B par autre chose de plus puissant vu mon installation?

## Specifications

  - OpenMPTCProuter version:  0.57.3
  - OpenMPTCProuter VPS version:  0.1024
  - OpenMPTCProuter platform:  Pi 4B 2Go

Ysurac, 2021-05-08:
Que donne un htop via SSH ? qui utilise le processeur ?

JeedHome44, 2021-05-08:
Bonjour,

Il n'y a que OMR sur la carte SD de ce Pi. Faut il faire la commande quand il est surchargé?

JeedHome44, 2021-05-08:
Peut on mettre OMR sur un NUC?

JeedHome44, 2021-05-08:
Ok donc ça vient donc peut être de la ma latence. Si le Pi est surchargé...
Un ami me dit que Open WRT est installable sur le Xiaomi AX3600. Je viens juste de l'avoir.

Ça serait donc super si OMR pouvait s'installer dessus et ça permettrait de passer le DHCP et l'agrégation directement sur le Xiaomi AX3600.

Voici le lien qu'il m'a passé:

https://openwrt.org/inbox/toh/xiaomi/xiaomi_ax3600

Peux tu me dire si on peut le faire avec OMR?

JeedHome44, 2021-05-08:
Ça a l'air complexe quand même à mettre sur le AX3600.

Quel appareil me conseilles tu supportant AES-NI

Ysurac, 2021-05-08:
Je supporte que le matériel que je possède.
Pour l'AES-NI, il y a pas mal de processeur Intel supportant ça. Il y a des NUC à base d'Intel qui sont pas mal vendu un peu partout.

Speedy37, 2021-05-11:
On dirait bien que c'est le même soucis que moi (fin du thread #1619).

Regarde si tu as des processes en charge: http://192.168.100.1/cgi-bin/luci/admin/status/processes

Ysurac, 2021-05-11:
Oui, il me faudrait le résultat d'un htop.

closed 2021-08-15 by github-actions[bot]