Ysurac/openmptcprouter #1813: High performance MPTCP

Corrado56, 2021-05-08:
Hi Yannick, I want only notify you of this video:

https://youtu.be/L7Wy_WKIUD0

This is an interesting Hardware / VPS choice, and the results are excellent. 
It's not cheap. but also linksys 3200 is not cheap
Unfortunately the video is in Italian language, but for technical people it's understandable also without audio.
At the end we have two series of Speed tests, the first bonding 3 connections, the second bonding 5 connections, from FTTC to 4g all togheter.

Regards

vinceducat, 2021-05-08:
@Corrado56  Thanks for this link ;-) @Ysurac  wath doo you think about this hardware ?

hikey, 2021-05-08:
I run the HUNSN.
Model：VMware, Inc. VMware Virtual Platform
Architecture：Intel(R) Core(TM) i5-4250U CPU @ 1.30GHz

But i run OpenMPTCP under esxi, i get the max speed of my connections.

Adorfer, 2021-05-08:
is there SOMETHING new in the video which hadn's been discussed in the previsous hardware-selection issue before?
https://github.com/Ysurac/openmptcprouter/issues/1108

hikey, 2021-05-08:
I have not watched the video, i can just tell you it work perfectly for me. (esxi 7.0)
However i do not use USB adapters 4/5g connections

Just Docsis3.1 and VDSL combined

Adorfer, 2021-05-08:
My feeling is, that the offered solution is more of the "mid performce" range, since it does not exceed 1GO/s.

hikey, 2021-05-09:
You can do 1gb/s cpu is powerfull enough for that.

Malaga82, 2021-05-09:
he used omr as default host OS, a bit over-rated...
as @hikey said, it's better to have it as a guest, under esxi or, if i can suggest, proxmox...or any other hypervisor: you have all benefits from virtualization without performance hits and waste of resources.
other than that, make sure both omr and vps uses AES, you'll have much better cpu usage.

Adorfer, 2021-05-09:
> You can do 1gb/s cpu is powerfull enough for that.

But you can not get more than 1GBit/s out of the device, since there are no 2.5G ethernet or SFP+ on the suggested device. 
And LAPC is -to my knowledge- not available in OpenWRT. (or the CPU will not handle it well due to the lack of offloading)

So in case you want "high performance", there is no way around virtualization on a host with capable networking.

hikey, 2021-05-10:
But what do you exactly want to achieve?
MPTCP , can handle any speed as long as you use the right hardware, doing 10Gb/s you need serious hardware on both ends.
You still can do NIC bonding , on the HUNSN you have 6 ports, i just can say that the HUNSN is a good minipc and is overkill for most of the users it can run 1GB/s easy.

Adorfer, 2021-05-11:
> You still can do NIC bonding ,

Does it work OpenWRT "bare metal" or do you need an hypervisor in between? 

> You still can do NIC bonding , on the HUNSN you have 6 ports,

The proposed solution here is not a HUNSN, as far as i understand. 
That's why i say: No, that's not a solution for "High performance", that's the mid-tier.

faddat, 2021-07-26:
My openptcp is bare metal and I love it.

I might try and make a lil portable openmptcprouter

closed 2021-10-30 by github-actions[bot]