Ysurac/openmptcprouter #1818: Install on Unsupported OpenWRT Router

GTiDon, 2021-05-10:
Hello

Is it possible to install Opem MPTCPRouter onto an unsupported router?

How do I set this up? I have OpenWRT running

Ysurac, 2021-05-10:
You need to compile the image: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform
Then this may or not work.

GTiDon, 2021-05-10:
Do I run these commands on the router? Then flash the firmware BIN it creates?

Sorry my Linux knowledge is very limited

GTiDon, 2021-05-10:
But then how is it compiled for my specific device?

GTiDon, 2021-05-14:
I am going over the Doc but dont understand it all.
I have built a linux VM and I have run the commands to install the tools (Some errors)

apt-get install busybox curl rsync build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler wget

I dont understand this next step?  WHat DO I enter for the platform? Which machine must this be run on?

# git clone https://github.com/Ysurac/openmptcprouter.git
# cd openmptcprouter
# git checkout master
# OMR_TARGET="myplatform" OMR_FEED_SRC="master" ./build.sh

Ysurac, 2021-05-14:
It does cross compilation, so on the machine you want to use for compilation.
For the myplatform, you need to enter the platform you want to compile to, like x86_64 or rpi4.

GTiDon, 2021-05-14:
It's a OpenWRT router I want to run it on

Ysurac, 2021-05-14:
If it's not a supported arch, you need to know the arch used and then you will need to configure it manually with ```make menuconfig```, as indicated in the doc.

GTiDon, 2021-05-14:
What Arch is this?

[image]

GTiDon, 2021-05-14:
[image]

Ysurac, 2021-05-14:
It's a ZBT WE-826, so based on MT7620 on ramips arch.
But with a so slow CPU, so small flash and really small ram, I think this will be difficult to use OpenMPTCProuter.

GTiDon, 2021-05-14:
I still want to test. It has 32Mb flash

Will it be possible?

Ysurac, 2021-05-14:
I don't think, but you can test if you want ;)

GTiDon, 2021-05-14:
Can you please help me with the cross platform build steps? I really don't understand from the documentation what I need to enter
[image]

Ysurac, 2021-05-14:
I don't see the image.

GTiDon, 2021-05-16:
I have selected the Arch on the list. Is there anything else I should select? what can I remove to make a lighter build?

[image]

Then what should the config file be saved as?

[image]

Ysurac, 2021-05-16:
You need at least to select Target, Subtarget and Target profile.
Keep default name for config.

GTiDon, 2021-05-16:
What is the default name? There is no name when I click save. It asks as above

Ysurac, 2021-05-16:
As above, and as default, it's ```.config```

GTiDon, 2021-05-19:
I keep getting errors.

[image]

Adorfer, 2021-05-19:
> Will it be possible?

No. flash and ram is both too small. and if you solder bigger on, then you will find a performance lower than 15MBit/s.

GTiDon, 2021-05-19:
[image]

Bumpalert, 2021-05-31:
@Ysurac  I tried to build for rpi4 target to make sure environment is correct. But it failed as below complaining ab out libparam & kmod-mt6625l-wlan-gen2.Which packages need to be included to the dependency list?

-------------
Overriding core package 'umbim' with version from openmptcprouter
Done
Building openmptcprouter for the target rpi4
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
#
# configuration written to .config
#
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
time: target/linux/prereq#0.26#0.13#0.37
 make[1] world
 make[2] tools/compile
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/patch compile
 make[3] -C tools/tar clean-build
 make[3] -C tools/tar compile
    ERROR: tools/tar failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/media/storage/openmptcprouter/rpi4/source/include/toplevel.mk:230: world] Error 1

Bumpalert, 2021-06-02:
@Adorfer @Ysurac can you help where can I get libpam. I tried openwrt compiling to clear this dependency but didn't work.

Ysurac, 2021-06-02:
@Bumpalert you don't need to solve the libpam warning to compile.

closed 2021-09-06 by github-actions[bot]