Ysurac/openmptcprouter #1821: Raspberry 3 interfaces

gianlucabarani, 2021-05-12:
Hi Raspberry 3 has 1 eth . I try to setup 2 Wan router to the same switch connected ETH Raspberry. But seem Only One work. It s because i Need several different interfaces? I try to setup vlmac but seems not work. So at the end with Raspberry i can connect 1 via USB , 2 via ETH and 3 via WiFi. Are there other option?

Ysurac, 2021-05-12:
MacVLAN create "virtual" interfaces. 2 interfaces are created by default on RPI.
You don't need to add USB adapters (that work badly on RPI <4).

gianlucabarani, 2021-05-12:
Ok when i create a MacVlan in phisical Tab What do i set? Leave blank or i set eth0? Thanks

Ysurac, 2021-05-12:
Use eth0. You can also use the wizard.

gianlucabarani, 2021-05-12:
Ok set eth0.1 but seem there Aren't packet TX and TX flowing in interfaces view. Packet flow Only of i set phisical interfaces eth0 sorry
[image]

Ysurac, 2021-05-12:
VLAN can't be used with MacVLAN.

Ysurac, 2021-05-12:
You can use VLAN without using MacVLAN in this case.

gianlucabarani, 2021-05-12:
Ok changing MacVlan to normal VLan packets now flowing only in One direction TX but i think that i set the same VLan tag on the Wan router Port side.

gianlucabarani, 2021-05-15:
Ok i buy a Layer2 switch and work fine using VLan. Thanks.

closed 2021-05-15 by gianlucabarani