Ysurac/openmptcprouter #1822: Remote access,

Nickmcl024, 2021-05-12:
Hello, is it possible to gain remote access to the router 
With a DDNS running on it,?

Ysurac, 2021-05-12:
You can use DDNS or set a domain on VPS IP and redirect port from VPS to router.

Bumpalert, 2021-05-12:
Did you get a chance to fix the issue when ddns domain name is used in vps
server address in 0.58beta

Ysurac, 2021-05-12:
which issue ?

Nickmcl024, 2021-05-12:
Amazing is there any setup guide or doc showing how to set it up and them, usb modems, are there any recommended modem that i can use the modem Manger, i have tried to flash my Huawei from hi link to stick mode but with no luck 

Then would this device work with it

ZAR 3,153.46  25%OFF | CHANEVE OpenMPTCPRouter 4 SIM Card LTE 4G Aggregation Wireless Router Multiple SIM Card Traffic Convergence Wi-Fi Router
https://a.aliexpress.com/_msykvop

Bumpalert, 2021-05-12:
https://github.com/Ysurac/openmptcprouter/issues/1619#issuecomment-830682956

@Ysurac/openmptcprouter ***@***.***>

Ysurac, 2021-05-12:
@Bumpalert I made some changes about that, but I need to test. But I think this issue is about remote access not using DDNS for VPS.
@Nickmcl024 for USB modems with ModemManager, you have to check on OpenWrt doc. I don't make a doc because I don't use that for now.
The device seems not so great for me, with a slow CPU if encryption is used, a low memory,... But I never tested.

Bumpalert, 2021-07-25:
> @Bumpalert I made some changes about that, but I need to test. But I think this issue is about remote access not using DDNS for VPS.
> @Nickmcl024 for USB modems with ModemManager, you have to check [...]

@Ysurac dynamic dns is still not able to be contacted by router.  but route is ressolving the IP

root@OpenMPTCProuter:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.255.255.1    0.0.0.0         UG    0      0        0 tun0
default         192.168.8.1     0.0.0.0         UG    11     0        0 eth1
10.255.255.1    *               255.255.255.255 UH    0      0        0 tun0
10.255.255.2    *               255.255.255.255 UH    1200   0        0 tun0
52.66.111.66    192.168.8.1     255.255.255.255 UGH   1      0        0 eth1
52.66.111.66    192.168.8.1     255.255.255.255 UGH   11     0        0 eth1
127.0.0.0       *               255.0.0.0       U     5      0        0 lo
192.168.0.0     *               255.255.0.0     U     7      0        0 usb0
192.168.8.0     *               255.255.255.0   U     11     0        0 eth1

root@OpenMPTCProuter:~# Sun Jul 25 12:26:38 2021 daemon.info omr-tracker-ss: Shadowsocks sss0 is up (can contact via http 198.41.212.162)
Sun Jul 25 12:26:38 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sun Jul 25 12:26:38 2021 user.notice Shadowsocks: Rules UP
Sun Jul 25 12:26:38 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Jul 25 12:26:38 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jul 25 12:26:40 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jul 25 12:26:41 2021 daemon.info dnsmasq[18264]: read /etc/hosts - 4 addresses
Sun Jul 25 12:26:41 2021 daemon.info dnsmasq[18264]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Jul 25 12:26:41 2021 daemon.info dnsmasq-dhcp[18264]: read /etc/ethers - 0 addresses
Sun Jul 25 12:26:41 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jul 25 12:27:16 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on vps.dnsfor.me, no server API answer on vps.dnsfor.me)
Sun Jul 25 12:27:31 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on vps.dnsfor.me, no server API answer on vps.dnsfor.me)
Sun Jul 25 12:27:37 2021 daemon.err /usr/bin/ss-redir[13109]: send: Connection reset by peer
Sun Jul 25 12:27:38 2021 daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1)
Sun Jul 25 12:27:38 2021 user.notice Shadowsocks: Rules DOWN
Sun Jul 25 12:27:49 2021 daemon.info omr-tracker-ss: Shadowsocks sss0 is up (can contact via http 151.101.129.164)
Sun Jul 25 12:27:49 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sun Jul 25 12:27:49 2021 user.notice Shadowsocks: Rules UP
Sun Jul 25 12:27:50 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Jul 25 12:27:50 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jul 25 12:27:52 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jul 25 12:27:52 2021 daemon.info dnsmasq[18264]: read /etc/hosts - 4 addresses
Sun Jul 25 12:27:52 2021 daemon.info dnsmasq[18264]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Jul 25 12:27:52 2021 daemon.info dnsmasq-dhcp[18264]: read /etc/ethers - 0 addresses
Sun Jul 25 12:27:52 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jul 25 12:27:56 2021 daemon.err /usr/bin/ss-redir[13109]: remote recv: Connection reset by peer
Sun Jul 25 12:28:12 2021 user.notice OMR-VPS: Can't get vps token, try later
Sun Jul 25 12:28:22 2021 user.notice OMR-VPS: Can't get vps token, try later
Sun Jul 25 12:28:50 2021 daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via http 77.88.55.77, 1.1.1.1, 74.82.42.42)
Sun Jul 25 12:28:50 2021 user.notice Shadowsocks: Rules DOWN
Sun Jul 25 12:28:57 2021 daemon.info omr-tracker-ss: Server sss0 (vps.dnsfor.me) seems down, no answer to ping
Sun Jul 25 12:29:03 2021 daemon.info omr-tracker-ss: Shadowsocks sss0 is up (can contact via http 198.41.212.162)
Sun Jul 25 12:29:03 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sun Jul 25 12:29:03 2021 user.notice Shadowsocks: Rules UP
Sun Jul 25 12:29:03 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Jul 25 12:29:03 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jul 25 12:29:06 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jul 25 12:29:07 2021 daemon.info dnsmasq[18264]: read /etc/hosts - 4 addresses
Sun Jul 25 12:29:07 2021 daemon.info dnsmasq[18264]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Jul 25 12:29:07 2021 daemon.info dnsmasq-dhcp[18264]: read /etc/ethers - 0 addresses
Sun Jul 25 12:29:07 2021 user.notice omr-bypass: OMR-ByPass is running

Bumpalert, 2021-07-25:
Just changed ddns name to actual ip & it connected like below

[image]