Ysurac/openmptcprouter #1824: Utilisation Zoom/Teams sur routeur OpenMPTCP

MathieuG9263, 2021-05-13:
Bonjour à tous 

Je cherche des retours d'experiences concernant l'usage de Zoom et Teams sur une solution OpenMPTPCP.

Pour vous brosser ma situation actuelle j'ai une OverTheBox chez OVH avec 2x8Mb en ADSL agrégés.

Malgré le paramétrage de règles de QoS sur leur solution, j'ai des coupures et du son métallique en communication.

Je ne cherche pas de support sur OTB car décidé à en sortir mais je cherche plutôt à savoir si certains d'entre vous ont cet usage visio, l'utilise sur OMPTCP et : 
- quel est le ressenti d'utilisation ?
- Ya t'il quelque chose à finetuner sur la solution pour accepter ce genre d'usage et éviter de retrouver les meme problèmes que j'ai chez OTB ?
- Plus globalement, peut on gérer la QoS sur la solution OpenMPTCP ?

Je vous pose ces questions en amont de la reception du matériel que j'ai commandé pour préparer un routeur MPTCP.
j'ai déjà monté le serveur.

Merci de votre aide :)

Mathieu

Ysurac, 2021-05-13:
Par défaut OpenMPTCProuter utilise SQM qui permet d'éviter la saturation d'une connexion et normalement les soucis que tu peux avoir.
Je l'utilise régulièrement pour Teams sans soucis.

Pour le "finetunage" je suis en train de créer une page "optimisation" dans la doc.

MathieuG9263, 2021-05-13:
Super 
Merci Yannick 
Je vais tester ça ;) 
Merci de m’avoir répondu aussi vite

MathieuG9263, 2021-05-14:
J'ai recu le matériel j'ai installé et tout fonctionne nickel mais je n'utiliseaps encore les 2 connexion adsl (j'ai testé sur de la fibre)

Si un utilisateur lance netflix (ou lance un gros telechargement) en meme temps qu'un autre utilisateur qui serait en session zoom, comment va se comporter le routeur ?  La session zoom pourrait elle etre perturbée ? 

Est ce que ca vaut le coup que j'enregistre toute les IP Zoom dans la section DSCP pour prioriser ? 

Merci à toi 
Mathieu

kobbybest, 2021-05-14:
Perso moi, c'est j'ai les mêmes problèmes que toi sur openmtcprouter que sur ta connexion, voix métallique, coupure que ce soit teams ou discord et c'est du au ping qui n'arrête pas de monter aléatoirement jusqu'à 1s, j'ai ouvert un ticket à ce sujet et ysurac m'a répondu que ce n'était pas grave d'avoir un ping qui bougeait sur une connexion internet, étant gamer j'ai pas compris, j'ai laissé tomber et je coupe l'agrégation pour les jeux, team et discord et je l'a remet qu'en j'ai besoin de télecharger ou regarder du contenu sur youtube ou netflix

Ysurac, 2021-05-15:
@MathieuG9263 tu peux mettre des domaines dans DSCP, il y a un onglet pour ça

@kobbybest Oui encore une fois le ping ne sert à rien, c'est juste des paquets ICMP qui ne transportent pas une information vitale.
Par contre les jeux, teams, discord,... peuvent utiliser UDP et la il y a des moyens d'améliorer les performances pour l'UDP comme utiliser V2Ray en proxy ou alors utiliser un autre VPN pour le transport.

kobbybest, 2021-05-19:
@Ysurac J'ai essayé tous les VPN disponibles ainsi que V2RAY, et j'ai le même soucis, le seul moyen d'avoir un ping stable et donc pas d'impact dans les jeux ou teams discord, c'est de désactiver le vpn mais ca implique d'autres problèmes de désactiver le vpn

closed 2021-08-23 by github-actions[bot]