Ysurac/openmptcprouter #1825: TP-Link TL-WR841N/ND v9

Need image

ssdeol84, 2021-05-13:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

I need snapshot image or firmware to connect my openwrt router with mptcp vps

Need your help to connect my mptcp router with mptcp vps.
# 

**Describe why you are asking for this support?**

**Additional context**

[image][]()[image][image]

Ysurac, 2021-05-13:
I don't understand what is the issue.

ssdeol84, 2021-05-13:
Issue is that I'm trying to connect my openwrt router with ss-tunnel 

On router it has only option to put remote server IP.
Method
Port 
And password

I've put all this information on router but still I'm not able to get connected with my remote server

Can you please help to fix this?

[image]

Ysurac, 2021-05-13:
I don't know what you want to do. You seems to be using another firmware than OpenMPTCProuter, this can't work with OpenMPTCProuter.

ssdeol84, 2021-05-14:
Yes it's different firmware..can you provide me firmware for TP-Link TL -WR841N/ND v9 router ?

Ysurac, 2021-05-14:
No. This device have a slow CPU, a too small flash and small memory. It can't support OpenMPTCProuter.

ssdeol84, 2021-05-14:
This have Qualcomm Atheros QCA9533 ver 2 rev 0 architecture

Memory size is 121.10mb

Plz check if we can place your firmware on this ?

Bumpalert, 2021-05-14:
> 
> 
> This have Qualcomm Atheros QCA9533 ver 2 rev 0 architecture
> 
> Memory size is 121.10mb
> 
> Plz check if we can place your firmware on this ?
which firmware you are using on WR841n ? it doesn't look to be original https://emulator.tp-link.com/Emulator_TL-WR841NV14(EU)/index.htm

ssdeol84, 2021-05-14:
[image]

ssdeol84, 2021-05-14:
This have 4 sim module 1 wan and 4 lan ports

ssdeol84, 2021-05-16:
root@AnyWiFi:~# cat /etc/os-release lsb_release
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0+11740-9b02d32"
OPENWRT_BOARD="ar71xx/tiny"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r0+11740-9b02d32"

This is the information of my router..

can i install any snapshot image ???

Hostname	AnyWiFi
Model	TP-Link TL-WR841N/ND v9
Architecture	Qualcomm Atheros QCA9533 ver 2 rev 0

Ysurac, 2021-05-16:
You need to compile an image yourself, I will not support this device as it's too slow (and really too many revision of the device).

ssdeol84, 2021-05-16:
Target mips not found ! You have to configure and compile your kernel manually.

Can you please let me know how i can compile by my own? I'm weak in linux and programing.

ssdeol84, 2021-05-16:
The seller from which I've bought this device already make a firmware which support mptcp, OpenVPN and shadowsocks. I'm trying to get connected with the shadowsocks server with its credentials its starts showing internet and also shows me VPS ip. but not getting mptcp speed. all 4 sim are not combining the internet. and that seller is speaking Chinese which i don't understand. that's why he unable to help me to setup.

now i'm really feeling depressed after spending amount on this.

Ysurac, 2021-05-16:
To compile it: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

ssdeol84, 2021-05-16:
I'm trying to build but getting this error:

Target mips not found ! You have to configure and compile your kernel manually.

Ysurac, 2021-05-16:
As indicated in the doc for not supported platoform, you have to do a make menuconfig.

ssdeol84, 2021-05-16:
But in menu Qualcomm Atheros QCA9533 is not available

Ysurac, 2021-05-16:
As you can see in the os-release, arch is mips_24kc, board is ar71xx/tiny and profile seems to be generic.

ssdeol84, 2021-05-16:
Thanks for clarifying...I'll give try ...hope it will be done...if I face any problem I'll let you know..

Thanks

ssdeol84, 2021-05-17:
Getting error in compiling:

root@mptcp:/home/mptcp/openmptcprouter/mips_24kc/source# make
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-tcp-bbr2', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
make[2]: Entering directory '/home/mptcp/openmptcprouter/mips_24kc/source'
make[3]: Entering directory '/home/mptcp/openmptcprouter/mips_24kc/source/scripts/config'
set -e; mkdir -p ./; trap "rm -f ./.mconf-cfg.tmp" EXIT; { /bin/sh mconf-cfg.sh; } > ./.mconf-cfg.tmp; if [ ! -r mconf-cfg ] || ! cmp -s mconf-cfg ./.mconf-cfg.tmp; then true '  UPD     mconf-cfg'; mv -f ./.mconf-cfg.tmp mconf-cfg; fi
cc -O2 -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600   -c -o mconf.o mconf.c
cc -O2 -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc -O2 -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc -O2 -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -O2 -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc -O2 -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600   -c -o lxdialog/util.o lxdialog/util.c
cc -O2 -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc -o mconf mconf.o lxdialog/checklist.o lxdialog/inputbox.o lxdialog/menubox.o lxdialog/textbox.o lxdialog/util.o lxdialog/yesno.o confdata.o expr.o lexer.lex.o parser.tab.o preprocess.o symbol.o util.o -lncursesw -ltinfo
make[3]: Leaving directory '/home/mptcp/openmptcprouter/mips_24kc/source/scripts/config'
make[3]: Entering directory '/home/mptcp/openmptcprouter/mips_24kc/source'
make[3]: Leaving directory '/home/mptcp/openmptcprouter/mips_24kc/source'
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-tcp-bbr2', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
No change to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

make[2]: Leaving directory '/home/mptcp/openmptcprouter/mips_24kc/source'
time: target/linux/prereq#1.12#0.11#1.21
 make[1] world
 make[2] tools/compile
 make[3] -C tools/flock clean-build
 make[3] -C tools/flock compile
 make[3] -C tools/xz clean-build
 make[3] -C tools/xz compile
 make[3] -C tools/sed clean-build
 make[3] -C tools/sed compile
 make[3] -C tools/patch clean-build
 make[3] -C tools/patch compile
 make[3] -C tools/tar clean-build
 make[3] -C tools/tar compile
    ERROR: tools/tar failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/mptcp/openmptcprouter/mips_24kc/source/include/toplevel.mk:230: world] Error 1

Ysurac, 2021-05-17:
```apt-get install tar```

Adorfer, 2021-05-18:
> This have Qualcomm Atheros QCA9533 ver 2 rev 0 architecture
> 
> Memory size is 121.10mb

Sorry, you are confusing Megabit and MegaByte!

If you do not believe Ysurac or me: 
https://openwrt.org/toh/tp-link/tl-wr841nd

to implement OMR on that platform is a COMPLETE waste of time. 

(and yes, i soldered bigger RAM and more than 4MB SPI on excatly this model and put a patched bootloader. it's the CPU which is FAR away from any use for an OMR.)

[image]

[image]

ssdeol84, 2021-05-19:
> `apt-get install tar`

root@mptcp:/home/mptcp/openmptcprouter# apt-get install tar
Reading package lists... Done
Building dependency tree
Reading state information... Done
tar is already the newest version (1.30+dfsg-6).
0 upgraded, 0 newly installed, 0 to remove and 28 not upgraded.

ssdeol84, 2021-05-19:
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/mptcp/openmptcprouter/mips_24kc/source/include/toplevel.mk:230: world] Error 1

ssdeol84, 2021-05-21:
Compiling is not happening

Adorfer, 2021-05-21:
what shall i say? i could teach you through 

https://openwrt.org/docs/guide-developer/build-system/use-buildsystem

But in the end it's shure and certain that you will run into a final error like: 

**"WARNING: Image file /.../...../squashfs-sysupgrade.bin is too big"**

closed 2021-08-24 by github-actions[bot]