Ysurac/openmptcprouter #1828: WAN 3 + WAN 4 will not connect

GTiDon, 2021-05-13:
Hello

I have setup a X86 Device that has 6 physical ports.

The original WAN 1 and WAN 2 ports connect to the server.

When I add more WAN ports eth2 + eth3 etc they will not connect to the server?

What can cause this? I have tried static and DHCP

Ysurac, 2021-05-13:
OpenMPTCProuter current release support 8 WAN, next release support 16 WAN on x86_64, so no problem for 6 WAN ports.
If you have multiples physical ports, check that MacVLAN is not enabled.

GTiDon, 2021-05-13:
I still want to use static IP addresses on the extra ports because they connect to the same POE switch that powers the modems.

Can it work like this?

GTiDon, 2021-05-13:
Thankyou Yannick you are amazing. You pointed me to the problem so quickly it now works

I still need help getting it running on my unsupported OPENWRT Router.. I want that as my primary device

[image]

Ysurac, 2021-05-13:
I can't help on unsupported devices. I only support devices that I own.
But you can compile it yourself using https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform and with some luck this may work.

closed 2021-08-18 by github-actions[bot]