Ysurac/openmptcprouter #1829: VPS LAN to OMR LAN Access

hrmtc, 2021-05-13:
## Expected Behavior
Being able to access the OMR LAN from the VPS LAN

## Current Behavior

I can't reach services on the OMR LAN side.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.57.3 r0+15225-bfc433efd4 / LuCI Master git-20.339.75073-e54708a
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

## Comments
I would like to reach devices on the OMR LAN side from the VPS LAN.
I guess that I should configure some routes and shorewall rules but I don't know where to start.
Could someone help me to understand this, please?

Thanks!

Ysurac, 2021-05-13:
On v0.58 you can access LAN from VPS by default, at least if you allow that from router firewall.

hrmtc, 2021-05-13:
Thanks. 
v0.58 will be released soon?

Ysurac, 2021-05-13:
I hope. For now there is 0.58rc2.

closed 2021-08-17 by github-actions[bot]