Ysurac/openmptcprouter #1831: WAN on status page copying the whois, public ip, and ping of master. Only gets the internal IP correct

darthclide, 2021-05-14:
Raspberry Pi 4
OMR router version: 57.3
VPS server version: 0.1024 5.4.86-mptcp

Description:
Upon reboot, a WAN on the status page reports the data for my master. The reason I say master, is without me knowing, the master switched to my Verizon. I then looked at the WAN giving problems, and it now was copying my Verizon information.

Steps to reproduce:
Simply restart the device.

uci show network
https://pastebin.com/JDdz0Xhi

ifstatus wan1
https://pastebin.com/NZLPaggA

ifstatus wan2
https://pastebin.com/U52VVZnC

ifstatus wan3
https://pastebin.com/egfe4vZz

ifstatus wan4
https://pastebin.com/Xn4ZiTgN

uci show openmptcprouter
https://pastebin.com/qjL0fGJS
(as you can see the ip address reported is the same between my DSL and wan2)

ip a
https://pastebin.com/JquxSJcC

ip r
default via 10.255.255.1 dev tun0
default via 192.168.1.1 dev eth1 metric 3
default via 192.168.42.129 dev usb1 metric 4
default via 192.168.124.1 dev eth2 metric 8
default via 192.168.42.129 dev usb0 metric 10
default via 10.255.255.1 dev tun0 metric 1200
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
10.255.255.2 dev tun0 scope link metric 1200
**.**.**.60 via 192.168.1.1 dev eth1 metric 1
**.**.**.60 via 192.168.1.1 dev eth1 metric 3
**.**.**.60 via 192.168.42.129 dev usb1 metric 4
**.**.**.60 via 192.168.124.1 dev eth2 metric 8
**.**.**.60 via 192.168.42.129 dev usb0 metric 10
127.0.0.0/8 dev lo proto static scope link metric 1
192.168.1.0/24 dev eth1 scope link metric 3
192.168.42.0/24 dev usb1 scope link metric 4
192.168.42.0/24 dev usb0 scope link metric 10
192.168.124.0/24 dev eth2 scope link metric 8

Interesting why it has my eth1 listed twice?

darthclide, 2021-05-17:
And just now after using QOS to force more data over that WAN, the connection got very shaky even though the network/mptcp/bandwidth showed a very stable line for all my connections. And there were 0 errors in my system log.

I think this has moved beyond just a visual error, and is now messing with my connection stability :/

Ysurac, 2021-05-17:
You can try 0.58rc2, maybe it will solve the issue or at least give more debug info.
QoS is not available by default on OpenMPTCProuter, SQM is used instead.

darthclide, 2021-05-17:
I just solved it... I will leave the other posts up just in case someone else pulls a dumb move like I did. I use some cheap Android phones (got lucky that they cover all bands in USA) and I sometimes am forced to use WiFi to download an update to them. I forgot to turn the WiFi off, so that is why it was putting my master's information on that WAN. Can you maybe add a huge disclaimer somewhere about using USB tether with phones and forgetting to turn off your WiFi when you do an update?

closed 2021-08-20 by github-actions[bot]