Ysurac/openmptcprouter #1832: Only one of two wan interfaces used (most of the time, but sometimes works)

linwong, 2021-05-15:
wan1 10/2 4G LTE
wwan 15/1 ADSL 

I started using it OMR about a month ago, and link aggregation sometimes worked. Sometimes it didn't. Sometimes it would work for a few hours, not work at all for a few hours, and then start working again by itself.  Rebooting the router normally does not fix the aggregation immediately. I just have to wait and it'll start working by itself. Most of the time, however, it only uses the interface that is set as the master.

The last few days link aggregation hasn't been working much at all. I've tried using .57.3, .58rc1 and .58rc2. All have worked properly at some point. I've clean flashed .57.3 and .58rc2 and there isn't much difference in stability.

omr-test-speed shows both links are working. I can see in Network -> MPTCP -> Bandwidth, when they are working. I can ping from both interfaces. The system log does not give any information at the time of when it goes down, and comes back up, although I may have missed it.

I'm not sure what else I should be looking for.

## Expected Behavior
15/1 Mbs ADSL line (connected as wifi client) + 10/2 Mbs 4G (USB ethernet) ~ 23/2.5 Mbs total
This works sometimes. I've also used a different USB ethernet adapter with the ADSL line, but the problem persists. (I'm not using the USB ethernet adapter because it is garbage; only getting 3-5Mbs on it)

## Current Behavior
Most of the time it's like this:  Only uses the interface set as master. Either 15/1 if ADSL is set as master, or 10/2 if 4G set as master.

## Specifications

  - OpenMPTCProuter version: .57.3 (and .58rc2)
  - OpenMPTCProuter VPS version: .1024 (and .1025-test)
  - OpenMPTCProuter platform: RPI4

[image]

Ysurac, 2021-05-15:
Wifi never give good result, it's far better to use macvlan and use a switch to connect all.
When you have issue, you should check if you have something strange in State->System log again.
You can, in Network->MPTCP, change scheduler.

linwong, 2021-05-17:
Yes, I understand that wifi is not ideal. To test, I removed the wifi client, and only used the onboard ethernet and USB3 ethernet for the WAN ports. It was working for a few minutes, and then it stopped. You can see the screenshot where it was working, and then it stopped and was only using the WAN set to master.

[image]

I've tried changing the MPTCP scheduler and that didn't worked. I tried each one. 

I checked the system log, and the only error messages of note were these, although I believe these have to do with the wifi. (I only started getting these messages with 0.58rc2 and after updating to the latest rpi4 firmware)

```
Sun May 16 23:56:52 2021 kern.err kernel: [  103.330613] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun May 16 23:56:55 2021 kern.err kernel: [  106.315170] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun May 16 23:56:58 2021 kern.err kernel: [  109.312986] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun May 16 23:56:58 2021 kern.err kernel: [  109.321970] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
Sun May 16 23:57:04 2021 kern.err kernel: [  115.162855] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun May 16 23:57:07 2021 kern.err kernel: [  118.329077] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun May 16 23:57:07 2021 kern.err kernel: [  118.338077] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun May 16 23:57:07 2021 kern.err kernel: [  118.347044] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Sun May 16 23:57:10 2021 kern.err kernel: [  121.336794] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun May 16 23:57:13 2021 kern.err kernel: [  124.337233] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun May 16 23:57:13 2021 kern.err kernel: [  124.346233] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Sun May 16 23:57:19 2021 kern.err kernel: [  130.341992] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun May 16 23:57:22 2021 kern.err kernel: [  133.359436] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sun May 16 23:57:22 2021 kern.err kernel: [  133.368437] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Sun May 16 23:57:25 2021 kern.err kernel: [  136.331965] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun May 16 23:57:28 2021 kern.err kernel: [  139.346124] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
```

There are a lot of other settings I don't quite understand. I just don't know where to look. For the most part, I've tried to stick with the defaults. I just don't know why it would be working, and then suddenly stop working. When it does work, it's been great.

Ysurac, 2021-05-17:
Error in log are related to wifi, it's often due to an unsupported configuration. You need to try to find parameters that the wifi driver allow, like a real country and not using world or driver default,...

linwong, 2021-07-14:
I took a break from this for a while. I replaced the wifi adapter with a USB ethernet adapter. However, I was still getting the same results where only one of the interfaces works.

A few weeks ago, I re-imaged the router and my server and tried just running MLVPN. After working and tweaking it, I was able to get link aggregation working over MLVPN. Many tutorials and docs so to use say to use the reorder_buffer_size to enable the reordering system. With this enabled, I was not able to get the link aggregation working. One of the links, particularly my ADSL, would have high pings and drop the link. It wasn't until I disabled the reordering system that link aggregation would work fairly stable.

I only mention this cause I don't know if there is something equivalent to this with the default settings on OMR. I've tried again with the latest versions, but I'm still getting the problem where link aggregation does not work. While MLVPN works, I'm only getting about 65-70% of the combined speed. Whereas when OMR was working, I would get close to 95% of the combined speed.

I don't know if this gives any more insight on this.

Ysurac, 2021-07-14:
Did you try latest image ?

linwong, 2021-07-14:
Yes, I tried the latest image.

Ysurac, 2021-07-14:
Check in Network->MPTCP, "MPTCP Fullmesh" tab that all IPs are here.
Use also ```omr-test-speed``` via SSH to get full aggregated speed, and ```omr-test-speed eth1``` (where eth1 is real interface, you can use ```ip a``` to get real interfaces), use this on all wan at least 2 minutes (Ctrl+C to stop).

linwong, 2021-07-14:
Yes. All IPs are there.

```
Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 5
3, 4, 0, 192.168.1.11, 12
4, 5, 0, 192.168.4.142, 11
IPv6, next v6-index: 0
```
Well, link aggregation is working at the moment. I'll update this thread when it stops working.

```
root@OpenMPTCProuter:~# omr-test-speed

Best server is http://nyc.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 9536M    2  260M    0     0  2404k      0  1:07:40  0:01:51  1:05:49 2435k^C

root@OpenMPTCProuter:~# omr-test-speed eth1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 9536M    1  122M    0     0  1051k      0  2:34:47  0:01:59  2:32:48 1023k^C

root@OpenMPTCProuter:~# omr-test-speed eth1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 9536M    2  200M    0     0  1587k      0  1:42:32  0:02:09  1:40:23 1614k^C
```

linwong, 2021-07-16:
The link aggregation was working for a good two days. That was probably the longest I have had it working at any point. When I woke up this morning, only one link (the master) was being used. I was testing throughout the day, but the second link never returned. I could switch which interface is the master, and then only that would be used.

```
root@OpenMPTCProuter:~# omr-test-speed eth1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 9536M    1  133M    0     0  1057k      0  2:33:58  0:02:08  2:31:50 1081k

root@OpenMPTCProuter:~# omr-test-speed eth1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  123M    0  123M    0     0   963k      0 --:--:--  0:02:11 --:--:--  929k

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 6106k    0     0  90417      0 30:43:18  0:01:09 30:42:09 13333
```

You can see with the last test, it was extremely slow. That only happened with eth1 as the master. I switched to eth2 as the master, and it was still using only one interface but the full speed of it.

I tried restarted the router. I restarted the modems for both connections. Nothing worked. I went to restart the server and noticed I had a few updates. So I applied the updates and after rebooting, the server went into a boot loop, kernel panicking less than 2 seconds after booting.

I brought up a new instance of the server with a new IP and reinstalled everything. I update the router settings with the new IP and key, and of course link aggregation starts working again.

It's possible it's the VPS, but I've tried using 3 different providers (Digital Ocean, Linode and now Ramnode) and they all eventually stop working properly.

I'm perplexed by all of this. I'm not sure what is going on, or what I can do to have link aggregation working stably. Is it common for others to have it stop working over time? Or need to reboot the router or server?

Ysurac, 2021-07-17:
Use ```omr-test-speed``` on all wan, not only eth1.
I tested Digital Ocean myself, so I'm sure that it's working.
You can try to enable Multipath debug in Network->MPTCP, then check in State->System log if there is errors.

linwong, 2021-08-13:
I think I just copied and pasted incorrectly on the eth1/eth2 command, but the speeds I believe were correct. Regardless, it's been running flawlessly for the last 3 weeks. I can bring the links up and down, and the link aggregation will return like it's supposed to.

On my VPS that broke, I was running Ubuntu 20.04, which may have been the reason why it broke on the update. I re-imaged the server with Debian 10, and it has been working flawlessly since. I had tried Debian 9 and 10 previously without success, so I'm not sure why it is working this time around. Perhaps it's just the updated software? I literally flashed and reflashed my Pi and VPS almost a dozen times before it finally worked. Maybe my Pi just needed some rest. 

I'm going to close this ticket since it's working. For those reading this ticket looking for help, I guess my only suggestion is too keep trying until it works. And use Debian. *shrug*

Thank you for the help.

closed 2021-08-13 by linwong