Ysurac/openmptcprouter #1833: DHCP loop in log

ciasco92, 2021-05-16:
## Expected Behavior

I should see all the logs

## Current Behavior

in the logs I always find these two alerts in loop
`daemon.info dnsmasq-dhcp[20061]: DHCPOFFER(eth0) 192.168.100.5 00:00:00:00:00:00`
`daemon.info dnsmasq-dhcp[20061]: DHCPDISCOVER(eth0) 00:00:00:00:00:00 `

i think this causes problems, often devices fail to get ip address from dhcp
I have a Huawei b818 modem and a mikrotik Chateau LTE 12, I have DHCP disabled in both, but I think the mikrotik gives problems because when it is off the messages in the logs disappear

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: 

v0.57.3 r0+15225-bfc433efd4
Vps 0.1025
RPI4

Ysurac, 2021-05-17:
If the mac address is really "00:00:00:00:00:00", then there is an issue. Check mikrotik mac and try to fix it.

ciasco92, 2021-05-17:
thanks for the reply, no the mac address is totally different, the problem is that the log is full of these two warnings, which appear every second in loop

Ysurac, 2021-05-17:
You can put the client with the mac in error on fixed IP as it seems to request an IP and seems to refuse the proposed IP.

Adorfer, 2021-05-18:
i had this on IOT device some time ago: during firmware update it lost it's MAC (was probably stored in a partition of the flash)and the presented 00:00:00:00:00:00 is clearly invalid and can not be served an IP.

ciasco92, 2021-05-18:
thanks for the answer, I have set to -> dhcp and DNS -> static leases the Mac 00:00 ... on the ip 192.168.100.2 but these logs still appear.  here is the result of the command

[image]

I think the problem is my mikrotik Chateau lte12 because when it is off the logs stop, the mikrotik has routeos in which I only disabled the DHCP server, but I left the bridge part active, only one ethernet cable comes out of the mikrotik that goes into a switch

Adorfer, 2021-05-18:
perhaps i was not clear in my last message: an "all-zero-MAC" is invalid. it is not legal. it can't work. 
You are having a device with either defective hardware or a defective network-setup in your LAN. 
There is nothing the OMR can do for you.

ciasco92, 2021-05-18:
thanks, i reset my mikrotik and i haven't loaded my backup, now i don't have this problem anymore

closed 2021-05-18 by ciasco92