Ysurac/openmptcprouter #1834: dedicated master & backup is a trouble in all wireless connection

Bumpalert, 2021-05-17:
**Is your feature request related to a problem? Please describe.**

When used 3-4 LTE routers to get a stable connection, even master can be unstable. dedicated master is impractical,when you don't have a cabled internet. The backup can be enabled,but those should be used as MPTCP when not acting as master.

**Describe the solution you'd like**

Option to have backup/mptcp enabled

**Describe alternatives you've considered**

adding a  2g connection which has range and slow speed 2G internet as a master to keep the connection alive.

**Additional context**

Bumpalert, 2021-05-17:
use dhcp always thinking that if static, every reboot pi will assign different interface id like usb0 usb1 interchangeable and master will go for a toss. Backup of will not be active until master is down and cannot be used for sharing the traffic. When all devices are lte depending on a single device is a problem. @Ysurac any solution to use other devices to share traffic as well as backup for master

Ysurac, 2021-05-17:
Check that in System->OpenMPTCProuter, advanced setting tab, "Disable interfaces auto rename" is not checked.

Bumpalert, 2021-05-17:
Thanks.. I will do that.. Is there anyway to use backup  as mptcp shared connection instead of waiting for a failure

Ysurac, 2021-05-17:
I don't understand what you mean. Backup is used only when others connections can't be used, if you want to use it all the time only set it as enabled.

Bumpalert, 2021-05-17:
i am using 2 sim of one LTE provider & 2 sim of another LTE provider and total 4 connection. What I understand by master, if master dies, our vpn tunnel will die, no matter all other 3 wans are up even after connection is established with vps already(Correct me if I am wrong here). So If I am keeping one master in one of the LTE vendor  and backup on another from another LTE vendor,I can add only 3 wans at a time & one will be unused. If backup or master can be used as mptcp enabled when not acting as master, we can utilize all 4 wans aggregated at any point of time and no downtime even if master got a problem as backup to master also  alive and doing the traffic. If we enable backup only after detecting master is down, there will be a break in the connection until backup to master is up and running.

Ysurac, 2021-05-17:
VPN tunnel should never die, if this happen you can try another VPN like Glorytun UDP.
You don't need to put any wan as backup for security, backup is more if you want a WAN to be used only when all others WAN are down, as a security but not used in normal use.
So you can utilize all WANs aggregated.

Bumpalert, 2021-05-18:
Are you sure tht vpn tunnel should not die even if master dies after setting up the tunnel?

Bumpalert, 2021-05-18:
I mean other connections exist to keep the tunnel but master disconnected after creating tunnel

gnosoz, 2021-05-20:
I think what you are trying to do can be slved with the following setup which automatically switches master depending on which connection it feels is best in terms of stability / RTT.
From experience best RTT is also the more stable of the connection pool...

[image]

Bumpalert, 2021-05-23:
Dynamic change helped me. Now it automatically switches over master and keep connection

closed 2021-06-02 by Bumpalert