Ysurac/openmptcprouter #1836: "no data when rebooted" on a configured router

Bumpalert, 2021-05-18:
after setting up the router, i rebooted yesterday.Now all setting gone and no data.

[image]

Interfaces are up and running,but no connection to the server.

[image]

## Expected Behavior

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Bumpalert, 2021-05-18:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].dnssec='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs7'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs4'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs4'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs4'
dscp.@classify[3].dest_port='65500'
dscp.@classify[3].comment='OMR API'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='tcp'
[... 1691 lines cut ...]
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.network='wifi'

Ysurac, 2021-05-18:
Check log in State->System log

Bumpalert, 2021-05-18:
here goes system log

Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty48, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty48
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty48, subsystem=tty
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty38, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty38
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty38, subsystem=tty
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=ip6tnl0, subsystem=net, sysfspath=/sys/devices/virtual/net/ip6tnl0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ip6tnl0, subsystem=net
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=sit0, subsystem=net, sysfspath=/sys/devices/virtual/net/sit0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=sit0, subsystem=net
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=gre0, subsystem=net, sysfspath=/sys/devices/virtual/net/gre0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=gre0, subsystem=net
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=gretap0, subsystem=net, sysfspath=/sys/devices/virtual/net/gretap0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=gretap0, subsystem=net
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=erspan0, subsystem=net, sysfspath=/sys/devices/virtual/net/erspan0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=erspan0, subsystem=net
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=ip6gre0, subsystem=net, sysfspath=/sys/devices/virtual/net/ip6gre0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ip6gre0, subsystem=net
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=bond0, subsystem=net, sysfspath=/sys/devices/virtual/net/bond0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=bond0, subsystem=net
Mon May 17 19:39:10 2021 daemon.notice netifd: usb0_jiofi3 (5666): udhcpc: sending select for 192.168.225.145
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=teql0, subsystem=net, sysfspath=/sys/devices/virtual/net/teql0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=teql0, subsystem=net
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=usb0, subsystem=net, sysfspath=/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.0/net/usb0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=usb0, subsystem=net
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: no need to wait for modem at sysfs path /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2
Mon May 17 19:39:10 2021 daemon.notice netifd: usb0_jiofi3 (5666): udhcpc: lease of 192.168.225.145 obtained, lease time 43200
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=wlan0, subsystem=net, sysfspath=/sys/devices/platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: event reported: action=add, name=wlan0, subsystem=net
Mon May 17 19:39:10 2021 daemon.debug dnsmasq[6103]: listening on usb0(#11): 192.168.225.145 port 53
Mon May 17 19:39:10 2021 daemon.notice netifd: Interface 'usb0_jiofi3' is now up
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon May 17 19:39:10 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=eth1, subsystem=net, sysfspath=/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1 [... cut ...]
Mon May 17 20:11:00 2021 user.notice post-tracking-post-tracking: huewei_eth2 (eth2) switched up
Mon May 17 20:11:01 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon May 17 20:11:03 2021 user.notice omr-bypass: Reload dnsmasq...
Mon May 17 20:11:03 2021 daemon.info dnsmasq[15011]: read /etc/hosts - 4 addresses
Mon May 17 20:11:03 2021 daemon.info dnsmasq[15011]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Mon May 17 20:11:03 2021 daemon.info dnsmasq-dhcp[15011]: read /etc/ethers - 0 addresses
Mon May 17 20:11:03 2021 user.notice omr-bypass: OMR-ByPass is running

Ysurac, 2021-05-18:
I see some USB device, what is the result if some are unplugged ?

Bumpalert, 2021-05-18:
Those 3 usb devices are my wan providers. I set it yesterday and turned off. Today powered on and it says no data. Even if i try to run wizard save button of wizard is not saving.

Ysurac, 2021-05-18:
So you have no data even if you unplug them ?

Bumpalert, 2021-05-18:
I didn't try unplugging as i turned off and stopped the experiment for today. But do you have any clue why the no data comes? Usb devices are functional and displaying dhcp leased ip addresses in the interface page.

Bumpalert, 2021-05-23:
This issue was fixed.  Added a powered usb hub. Now reboots has no issue. Some usb chargers has power issues & some cable has issue. But identified a configuration of charger and cables which comes and works stable.