Ysurac/openmptcprouter #1837: WAN connection down after heavy network traffic

mathewinwood, 2021-05-18:
I've been having problems with one of two wan connections going down (`Gateway down` in the MPTCP status page).

Any ideas what might be causing this and how I can diagnose/fix it?

## Expected Behavior
WAN connections should stay up, even after heavy network traffic.

## Current Behavior
Usually after a period of heavy network traffic, my primary WAN connection does down (`Gateway down`). The secondary WAN connection is reliable.

Other symptoms:
- rebooting openmptcprouter always fixes is
- Restarting the WAN1 interface also fixes it (`Network` -> `Interfaces` -> `WAN1` ->  `Restart` button)
- When the gateway is down, running `omr-test-speed wan1` fails (no ping responses from any server)
- When the gateway is working, I see a different error from `omr-speed-test wan1`:

  ```
  root@OpenMPTCProuter:~# omr-test-speed wan1
  Select best test server...
  host: speedtest.frankfurt.linode.com - ping: 21
  host: speedtest.tokyo2.linode.com - ping: 247
  host: speedtest.singapore.linode.com - ping: 167
  host: speedtest.newark.linode.com - ping: 80
  host: speedtest.atlanta.linode.com - ping: 94
  host: mia.download.datapacket.com - ping: 105
  ...
  host: nyc.download.datapacket.com - ping: 79
  Best server is http://ams.download.datapacket.com/10000mb.bin, running test:
  ipset v7.6: Syntax error: cannot parse 185.59.222.18185.59.222.21195.181.172.2: resolving to IPv4 address failed
    % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                   Dload  Upload   Total   Spent    Left  Speed
    0     0    0     0    0     0      0      0 --:--:--  0:00:46 --:--:--     0
  curl: (28) Failed to connect to ams.download.datapacket.com port 80: Operation timed out
  
  ipset v7.6: Syntax error: cannot parse 185.59.222.18185.59.222.21195.181.172.2: resolving to IPv4 address failed 
  ```
  Not sure if this is related or not?

Looking in the system log, I can see this from (I think) when `wan1` went down:

```
Tue May 18 13:53:33 2021 user.notice post-tracking-post-tracking: wan1 (wan1) switched off because gateway down and ping from 192.168.8.2 error (114.114.114.114,1.1.1.1,4.2.2.1)
Tue May 18 13:53:33 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev wan1
Tue May 18 13:54:39 2021 daemon.err /usr/bin/ss-redir[14942]: remote recv: Operation timed out
Tue May 18 13:54:48 2021 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because ping from 192.168.129.3 error (9.9.9.9,1.0.0.1,114.114.115.115)
Tue May 18 13:54:48 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.129.1 dev wan2
Tue May 18 13:54:51 2021 daemon.err /usr/bin/ss-redir[14944]: remote recv: Operation timed out
Tue May 18 13:54:54 2021 daemon.err /usr/bin/ss-redir[14942]: remote recv: Operation timed out
Tue May 18 13:54:57 2021 daemon.err /usr/bin/ss-redir[14942]: remote recv: Operation timed out
Tue May 18 13:54:57 2021 daemon.err /usr/bin/ss-redir[14942]: remote recv: Operation timed out
Tue May 18 13:54:58 2021 daemon.err /usr/bin/ss-redir[14942]: remote recv: Operation timed out
Tue May 18 13:55:03 2021 user.notice post-tracking-post-tracking: wan2 (wan2) switched up
```

The `remote recv: Operation timed out` then continues indefinitely until a reboot or restart of the interface.

## Specifications

  - OpenMPTCProuter version: `openmptcprouter v0.57.3 r0+15225-bfc433efd4`
  - OpenMPTCProuter VPS version: `5.4.86-mptcp 0.1024`
  - OpenMPTCProuter platform: `Raspberry Pi 4 Model B Rev 1.2`
 
Config:
- WAN1 is fixed line broadband, gateway is a FritzBox 7490
- WAN2 is a 4G modem, Huawei B535

Ysurac, 2021-05-18:
In the wizard, enable SQM.

mathewinwood, 2021-05-18:
Thanks! You mean for the fixed line broadband? (I already have it enabled for the 4G connection).

I'll give it a go and see if it fixes the problem.

Bumpalert, 2021-05-18:
Sqm is not recommended for 4G right @Ysurac

Ysurac, 2021-05-18:
True. SQM should be enabled on the fixed line in priority with download and upload set (it's more difficult to know speed of a 4G link, often not stable).

mathewinwood, 2021-05-31:
Thanks.

FYI, I tracked this down to a faulty gateway after I noticed the same symptoms on a computer using that gateway directly. Replacing that hardware fixed this particular issue.

closed 2021-05-31 by mathewinwood