Ysurac/openmptcprouter #1840: qualcomm qca9531 soc support

Bumpalert, 2021-05-19:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

https://www.aliexpress.com/item/1005001685713486.html

The below router claims openmptcp router, but no builds found  officially for qualcomm SOC. 

**Describe why you are asking for this support?**

The router was found with reasonable pricing, but as it looks like the custom build bug fixing and support is expected as a problem. Banana PI R64 doesn't come with multiple sim card option which is why I am not considering it.

**Additional context**

Lookig for any other router hardware with multiple sim options but support original openmptcp github builds.

Kalimeiro, 2021-05-20:
if you want it for your qualcomm qca9531, buy it an send them to ycarus.

Bumpalert, 2021-05-20:
@Ysurac If you can add qualcomm qca9531 to your builds, I will send you a device. Please dm me your shipping address on @johns_s on twitter.

Bumpalert, 2021-05-20:
@Ysurac  Got your address from webpage. 

Address:
Yannick Chabanois
3892 route de Romanèche
01270 Coligny
FRANCE

Ysurac, 2021-05-20:
There is several problems with this product:
* slow CPU
* small RAM
* small flash
* 100Mbps interfaces

So if it's not possible to boot from SDcard, OpenMPTCProuter will have lot of features removed (no proxy choices, one VPN,...).
I think encyption must be disabled with this CPU and speed will never be really great.

Bumpalert, 2021-05-20:
100mbps maximum wan is still good for a work from home scenario. I will be happy if it can handle maximum 10mbps upload and download after aggregation. 4G speed normally won't exceed 20mbps in best case in India( Worst cases go down <1mbps). The above product has 4 LTE modules. Which encryption VPN or shadowsocks or some other encryption we need to avoid.

Ysurac, 2021-05-20:
I'm not sure it can reach 100Mbps, 10Mbps seems possible I think. Encryption need to be disabled for proxy (shadowsocks, v2ray require too much CPU/RAM), encryption can't be disabled on most VPN (only OpenVPN can have encryption disabled I think).

Bumpalert, 2021-05-20:
I know people buying the below item in my area for 5mbps throughput( In fact I started on MPTCP looking for an economical alternative). Stable internet even in comparatively low band width also great support. The compromise on encryption also understandable. People who are trying to get bandwidth to get thing done somehow are not worried much about encryption. Even though 5G is coming in next 2 years, there is no magic going to happen in wireless internet as it is not economical to put tower everywhere.

The below device cost about double of listed price, when it comes in end user hands, after clearing all customs & formalities.

https://www.aliexpress.com/item/1005001407248089.html

Bumpalert, 2021-05-20:
same qca9531 requested by another user in this thread https://github.com/Ysurac/openmptcprouter/issues/1822

closed 2021-08-24 by github-actions[bot]