Ysurac/openmptcprouter #1844: glorytun-tcp server port can not connect sometimes

johnlipss, 2021-05-20:
glorytun-tcp listened port 65001 in openmptcprouter-vps side, but every once in a while after Glorytun VPN connected it becomes unreachable, causing Glorytun VPN restart.
## Expected Behavior
glorytun-tcp port 65001 should always connect able, Glorytun VPN should not restart every once in a while.

## Current Behavior
Glorytun VPN restart every once in a while, after I check the network status, found out that the port 65001 listened by glorytun-tcp become can not connect successful after minutes of Glorytun VPN connected(1-5min), even netstat command show me it still listening, but I can connect it even to 127.0.0.1:65001.
[image]

The connections already connected to 65001 will not be disconnected, but no new one can connect.
After a while(20-40min), the client side of openmptcprouter will try to restart Glorytun VPN, and few moments later 65001 become connect able again, then Glorytun VPN connect successful, and then port 65001 unreachable again, loop ...

This is the log in OMR client side:
```
Wed May 19 11:13:38 2021 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Wed May 19 11:13:38 2021 user.notice post-tracking-post-tracking: Glorytun VPN down, restart it
Wed May 19 11:13:38 2021 daemon.info glorytun[29158]: STOPPED tun0
Wed May 19 11:13:38 2021 daemon.notice netifd: Network device 'tun0' link is down
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' is now down
Wed May 19 11:13:38 2021 daemon.debug dnsmasq[7287]: stopped listening on tun0(#3264): 10.255.255.2 port 53
Wed May 19 11:13:38 2021 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Wed May 19 11:13:38 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Wed May 19 11:13:38 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed May 19 11:13:38 2021 daemon.info glorytun[4354]: INITIALIZED tun0
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed May 19 11:13:38 2021 daemon.notice netifd: Network device 'tun0' link is up
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed May 19 11:13:38 2021 daemon.notice netifd: Interface 'omrvpn' is now up
Wed May 19 11:13:38 2021 user.notice ModemManager: hotplug: add network interface tun0: event processed
Wed May 19 11:13:38 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Wed May 19 11:13:38 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed May 19 11:13:38 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed May 19 11:13:42 2021 daemon.err /usr/bin/ss-local[18208]: getpeername: Socket not connected
Wed May 19 11:13:48 2021 user.notice OMR-VPS: Can't get vps token, try later
Wed May 19 11:13:48 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed May 19 11:13:48 2021 user.notice omr-bypass: Reload dnsmasq...
Wed May 19 11:13:48 2021 daemon.info dnsmasq[7287]: read /etc/hosts - 4 addresses
Wed May 19 11:13:48 2021 daemon.info dnsmasq[7287]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Wed May 19 11:13:48 2021 user.notice omr-bypass: OMR-ByPass is running
Wed May 19 11:13:48 2021 daemon.debug dnsmasq[7287]: listening on tun0(#3265): 10.255.255.2 port 53
Wed May 19 11:13:54 2021 user.notice MPTCP: No master multipath defined, setting it to eth1
Wed May 19 11:13:54 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Wed May 19 11:13:57 2021 daemon.err /usr/bin/ss-local[18208]: getpeername: Socket not connected
Wed May 19 11:14:04 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Wed May 19 11:14:07 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 1.1.1.1, x.x.x.42, x.x.x.130)
Wed May 19 11:14:07 2021 user.notice Shadowsocks: Rules DOWN
Wed May 19 11:14:10 2021 daemon.err glorytun[4354]: OMR_VPS_IP.65001: connected
Wed May 19 11:14:10 2021 daemon.info glorytun[4354]: STARTED tun0
Wed May 19 11:14:11 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Wed May 19 11:14:12 2021 daemon.err /usr/bin/ss-local[18208]: getpeername: Socket not connected
Wed May 19 11:14:13 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.0.0.1)
Wed May 19 11:14:13 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Wed May 19 11:14:13 2021 user.notice Shadowsocks: Rules UP
Wed May 19 11:14:13 2021 user.notice Shadowsocks: Reload omr-bypass rules
Wed May 19 11:14:14 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed May 19 11:14:14 2021 user.notice omr-bypass: Reload dnsmasq...
Wed May 19 11:14:14 2021 daemon.info dnsmasq[7287]: read /etc/hosts - 4 addresses
Wed May 19 11:14:14 2021 daemon.info dnsmasq[7287]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Wed May 19 11:14:14 2021 user.notice omr-bypass: OMR-ByPass is running
```

## Possible Solution
no idea

## Steps to Reproduce the Problem

  1.Use OMR with 4 LTE routers
  2.check the OMR log and glorytun-tcp server port

## Context (Environment)
This bug made OMR not stable, my connections always break.
Please help to find out the reason why glorytun-tcp port 65001 become unreachable causing Glorytun VPN restart.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-05-20:
You need to set a connection as Multipath Master.

johnlipss, 2021-05-20:
> You need to set a connection as Multipath Master.

Still not working.
Why glorytun-tcp server port 65001 get unreachable after a while glorytun-tcp connected(even 127.0.0.1:65001)?
In my opinion this is a omr-vps problem, maybe iptables or ipset?

this is the new log:
```
Thu May 20 09:48:37 2021 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Thu May 20 09:48:37 2021 user.notice post-tracking-post-tracking: Glorytun VPN down, restart it
Thu May 20 09:48:37 2021 daemon.err glorytun[12928]: read: Operation timed out
Thu May 20 09:48:37 2021 daemon.info glorytun[12928]: STOPPED tun0
Thu May 20 09:48:37 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Thu May 20 09:48:42 2021 daemon.info procd: Instance glorytun::instance1 pid 12928 not stopped on SIGTERM, sending SIGKILL instead
Thu May 20 09:48:42 2021 daemon.notice netifd: Network device 'tun0' link is down
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' is now down
Thu May 20 09:48:42 2021 daemon.debug dnsmasq[29715]: stopped listening on tun0(#3345): 10.255.255.2 port 53
Thu May 20 09:48:42 2021 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Thu May 20 09:48:42 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Thu May 20 09:48:42 2021 daemon.info glorytun[17397]: INITIALIZED tun0
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu May 20 09:48:42 2021 daemon.notice netifd: Network device 'tun0' link is up
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu May 20 09:48:42 2021 daemon.notice netifd: Interface 'omrvpn' is now up
Thu May 20 09:48:42 2021 user.notice ModemManager: hotplug: add network interface tun0: event processed
Thu May 20 09:48:42 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Thu May 20 09:48:42 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu May 20 09:48:42 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Thu May 20 09:48:44 2021 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.101.1 dev eth1 weight 2 nexthop via 192.168.102.1 dev eth2 weight 1 nexthop via 192.168.103.1 dev eth3 weight 1 nexthop via 192.168.104.1 dev eth4 weight 1
Thu May 20 09:48:50 2021 daemon.err /usr/bin/ss-local[18208]: getpeername: Socket not connected
Thu May 20 09:48:52 2021 user.notice OMR-VPS: Can't get vps token, try later
Thu May 20 09:48:52 2021 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 20 09:48:53 2021 user.notice omr-bypass: Reload dnsmasq...
Thu May 20 09:48:53 2021 daemon.info dnsmasq[29715]: read /etc/hosts - 4 addresses
Thu May 20 09:48:53 2021 daemon.info dnsmasq[29715]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Thu May 20 09:48:53 2021 user.notice omr-bypass: OMR-ByPass is running
Thu May 20 09:48:53 2021 daemon.debug dnsmasq[29715]: listening on tun0(#3346): 10.255.255.2 port 53
Thu May 20 09:48:58 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Thu May 20 09:48:59 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http x.x.x.164, x.x.x.77, 1.1.1.1)
Thu May 20 09:48:59 2021 user.notice Shadowsocks: Rules DOWN
Thu May 20 09:49:05 2021 daemon.err /usr/bin/ss-local[18208]: getpeername: Socket not connected
Thu May 20 09:49:07 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)
Thu May 20 09:49:09 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http x.x.x.42)
Thu May 20 09:49:09 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Thu May 20 09:49:09 2021 user.notice Shadowsocks: Rules UP
Thu May 20 09:49:09 2021 user.notice Shadowsocks: Reload omr-bypass rules
Thu May 20 09:49:09 2021 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 20 09:49:09 2021 user.notice omr-bypass: Reload dnsmasq...
Thu May 20 09:49:09 2021 daemon.info dnsmasq[29715]: read /etc/hosts - 4 addresses
Thu May 20 09:49:09 2021 daemon.info dnsmasq[29715]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Thu May 20 09:49:09 2021 user.notice omr-bypass: OMR-ByPass is running
Thu May 20 09:49:11 2021 daemon.err glorytun[17397]: OMR_VPS_IP.65001: connected
Thu May 20 09:49:11 2021 daemon.info glorytun[17397]: STARTED tun0
Thu May 20 09:49:13 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
```

Ysurac, 2021-05-20:
The problem is that VPN doesn't answer to ping. You have only one router using this server ?
Did you check log on the VPS, in /var/log/daemon.log ?

johnlipss, 2021-05-24:
> The problem is that VPN doesn't answer to ping. You have only one router using this server ?
> Did you check log on the VPS, in /var/log/daemon.log ?

Yes only one router.
I didn't find daemon.log file in /var/log/.
I found out OMR network(both UDP and TCP) and glorytun breaks every 20 minutes.

I wrote a script `tcping -p 65001 -c 1 my_omr_server` test glorytun-tcp port in loop, find out glorytun-tcp port get unreachable every time after 90 seconds of glorytun vpn connected, but glorytun vpn break every 20 minutes, not 90 seconds.

Ysurac, 2021-05-24:
What do you use as VPS server ? Ubuntu ? Try Debian 10.

johnlipss, 2021-05-25:
> What do you use as VPS server ? Ubuntu ? Try Debian 10.

Yes, Ubuntu 20.04.
Ok, I'll try debian 10.

closed 2021-08-28 by github-actions[bot]