Ysurac/openmptcprouter #1845: VLAN tags for LAN interface

gnosoz, 2021-05-20:
**Is your feature request related to a problem? Please describe.**
Setting up VLANs for a LAN interface

**Describe the solution you'd like**
Would be great to be able to add VLAN_ID tag field in the interface configuration to simplify the creation of tagged lan traffic on a specific interface 

**Describe alternatives you've considered**
Manual creation of the bridge / interface with VALN tag specification

**Additional context**
This would be extremely useful when passing traffic on 2 vlans to managed switches for example XX for voice directly from main router and vlan tag XX for data from OMR to the switch.

Ysurac, 2021-05-20:
I didn't added it in wizard because it's a more advanced usage.
In some case you need/want a new firewall zone and sometimes no.
I can add a "vlan" box in wizard to choose vlan, but more difficult to add a new interface button creation.

For now you can add/set a vlan in Network->Interfaces and use "eth0.2", as physical interface, for vlan 2 on interface eth0.
(I think I will add a vlan box in Network->Interfaces)

gnosoz, 2021-05-31:
Hi Yannick,
I see your point and it does make sense :)

Adorfer, 2021-05-31:
it would be nice to have some kind of "automatic default-vlan-mode" for devices with only one eth, like Raspi by default. 

something like "Wan1:-> VLAN1, WAN2->VLAN2.... LAN->VLAN7" plus dhcp client enabled on all those WANs.
Then people can get a cheap 30€/USD vlan switch, make a simple tagging setup (or even take the option "port-VLAN" where the config above is made with just one click.

closed 2021-09-03 by github-actions[bot]