Ysurac/openmptcprouter #1846: Why do modems need static addresses?

MatthewFrench, 2021-05-21:
The guide I’m following: https://milankragujevic.com/openmptcprouter-true-bonding-of-2-wan-connections-for-cheap

Piece: `Set the modems’ LAN IP addresses to 192.168.10.1 for the first and 192.168.11.1 for the second modem `

Why do the providers being bonded need static addresses with dhcp disabled? Could the software be made to work with any given connection?

Bumpalert, 2021-05-21:
What i understand is same make modems set same gateway and routing will confuse if 2 devices with a me gateway on wan side. If you change gateway ips you can run on dhcp. I use dhcp as my wan  gateways are different from each other

Ysurac, 2021-05-21:
@MatthewFrench DHCP need to be disabled on modem and use fixed IPs, when modems are connected to a switch and not directly to a network port on OpenMPTCProuter, else multiples DHCP on same network will make a conflict.
It's also possible to use VLAN.

jrowny, 2021-06-14:
I had a modem which doesn't allow disabling DHCP and I just stuck it behind a cheap glinet as WAN with dhcp disabled on the glinet. Not sure if there's a better way to do it, but it worked for me.

closed 2021-09-19 by github-actions[bot]