Ysurac/openmptcprouter #1858: How to get the wan signal strength and operator info from LTE modem?

gudokim, 2021-05-24:
[image]

I have a project to get the modem info and would like to know how.

To get the wan signal strength and operator info from LTE hilink modem..
As I know , we can get info only via web from modems in the hilink mode.

Every E8372h-320 lte modem in the hilink mode has different password for log-in to get info.
I wonder how openmptcprouter can get the modem info without knowing password.

best regards,

Ysurac, 2021-05-24:
This is the code that is used to get info from huawei: https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/luci-app-openmptcprouter/root/bin/omr-huawei
These infos doesn't need any login.

gudokim, 2021-05-24:
Wow~~ Thank you so much!!

closed 2021-05-24 by gudokim