Ysurac/openmptcprouter #1859: Couple questions regarding IPv6 functionality

Leseratte10, 2021-05-24:
I have not yet installed OpenMPTCProuter anywhere, I'd like to ask a couple questions before I buy hardware to do that. I hope that that's fine in a Github issue report, I didn't find a forum or anything else to ask questions. 

Out of the internet connections I want to merge together with OMR, one of them will be an IPv6 connection with IPv4 only provided over CGNAT. Others will either be IPv4-only or IPv4/IPv6 dual stack. 

I've read #1619 and that the next release should support IPv6 on the WAN side, but I have a couple of questions about that. 
I hope it's fine that I created my own issue report about that. 

Question 1: 
Does it also support combining different connection types? If I have a VPS with IPv4 1.2.3.4 and IPv6 2001:db8::1, can I configure OMR to contact my VPS through that IPv4 IP on internet connections 1 and 2, and to that IPv6 IP on internet connections 3 and 4? Some of the connections won't have IPv6 so I can't use IPv6 for all of them, but some also won't have native IPv4 and I don't want to run my connection through an ISP-provided CGNAT. 

Question 2: 
If I have a particular public IPv4 and a particular IPv6 /62 subnet available on the VPS, am I understanding it correctly that A) any IPv4 connection done from a device behind the OMR will use that VPS' public IPv4, and any IPv6 connection will use an IP from that IPv6 subnet?
How does that work? Does OMR run a Route Advertisement Daemon (radvd or something) so that all network clients believe that the VPS' prefix is provided by a local router? That would be my preferred solution. Or does OMR provide link-local / ULA adresses and routes them all to the internet through a single IP? That'd be pretty ugly. 
If I have X devices behind the OMR, they will all use their own dedicated IPv6 from that VPS' subnet - right? And they all can receive incoming connections?

Question 3: 
I read on the documentation that I can make the VPS port-forward all IPv4 ports to the OMR, and then add port forwardings in the OMR interface to particular internal devices, basically as if the OMR was a "regular consumer router" and I'd add port forwardings on that. How does that work on IPv6? I'm assuming there will be no need for port forwardings per-se as there's no NAT with IPv6, but does OMR support incoming connections on IPv6? Is there a way to firewall / filter them or will my local devices behind the OMR have all of their IPv6 ports exposed to the internet?

Sorry for asking so many technical questions, but IPv6 is important to my network so if that won't work properly I probably won't be able to use OpenMPTCProuter (sadly); and I don't want to buy hardware just to test that.

Ysurac, 2021-05-24:
1. It support combining different connection types (on next 0.58), VPS need to be set with one IPv4 and one IPv6. But in fact CGNAT should not be a real problem.
2. OMR will use default output IPs of the VPS. You can delegate an IPv6 prefix to internal network (else it's ULA addresses), but output IP will still be default IPv6 output of the VPS (this is due to the use of a proxy, no problem with proxy disabled but VPN are far slower).
3. You can redirect all ports to OMR (port 0-64999), but it's a bad idea, far better to only open needed port on the router and this will open port on the VPS side. NAT is used with ULA IPv6, else it's open opening ports. All ports are blocked in entry side by default from the VPS (and WAN) in IPv4 and IPv6.

Leseratte10, 2021-05-24:
Wow, thanks for the quick response. Couple follow-up questions to that: 
1. Okay, it does support combining IPv4 and IPv6, but can I manually configure, for each internet connection, whether it should use IPv4 or IPv6? If I have two internet connections, each providing IPv4 and IPv6, but on one of them IPv4 is a tunnel and on the other IPv6 is a tunnel, can I tell OMR to only use IPv6 on the 1st one and only IPv4 on the 2nd one?

2. Okay, I can delegate the VPS' IPv6 prefix to my internal devices but they'll be NATed to the one main VPS IPv6, and not all routed to the internet as-is? Okay, that's pretty ugly. When I turn the proxy setting off that issue will be fixed and each device will communicate with the internet with its own actual IPv6 address?

3. So, if I open IPv4 port 80 in OMR it will automatically configure the VPS to forward port 80 to the OMR. But I didn't fully understand yet how that will work with IPv6. Can I tell it "Okay, forward port 80 for this one device but don't forward it for this other device"? Or can I only say "forward port 80" and then *every* IPv6-capable device will be reachable on port 80?

I guess I should definitely go grab a cheap-ish device like a Raspberry with a bunch of USB-to-LAN dongles to do a test run before I buy actual expensive hardware to make sure IPv6 is working like I need it ...

Ysurac, 2021-05-24:
1. If you don't set IPv6 or IPv4 IPs/gateway on a connection it doesn't use it. Same if you don't set DHCPv6 for IPv6 and DHCP for IPv4.
2. Not NATed, it's over a proxy. Ugly but no real choices when a proxy is used. Yes without the proxy all should communicate directly with internet but at a slow speed.
3. Yes, you can set public IPv/IPv6 allowed for the redirection. I mean "only" opening ports. Even with ULA disabled, firewall is still active on VPS side so it work same way and you have to open needed ports.

No need to use USB-to-LAN dongle, "virtual" port using MacVLAN are created on the device, you only need to have all connected to same switch. Else you can also use VLAN.

Leseratte10, 2021-05-26:
I have just set up OpenMPTCProuter in a virtual machine, and wow does it have many settings. I'm impressed. 

So there's two different ways to get my internet traffic to my VPS and from there into the internet, VPN and Proxy. I can either pick "Proxy" and have a slow connection but with the correct source IPv6 addresses for outgoing traffic, or I can pick "VPN" and have a faster connection but wrong (translated to the VPS public IPv6 like in a NAT) IPv6 address?

Is there any plans to add full IPv6 support to the VPN route to have that work at a faster speed, too?

Ysurac, 2021-05-26:
No. It's the opposite. Proxy is fast but you have VPS main IPv6, VPN is slow but you should have client IPv6.
Not possible to add full IPv6 route support to the proxy.

closed 2021-08-30 by github-actions[bot]