Ysurac/openmptcprouter #1861: 192.168.100.1 refused to connect.

Yersi88, 2021-05-24:
## Expected Behavior
web interface loads at 192.168.100.1

## Current Behavior

192.168.100.1 refused to connect. When trying to access router from web browser. When I try to ping 192.168.100.1 from cmd it responds. 
[image]

 computer is connected straight with ethernet cable to eth0 of OpenMPTCProuter box. IP of computer is set to 192.168.100.13.

## Specifications

  - OpenMPTCProuter version:  openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined-efi.img.gz 
  - OpenMPTCProuter VPS version: OpenMPTCProuter Server 0.1024 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-05-25:
Can you try to go on it with SSH (user root, no password by default and port 22) ?

Yersi88, 2021-05-25:
> Can you try to go on it with SSH (user root, no password by default and port 22) ?

I tried to ssh into 192.168.100.1 but was not successful. 

I also noticed that pinging 192.168.100.1 still works when network cable is unplugged from PC, not sure what to think about that. (Computer is set up with gateway: 192.168.100.1, mask: 255.255.255.0 and ip: 192.168.100.13 )

Also, isn't it supposed to have DHCP server running as default? (It doesn't give IP address to PC when I plug PC's network card to eth0)

Ysurac, 2021-05-25:
You seems to have something that answer at IP 192.168.100.1 but it's not OpenMPTCProuter.
Yes, OpenMPTCProuter have a DHCP server running as default.

Yersi88, 2021-05-25:
> You seems to have something that answer at IP 192.168.100.1 but it's not OpenMPTCProuter.
> Yes, OpenMPTCProuter have a DHCP server running as default.

There shouldn't be anything with that address, I have all other network adapters disabled just the one that's connected directly to OpenMPTCProuter box is enabled. When I remove the manual ipv4 configuration than there's no response from 192.168.100.1

Why is the dhcp not working though? Maybe I'm doing something stupid. I just booted the OpenMPTCProuter  from usb stick and plugged in the network cable to eth0 and to my pc.

hairy-dog, 2021-05-25:
I'd check the PC's network settings next.
"ipconfig /all"

Yersi88, 2021-05-25:
> I'd check the PC's network settings next.
> "ipconfig /all"

[image]

When IP is not set manually I get no IP from OpenMPTCProuter

Ysurac, 2021-05-25:
Check that you don't have a firewall on your client PC.
Check also that ethernet connection is correct (right port on the router).
Check that router is really booted to OpenMPTCProuter and that there is no error.

Yersi88, 2021-05-25:
> Check that you don't have a firewall on your client PC.
> Check also that ethernet connection is correct (right port on the router).
> Check that router is really booted to OpenMPTCProuter and that there is no error.

1. No firewall on client
2. I'm running it on mini pc which only has one ethernet port. 
3. 
[image]

Ysurac, 2021-05-25:
You can try to check IP on OpenMPTCProuter via ```ip a``` and set fixed IP on your client like before and try to ping it from the router.

Yersi88, 2021-05-25:
> You can try to check IP on OpenMPTCProuter via `ip a` and set fixed IP on your client like before and try to ping it from the router.

[image]

[image]

Can't ping the client as well. This is odd.

hairy-dog, 2021-05-25:
I wonder about the mention of connectify in the ipconfig results. Doest connectify use some network software in the background?

hairy-dog, 2021-05-25:
The ipconfig screengrab was of a pc with no network cable connected.

Yersi88, 2021-05-25:
> I wonder about the mention of connectify in the ipconfig results. Doest connectify use some network software in the background?

Connectify was running on different PC never on this one.

> The ipconfig screengrab was of a pc with no network cable connected.

Network cable is connected. But no ip allocated from OpenMPTCProuter.

You can see two network cards Ethernet and Ethernet 2. One is 'Media disconnected' the other one is connected to the OpenMPTCProuter direct via ethernet cable.

Adorfer, 2021-05-26:
other approach: 
1) "ping 192.168.100.1"
2) "arp -a", look for the 192.168.100.1 and the IEEE/MAC associated. hopefully it's 68-1d-ef-22-f8-cf.

hairy-dog, 2021-05-26:
I think the second connection in ipconfig is showing an IP that displays when no dhcp server has been found: effectively a network that is disconnected.
My suspicion is that the router is set to prevent clients contacting one another, and only connecting out to the internet. Can other devices on the network ping each other?

Yersi88, 2021-05-26:
> I think the second connection in ipconfig is showing an IP that displays when no dhcp server has been found: effectively a network that is disconnected.
> My suspicion is that the router is set to [...]

There are no devices. It's my PC (client) connecter through ethernet cable straight to the PC where OpenMPTCProuter is running. One network adapter connected straight to other network adapter.

Adorfer, 2021-05-26:
if you do not like to do the arp thing, please "tracert 192.168.100.1"

flimo44, 2021-05-26:
Hello ,
On screem shoot i see your manuel Pc address set at  192.168.100.1  and gateway 192.168.100.1 ... The IP's are reversed I think, right?

Ysurac, 2021-05-26:
True, error in Windows PC config here.

Adorfer, 2021-05-26:
so either open the firwall on the PC to access the local webserver. 

Or change the address in the browser to 192.167.100.13 and change the IP in the OMR as well.

Yersi88, 2021-05-27:
> True, error in Windows PC config here.

Indeed a really silly mistake! Thanks for all the support <3

closed 2021-05-27 by Yersi88