Ysurac/openmptcprouter #1863: Issues with openmtcprouter and modem/adapter

Leaptear, 2021-05-27:
## Expected Behavior
Connect to Modem & show up as a port

## Current Behavior
The current behaviour is im using a PCI-E Mini to USB and It connects fine and works on Windows but on Linux which is supported it seems OPENMPTCRouter shows this issue below 

[image]

It establishes the connection and then drops the adapter any idea?

## Specifications

  - OpenMPTCProuter version:  Latest Version
  - OpenMPTCProuter VPS version:  Latest Version
  - OpenMPTCProuter platform: Latest Version

Ysurac, 2021-05-27:
It seems to attach modem to ttyUSB0 to ttyUSB4 and you should be able to use it with modemmanager or mbim.

Leaptear, 2021-05-27:
Hi @Ysurac Thanks for your reply, I believe that your OpenMPTCRouter has modemmanager included would you be abl e to advise how I make it be seen as a port to select to use? as it seems that it binds as wwan0 but then unregisters thank you.

Ysurac, 2021-05-27:
modemmanager is installed by default on OpenMPTCProuter.
Your modem doesn't seems to be used as cdc_ether (it's why there is no ethX port created), I think it's possible to use it with modemmanager, but you can search on openwrt forum/doc if you can find more info.
(I don't use modemmanager, I have huawei keys recognized as eth)

Leaptear, 2021-05-28:
Thanks, Last question when using USB modems it works fine but I am curious it seems like it's drawing connection from both when at low network usage but when I use a lot of network for example speedtest it just uses 1 modem instead of combining the two to give a combined throughput? any ideas? thanks @Ysurac

closed 2021-09-01 by github-actions[bot]