Ysurac/openmptcprouter #1865: iPad tether with error ipheth 1-1.1:4.2: ipheth_rcvbulk_callback: urb status: -75

darthclide, 2021-05-28:
## Expected Behavior
iPad tether doesn't give up reconnecting after 1 failed ping.

## Current Behavior
Seemingly at random the gateway will go down, and the following errors are in the log (not necessarily at the same time):

wan5 (eth3) switched off because gateway down and ping from 172.20.10.3 error (80.67.169.12,80.67.169.40)
ipheth 1-1.1:4.2: ipheth_rcvbulk_callback: urb status: -75

Upon restarting the interface  on this page-> http://192.168.100.1/cgi-bin/luci/admin/network/network either of 2 things happen:
1. It reconnects with a green check mark and works for a random amount of time before turning off again
2. Does not reconnect and still says gateway down.

## Possible Solution
https://lore.kernel.org/netdev/d61ad9565e29a07086e52bc984e8e629285ff8cf.camel@suse.com/T/
This thread seems to be talking about my exact same issue even if the number is slightly different. Do you think it would help to reduce the URB buffer size? I already tried reducing the MTU to 1498 like the OP suggested, but it did not work. How can we ensure it doesn't give up after 1 failed ping? I have seen it fail even on 8.8.8.8 and 8.8.4.4

## Steps to Reproduce the Problem

  1. Plug in iPad
  2. Turn on hotspot
  3. Wait a random amount of time

## Context (Environment)
I can't have it randomly disconnecting on me. It makes my stream crash.

## Specifications

  - OpenMPTCProuter version: 57.3
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: RPI4

darthclide, 2021-05-28:
Unsure if it is just a coincidence, but after manually adding a OMR-Tracker for the iPad that uses httping instead of ping, I haven't had it switch to a permanent Gateway Down for ~10 hours now. Will report back later if it breaks again.

Ysurac, 2021-06-05:
I can't really help but you can try 0.58 that have the patch for 1514 MTU if it's iOS 14, but as you already tried to lower MTU not sure that this will really change anything.

darthclide, 2021-06-05:
Why do you think httping worked for 1 week before going out? I stupidly forgot to try restarting the interface with OMR instead of physically turning off and on the TetherMe. We need to determine if this is maybe a TetherMe issue and not something you can control in OMR.

You say if it's iOS 14, but does that include jailbreaking a device that was formerly on iOS 14.5?

Is there a chance I need to try lowering the MTU by a lot more? Is MTU the same thing as USB packet size?

closed 2021-09-08 by github-actions[bot]