Ysurac/openmptcprouter #1866: Android USB Tethering, after re-plugging USB interface doesn't go back up

Yersi88, 2021-05-28:
I have one wan connected via usb0 to my Android phone. It works nicely but when I disconnect the phones usb cable for a while and than I plug it back it the interface doesn't come back again. Even restarting the interface (Network>Interfaces>Restart) doesn't help. Only thing that helps is rebooting through System > Reboot, after that my android usb0 wan is up again. 

Is there more convenient way to bring that usb0 wan up, other than rebooting the router?

Ysurac, 2021-05-28:
Can you check what you have in State->System log when you plug it back ?

Yersi88, 2021-05-28:
Fri May 28 09:33:31 2021 daemon.notice netifd: wan3 (26316): udhcpc: sending discover
Fri May 28 09:33:31 2021 user.notice ModemManager: hotplug: add network interface usb0: event processed
Fri May 28 09:33:31 2021 user.notice ModemManager: hotplug: event reported: action=add, name=usb0, subsystem=net
Fri May 28 09:33:31 2021 user.notice ModemManager: hotplug: no need to wait for modem at sysfs path /sys/devices/pci0000:00/0000:00:15.0/usb1/1-4
Fri May 28 09:33:31 2021 user.notice OMR-Rename: dir:  - modalias: usb:v0BDAp8153d3000dc00dsc00dp00icFFiscFFip00in00 - chk_modalias: usb:v04E8p6864d0C00dc00dsc00dp00icE0isc01ip03in00 - ifname: eth1 - INTERFACE: usb0
Fri May 28 09:33:31 2021 user.notice OMR-Rename: dir:  - modalias: usb:v04E8p6864d0C00dc00dsc00dp00icE0isc01ip03in00 - chk_modalias: usb:v04E8p6864d0C00dc00dsc00dp00icE0isc01ip03in00 - ifname: usb0 - INTERFACE: usb0
Fri May 28 09:33:31 2021 user.notice OMR-Rename: dir:  - modalias: usb:v0BDAp8153d3000dc00dsc00dp00icFFiscFFip00in00 - chk_modalias: usb:v04E8p6864d0C00dc00dsc00dp00icE0isc01ip03in00 - ifname: eth1 - INTERFACE: usb0
Fri May 28 09:33:31 2021 user.notice OMR-Rename: dir:  - modalias: usb:v04E8p6864d0C00dc00dsc00dp00icE0isc01ip03in00 - chk_modalias: usb:v04E8p6864d0C00dc00dsc00dp00icE0isc01ip03in00 - ifname: usb0 - INTERFACE: usb0
Fri May 28 09:33:31 2021 kern.info kernel: [ 2752.705253] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev d)
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707821] sdhci-pci 0000:00:1b.0: GPIO lookup for consumer cd
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707825] sdhci-pci 0000:00:1b.0: using ACPI for GPIO lookup
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707829] acpi device:3f: GPIO: looking up cd-gpios
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707832] acpi device:3f: GPIO: looking up cd-gpio
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707833] sdhci-pci 0000:00:1b.0: using lookup tables for GPIO lookup
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707835] sdhci-pci 0000:00:1b.0: No GPIO consumer cd found
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707837] sdhci-pci 0000:00:1b.0: GPIO lookup for consumer (null)
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707838] sdhci-pci 0000:00:1b.0: using ACPI for GPIO lookup
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707839] acpi device:3f: GPIO: looking up gpios
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707841] acpi device:3f: GPIO: looking up gpio
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707842] acpi device:3f: GPIO: looking up 0 in _CRS
Fri May 28 09:33:31 2021 kern.debug kernel: [ 2752.707891] sdhci-pci 0000:00:1b.0: No GPIO consumer (null) found
Fri May 28 09:33:34 2021 daemon.info [7367]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:15.0/usb1/1-4': not supported by any plugin
Fri May 28 09:33:34 2021 daemon.notice netifd: wan3 (26316): udhcpc: sending discover
Fri May 28 09:33:37 2021 daemon.notice netifd: wan3 (26316): udhcpc: sending discover

Yersi88, 2021-05-28:
This turns out to be glitch in my Android. I have the default USB option set to Tethering. I either need to switch tethering off and on in Android or just don't have it as default. Nothing to do with  openmptcprouter itself!

closed 2021-05-28 by Yersi88