Ysurac/openmptcprouter #1869: link between wifi and lan interfaces

SylvainLegrand, 2021-05-30:
## Expected Behavior
I would like the wired and wifi networks to be linked.
For example, I would like to use a wifi printer from a computer connected to the wired network.

## Current Behavior
My wifi printer works well and I can use it from a phone or any devices connected to the wifi interface.
My computer reach the internet through the router but can't be link to any devices in the wifi interface.
It is as if I had built 2 different networks without any connection between them.

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: Server 0.1024
  - OpenMPTCProuter platform: RPI4/x86_64
 
  - Wifi configuration: Mode => Access point / Network => wifi
  - Wifi interface configuration: IPV4 192.168.100.2 / gateway 192.168.100.1 / type bridge / firewall settings => in the same zone with the lan interface
  - Lan interface: IPV4 192.168.100.1

Thanks for your help.

Ysurac, 2021-06-01:
You can try to follow documentation: https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi

SylvainLegrand, 2021-07-29:
Hi,

I have already followed this documentation.
Can you take a look at the screenshots I made?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Thanks.

Sylvain Legrand.

closed 2021-11-01 by github-actions[bot]