Ysurac/openmptcprouter #1870: OMR-ByPass Amazon pbs

JmRy, 2021-05-31:
- Version du micrologiciel：openmptcprouter v0.57.3 r0+15225-bfc433efd4
- Modèle：Raspberry Pi 4 Model B Rev 1.2
- VPS : OVH kimsufi
- IPV6 activé
- 4 WAN (Adsl + 4G)

Bonjour,

J'ai actuellement un problème avec les ORM-Bypass d'Amazon.

Si je veux utiliser PrimeVideo, je suis obligé d'ajouter les services `amazon` et `amazonvideo` en les redirigeant sur le WAN1 pour que cela fonctionne.

Par contre, il ne m'est plus possible de me connecter au site www.amazon.fr depuis mon pc. La résolution dns se lance, puis le browser indique un timeout.

Si j'enlève le service `amazon` en laissant uniquement `amazonvideo`, le pc se connecte sans problème à www.amazon.fr, mais du coup Prime m'indique que je suis connecté à un proxy donc me bloque.

Merci de votre aide

Cordialement

Ysurac, 2021-06-01:
Pour PrimeVideo il est nécessaire d'ajouter uniquement amazonvideo, du moins ça fonctionne dans mon cas.
Qu'est-ce que ça donne pour accéder à amazon.fr si IPv6 est désactivé dans le navigateur ?

JmRy, 2021-06-02:
Sous Chrome lancé dans le terminal en désactivant IPV6 avec la commande `./'Google Chrome' --disable-async-dns` *(seule méthode que j'ai trouvé sur le net pour désactivé IPV6)* toutes les demandes vers un site amazon .fr, .com, etc... provoque un timeout avec au final l'erreur ci-dessous affiché dans le terminal.
`[27564:12291:0602/101730.404190:ERROR:ssl_client_socket_impl.cc(947)] handshake failed; returned -1, SSL error code 1, net_error -7`

Note qu'un `ping6 amazon.fr` renvoi  `nodename nor servname provided, or not known` donc à priori pas de serveur ipv6 pour leur site. Mais peut-être le sais-tu déjà ;-)

Si je n'ajoute pas le bypass amazon quel que soit l'appareil (PC Win ou Mac, Google TV, Apple TV, iPhone, Tablette Android) je suis toujours détecté comme me connectant via un proxy donc bloqué.

JmRy, 2021-06-02:
Oui le DNS de la Google TV pointe bien vers l'IP d'OpenMPTCrouter
[image]

Pour info, le routeur principal de mon réseau est une appliance OPNSense et je ne pense pas qu'elle soit en cause. J'ai fait plusieurs essais en désactivant tous les filtres et sécurités (Suricata, Pare-Feu, Squid, etc...). De plus, j'ai directement branché mon portable avant l'appliance pour qu'il obtienne son adresse d'OpenMPTCrouter. Dans le 2 cas cela ne change rien.

Je confirme _(enfin je ne pense pas...)_ que cela ne provient pas d'IPv6, car j'avais le même pbs lorsque j'ai installé OpenMTCProuter sur mon réseau. Ce dernier étant à l'époque uniquement en IPv4 (box comprises)

C'est peut-être un pbs de résolution DNS ? À ce sujet, si j'ai bien compris on doit **obligatoirement** faire pointer les clients du réseau vers l'IP d'OpenMPTCrouter ce **seulement** si on utilise ORM-Bypass ?

Ysurac, 2021-06-02:
Oui effectivement le DNS doit pointer vers OpenMPTCProuter uniquement si OMR-ByPass est utilisé.
Il peut y avoir un soucis de MTU.
Si tu ajoutes un domaines comme "ifconfig.co" dans omr-bypass, qu'est-ce que ça donne ? Ca fonctionne ou pas ?

JmRy, 2021-06-02:
Oui en ajoutant ifconfig.co comme demandé et en le dirigeant vers un WAN, l'IP affichée est celle de la box.

Ysurac, 2021-06-02:
dommage, ça m'arrange pas pour trouver le soucis...

Ysurac, 2021-06-02:
Il me faudrait la configuration d'omr-bypass, via ssh ```uci show omr-bypass``` ou section commençant par "omr-bypass" dans Systèmes->OpenMPTCProuter->Voir tous les paramètres.

JmRy, 2021-06-02:
```
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.lo=interface
omr-bypass.lo.id='5'
omr-bypass.eth0=interface
omr-bypass.eth0.id='6'
omr-bypass.wan1=interface
omr-bypass.wan1.id='3'
omr-bypass.wan2=interface
omr-bypass.wan2.id='4'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1200'
omr-bypass.wan3=interface
omr-bypass.wan3.id='7'
omr-bypass.wan4=interface
omr-bypass.wan4.id='8'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='ifconfig.co'
omr-bypass.@domains[0].interface='wan4'
omr-bypass.@dpis[0]=dpis
omr-bypass.@dpis[0].proto='amazonvideo'
omr-bypass.@dpis[0].interface='wan4'
omr-bypass.@dpis[1]=dpis
omr-bypass.@dpis[1].proto='amazon'
omr-bypass.@dpis[1].interface='wan4'
```

Ysurac, 2021-06-02:
Aucun soucis dans cette configuration.
Il me faudra la section "network" et "dhcp" ainsi que le résultat de la commande ```ip r``` (en enlevant l'IP du VPS avant de copier ici).

JmRy, 2021-06-02:
j'ai eu l'idée d'ajouter le domaine `amazon.fr` dans la config d'omr-bypass sur un WAN différent des services `amazon` et `amazonvideo` et du coup j'ai accès au site ...?!

Par contre et pour info, pour enlever le blocage proxy/vpn d'amazon sur les appareils précédemment utilisés, il faut les supprimer du compte Prime Video, purger l'appli (install/resintall) et refaire le process d'authentification de l'appareil. Du coup, Prime fonctionne sur mon iPhone et l'Apple TV sur le WAN4 (box 4G), mais toujours pas sur la Google TV dont l'appli Prime indique un pbs de connexion internet. Bref, la _LEM_ ... ;-) 

Voici les infos demandées

**ip r**
```
default via 10.255.255.1 dev tun0 
default via 192.168.30.10 dev wan1 metric 3 
default via 192.168.6.1 dev wan3 metric 7 
default via 192.168.7.1 dev wan4 metric 8 
default via 10.255.255.1 dev tun0 metric 1200 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
xx.XX.XX.XX
        nexthop via 192.168.30.10 dev wan1 weight 10 
        nexthop via 192.168.6.1 dev wan3 weight 1 
        nexthop via 192.168.7.1 dev wan4 weight 1 
        nexthop via 192.168.1.1 dev wan2 weight 1 
127.0.0.0/8 dev lo proto static scope link metric 5 
192.168.1.0/24 dev wan2 scope link metric 4 
192.168.6.0/24 dev wan3 scope link metric 7 
192.168.7.0/24 dev wan4 scope link metric 8 
192.168.30.0/24 dev wan1 scope link metric 3 
```
**Section dhcp**
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/ifconfig.co/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/s3.ll.dash.row.aiv-cdn.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/s3-dub.cf.dash.row.aiv-cdn.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/dmqdd6hw24ucf.cloudfront.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/d25xi40x97liuc.cloudfront.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/aiv-delivery.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/aiv-cdn.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/1s3.lvlt.dash.us.aiv-cdn.net.c.footprint.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/s.loris.llnwd.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/atv-ext.amazon.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/c.media-amazon.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/amazon.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/images-amazon.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/amazonaws.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/amazon-adsystem.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/cloudfront.net/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/us-west-2.compute.amazonaws.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.force='1'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_default='1'
dhcp.lan.ra_preference='high'
dhcp.lan.ra_management='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
[... cut ...]
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.metric='1201'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
network.omr6in4.force_link='1'
network.wan3_dev=device
network.wan3_dev.name='wan3'
network.wan3_dev.type='macvlan'
network.wan3_dev.ifname='eth0'
network.wan3_dev.macaddr='92:bb:0c:d8:7a:09'
network.wan4_dev=device
network.wan4_dev.name='wan4'
network.wan4_dev.type='macvlan'
network.wan4_dev.ifname='eth0'

JmRy, 2021-06-04:
Je viens de me rendre compte que d'être obligé d'ajouter le service `amazon` au by-pass, cela force tous les sites ou services qui utilisent un stockage AWS a transiter par le WAN désigné. Du coup c'est assez gênant, car on perds l'agrégation de lignes pour ces sites.

Ysurac, 2021-06-04:
Oui logique, AWS c'est amazon.
Sinon pas de soucis dans ta configuration tout est bon.
Il y a eu quelques corrections à ce niveau dans la 0.58rc*, peut être que ça corrigera le soucis...

JmRy, 2021-06-04:
Ok merci.

Où est-ce que je trouve la liste des domaines/ip que tu utilises dans le service `amazonvideo` ? 

En étudiant les connexions qui sortent de mon réseau, je peux peut-être trouver quel est l'ip/domaine qui me force à utiliser `amazon`

Ysurac, 2021-06-04:
Les IPs sont via SSH dans ```/proc/net/xt_ndpi/ip_proto``` et les hosts dans ```/proc/net/xt_ndpi/host_proto```

JmRy, 2021-06-04:
Je viens de faire une première recherche par rapport à la liste de tes ip/hosts. 

Voici les ip utilisée par mon iPhone lors d'une lecture de l'appli Prime
[image]

Je retrouve pas l'ip qui charge le plus dans tes fichiers ip_proto et host_proto
52.222.149.90 = server-52-222-149-90.cdg52.r.cloudfront.net

Dans ip_proto, tu n'indique que 8.248.79.0/24 pour `amazonvideo` et .r.cloudfront.net n'est pas indiqué dans host_proto. Par contre, je trouve bien 52.222.0.0/16 dans ip_proto avec le label Amazon.

Je me trompe en pensant qu'il est logique que je sois obligé d'ajouter le service [amazon](url) pour ne pas être bloqué ?

Ysurac, 2021-06-04:
Normalement il n'est pas nécessaire, mais tu peux toujours ajouter le domaine "cloudfront.net" dans les domaines à bypasser pour voir.

JmRy, 2021-06-04:
J'ai essayé de trouver une solution pour ne pas devoir utiliser l'omr-bypass amazon, mais je n'arrive à rien :(

Que ce soit en ajoutant toutes les IPs ou les hosts (ex: cloudfront.net, r.cloudfront.net, .cloudfront.net) qui se connecte lors de l'utilisation de Prime, je suis toujours bloqué. 

Concernant les IPs et host listés dans /proc/net/xt_ndpi/, les sous-domaines cloudfront.net et llnwd.net que tu utilises dans host_proto ne sont pas concordants avec mes connexions. Exemple :
95.140.239.85 = https-95-140-239-85.cdg.llnw.net VS .s.loris.llnwd.net
52.222.149.90 =  server-52-222-149-90.cdg52.r.cloudfront.net VS d25xi40x97liuc.cloudfront.net

Bref pour que ça marche, il me faut impérativement activer le service amazon 

J'ai aussi testé la dernière version 058.rc5 sans plus de résultat. A son sujet, c'est normal que ma connexion 4G (Ipv4 seulement) affiche le msg : Multipath semble bloquer sur la connexion

Ysurac, 2021-06-04:
Pour les hosts, en mettant "cloudfront.net" ça bloque "*.cloudfront.net" pas besoin d'autre chose.
Les IPs et hosts viennent de nDPI, pas de moi :)

Pour la 0.58rc5, c'est normal uniquement si Multipath est vraiment bloqué sur la connexion, mais dans ce cas pas d'agrégation avec cette connexion. Il faut vérifier dans Réseau->MPTCP, onglet "Vérification du support MPTCP".

JmRy, 2021-06-04:
Bonne nouvelle :) En ajoutant *.cloudfront.net et *.llnw.net ca fonctionne !

-------

Concernant le support MPTCP, j'ai fait le test. Ca mouline pendant plusieurs minutes sur la ligne 4G avec au final `Result: nil`
_Si c'est un bug et que cela t'intéresse de le résoudre on peut clore ce sujet et en ouvrir un autre pour ne pas mélanger_

Ysurac, 2021-06-04:
Parfait pour amazon vidéo, je vais essayer d'ajouter ces domaines par défaut (bizarre que ça fonctionne sans pour moi).

Pour MPTCP oui je veux bien que tu ouvres un autre sujet, mais j'ai potentiellement une idée de la raison ;)

closed 2021-06-05 by JmRy