Ysurac/openmptcprouter #1871: Memory leak in server?

mathewinwood, 2021-05-31:
## Expected Behavior
No memory leak, mptcp connection remains reliable

## Current Behavior
I've been seeing reliability issues where the connection seems to get wedged. Restarting the VPS often fixes this. Looking the memory usage monitoring (see below for last 30 days), I can see that the memory usage crept up over a period of time until I rebooted. More recently, memory usage has grown much more quickly, meaning I sometimes have to reboot the VPS more than once per day.

Any help greatly appreciated!

[image]

## Steps to Reproduce the Problem

  1. VPS running on Google Cloud
  2. Leave it for a while. Certain types of network traffic *may* trigger it, e.g. large uploads or video calling. Not certain.

## Context (Environment)
I don't think anything has changed on the server recently that could have triggered this (I've not changed anything).

Memory usage right now after rebooting around 5 hours ago:

```
$ ps --sort -rss aux | head
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root       449  1.2 38.8 686624 392876 ?       Ss   14:34   3:20 ss-server --manager-address 127.0.0.1:8839 -f /root/.shadowsocks/.shadowsocks_65101.pid -c /root/.shadowsocks/.shadowsocks_65101.conf --acl /etc/shadowsocks-libev/local.acl -t 1000 -u --fast-open --no-delay --mptcp -6 -s 0.0.0.0
root       974  0.1  4.6 899816 46780 ?        Ssl  14:34   0:32 python3 /usr/local/bin/omr-admin.py
mlvpn      539  0.0  2.8  33948 28912 ?        S    14:34   0:01 mlvpn: mlvpn0 !wan8 !wan7 !wan6 !wan5 !wan4 !wan3 !wan2 !wan1
root       691  0.4  2.1 1057852 21244 ?       Sl   14:34   1:18 /opt/stackdriver/collectd/sbin/stackdriver-collectd -C /etc/stackdriver/collectd.conf -f
root       440  0.0  1.7 109116 17628 ?        Ssl  14:34   0:00 /usr/bin/python3 /usr/share/unattended-upgrades/unattended-upgrade-shutdown --wait-for-signal
root       563  0.0  1.6 4916560 16536 ?       Ssl  14:34   0:00 /usr/bin/v2ray -config /etc/v2ray/v2ray-server.json
root       425  0.0  1.4 119784 14496 ?        Ssl  14:34   0:02 /usr/bin/google_osconfig_agent
root       695  0.0  1.2 114292 12340 ?        Ssl  14:34   0:02 /usr/bin/google_guest_agent
root         1  0.0  0.9  22216  9132 ?        Ss   14:33   0:01 /sbin/init
```

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 5.4.86-mptcp running on google cloud
  - OpenMPTCProuter platform: RPi4

Ysurac, 2021-05-31:
How many memory do you have on VPS ?

mathewinwood, 2021-06-01:
1 GB.

Not sure how much it needs?

Ysurac, 2021-06-01:
It's working with 1GB.
What is the result of ```free -m``` and ```top```, or your ps command, when you have high memory usage ? It's always ss-server that use much memory ?

mathewinwood, 2021-06-02:
Right now (with memory usga eat around 55% having crept up over the last day):

```
$ free -m
              total        used        free      shared  buff/cache   available
Mem:            987         660         189          10         137         184
Swap:             0           0           0
```

and top output:

```
top - 08:53:17 up 16:57,  1 user,  load average: 0.02, 0.05, 0.06
Tasks: 106 total,   1 running, 104 sleeping,   0 stopped,   1 zombie
%Cpu(s):  0.7 us,  1.3 sy,  0.0 ni, 95.8 id,  0.0 wa,  0.0 hi,  2.0 si,  0.2 st
MiB Mem :    987.6 total,    188.9 free,    660.8 used,    137.9 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.    184.2 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                       
 6019 root      20   0    2956    764    672 S   2.0   0.1   1:46.42 glorytun-tcp                                  
  577 root      20   0  559396 383976   2672 S   1.3  38.0  11:55.68 ss-server                                     
   10 root      20   0       0      0      0 I   0.3   0.0   1:30.66 rcu_sched                                     
  316 root      20   0    8080   4764   1640 S   0.3   0.5   0:27.95 haveged                                       
  423 root      20   0  905400  47928  11376 S   0.3   4.7   2:26.83 python3                                       
 4865 root      20   0       0      0      0 I   0.3   0.0   0:01.95 kworker/u4:0-mptcp_wq                         
```

Watching top, `glorytun-tcp` and `ss-server` are usually the top 2, the others vary.

I don't know if it's always `ss-server` consuming the most memory, but the few times I've checked it that's been the case. I'll monitor it.

Thanks for your help!

Adorfer, 2021-06-02:
Reboot end of January 2021 (update to stable version)
Leak not drastic or critical, but definitly visible. 
[image]

Ysurac, 2021-06-02:
I update shadowsocks, maybe this will solve the issue.

closed 2021-09-06 by github-actions[bot]