Ysurac/openmptcprouter #1874: VPN in?

hairy-dog, 2021-06-01:
## Expected Behavior
I would like to connect a different network to my network , which is using OMR

## Current Behavior
I have OMR running well, combining several different internet connections.

I shall have a different building with a fast enough internet connection, not using OMR.

I would like to connect the two networks together using a VPN. I wonder if some of the work is already done in OMR, but I don't know.

## Specifications

  - OpenMPTCProuter version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter VPS version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.1

Ysurac, 2021-06-01:
Several way, but the best is to install/configure VPN on the VPS side, then only some routes are needed.

hairy-dog, 2021-06-01:
Is there an idiot's guide about how to do that? I don't want to break my working system!

Ysurac, 2021-06-01:
No there is not. Wireguard is installed on VPS, you can use it but use another interface as wg0 used by OpenMPTCProuter.

hairy-dog, 2021-06-01:
OK, thank you.

Ysurac, 2021-06-01:
You can also use OpenVPN that is also installed by default, you only need to not use tun0 and tun1 as interface.

closed 2021-06-02 by hairy-dog