Ysurac/openmptcprouter #1875: Bloquer MAC sur le LA

JeedHome44, 2021-06-02:
Bonjour,

J'ai décidé de prendre un routeur Xiaomi 4C dédié à faire un wifi invité. Je lui ai donc laissé son DHCP activé mais j'arrive quand même à aller partout sur le LAN en étant sur le wifi invité. Le LAN me reconnait avec l'IP 192.168.100.240 du routeur 4C.

Je souhaite donc mettre une règle sur OMR pour autoriser l'adresse MAC du routeur 4C pour aller sur internet mais pas sur le LAN.

On peut faire ça?
Si oui, quelle option?

JeedHome44, 2021-06-03:
Je réouvre car je pensais avoir réussi mais non.

Ysurac, 2021-06-03:
Ce que je ferais c'est créé une autre interface dans OpenMPTCProuter, basée sur eth0 avec une IP dans un autre sous réseau, par exemple 192.168.101.1 et la mettant dans une autre zone dans le firewall similaire à la zone LAN.

JeedHome44, 2021-06-03:
Est-ce possible de faire des screenshot de ce qu'il faut faire? Je ne vois pas trop comment y arriver.

Mais les autres appareils ne vont pas se mettre sur le 101,X?

La je veux juste bloquer sur le LAN l'ip 192.168.100.214

Ysurac, 2021-06-03:
En mettant le DHCP sur l'interface wifi du routeur wifi non, ou alors en utilisant des VLANs, mais faut avoir du matos qui gère ça.
La actuellement comme tout est dans le même réseau, ça ne passe même pas par le routeur.

closed 2021-09-06 by github-actions[bot]