Ysurac/openmptcprouter #1876: Why VPS has established connection to Russia?

wicadmin, 2021-06-02:
I see the connections from v2ray. This is a new server with just the OMR VPS script ran on it.

```
root@omr:~# netstat -aon | grep -i establish
Active Internet connections (servers and established)
tcp        0      0 <redact>:65400    185.202.2.36:26581      ESTABLISHED off (0.00/0/0)

Active UNIX domain sockets (servers and established)
root@omr2:~# whois 185.202.2.36
% This is the RIPE Database query service.
% The objects are in RPSL format.
%
% The RIPE Database is subject to Terms and Conditions.
% See http://www.ripe.net/db/support/db-terms-conditions.pdf

% Note: this output has been filtered.
%       To receive output for a database update, use the "-B" flag.

% Information related to '185.202.2.0 - 185.202.2.255'

% Abuse contact for '185.202.2.0 - 185.202.2.255' is 'abuse@zservers.ru'

inetnum:        185.202.2.0 - 185.202.2.255
netname:        xweb-ltd
org:            ORG-XA152-RIPE
country:        NL
admin-c:        OD1924-RIPE
tech-c:         OD1924-RIPE
status:         ASSIGNED PA
mnt-by:         FOXLABMNT
created:        2020-02-12T14:24:55Z
last-modified:  2020-05-04T16:42:50Z
source:         RIPE

organisation:   ORG-XA152-RIPE
org-name:       xWEBltd
org-type:       OTHER
address:        Moscow, pos. Sosenskoe, ul. Servantesa 1/3 kv. 125
abuse-c:        ZA1764-RIPE
mnt-ref:        mnt-wellweb
mnt-ref:        FOXLABMNT
mnt-ref:        ru-avm-1-mnt
mnt-by:         mnt-wellweb
mnt-by:         mnt-xweb
created:        2020-05-04T13:22:01Z
last-modified:  2020-05-04T20:32:44Z
source:         RIPE # Filtered

person:         Oleinichenko Denis
address:        Netherland
phone:          +7 925 584 32 84
nic-hdl:        OD1924-RIPE
mnt-by:         mnt-xweb
created:        2018-12-06T20:37:56Z
last-modified:  2020-02-07T13:39:38Z
source:         RIPE

% Information related to '185.202.2.0/24AS44477'

route:          185.202.2.0/24
origin:         AS44477
mnt-by:         FOXLABMNT
created:        2020-05-04T10:35:11Z
last-modified:  2020-05-04T10:35:11Z
source:         RIPE

```

/tmp/v2rayError.log
```
2021/04/19 03:32:56 [Warning] v2ray.com/core: V2Ray 4.31.0 started
2021/05/01 00:03:47 [Info] [707656808] v2ray.com/core/proxy/vless/inbound: firstLen = 0
2021/05/01 00:03:47 [Warning] [707656808] v2ray.com/core/app/proxyman/inbound: connection ends > v2ray.com/core/proxy/vless/inbound: invalid request from 185.202.2.32:45222 > v2ray.com/core/proxy/vless/encoding: failed to read request version > read tcp <redact>:65228->185.202.2.32:45222: read: connection reset by peer
2021/05/01 00:03:48 [Info] [3103994199] v2ray.com/core/proxy/vless/inbound: firstLen = 0
2021/05/01 00:03:48 [Warning] [3103994199] v2ray.com/core/app/proxyman/inbound: connection ends > v2ray.com/core/proxy/vless/inbound: invalid request from 185.202.2.32:45401 > v2ray.com/core/proxy/vless/encoding: failed to read request version > tls: first record does not look like a TLS handshake
2021/05/01 00:03:48 [Info] [3909004645] v2ray.com/core/proxy/vless/inbound: firstLen = 0
2021/05/01 00:03:48 [Warning] [3909004645] v2ray.com/core/app/proxyman/inbound: connection ends > v2ray.com/core/proxy/vless/inbound: invalid request from 185.202.2.32:47283 > v2ray.com/core/proxy/vless/encoding: failed to read request version > tls: first record does not look like a TLS handshake
2021/06/02 21:25:48 [Debug] [899120360] v2ray.com/core/proxy/dokodemo: processing connection from: 127.0.0.1:49247
2021/06/02 21:25:48 [Info] [899120360] v2ray.com/core/proxy/dokodemo: received request for 127.0.0.1:49247
2021/06/02 21:25:48 [Info] [899120360] v2ray.com/core/app/dispatcher: taking detour [api] for [tcp:127.0.0.1:0]
2021/06/02 21:25:48 [Info] [899120360] v2ray.com/core/app/proxyman/inbound: connection ends > v2ray.com/core/proxy/dokodemo: connection ends > context canceled
2021/06/02 21:26:28 [Info] [3228253560] v2ray.com/core/proxy/vless/inbound: firstLen = 0
2021/06/02 21:26:28 [Warning] [3228253560] v2ray.com/core/app/proxyman/inbound: connection ends > v2ray.com/core/proxy/vless/inbound: invalid request from 127.0.0.1:28367 > v2ray.com/core/proxy/vless/encoding: failed to read request version > tls: first record does not look like a TLS handshake

```

```
root@omr:~# netstat -aon | grep -i listen
tcp        0      0 127.0.0.1:10085         0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        0      0 0.0.0.0:65222           0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        0      0 0.0.0.0:65001           0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        0      0 0.0.0.0:65228           0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        0      0 0.0.0.0:65101           0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        0      0 0.0.0.0:65301           0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        2      0 0.0.0.0:65400           0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        0      0 0.0.0.0:65401           0.0.0.0:*               LISTEN      off (0.00/0/0)
tcp        0      0 0.0.0.0:65500           0.0.0.0:*               LISTEN      off (0.00/0/0)
```

Ysurac, 2021-06-03:
At port 65400 it's not v2ray but iperf3 that is running.
I think the ip scan all servers ports but every app running need an authentication, so nothing really special. You can install fail2ban on the server if you want.

gnosoz, 2021-06-03:
If you need you can find an example of fail2ban script here below which includes a portscan plus ssh check and ban 

```
#!/bin/bash
apt install fail2ban -y
echo '[Definition]
failregex = DROP.* SRC=<HOST>
ignoreregex =' | sudo tee /etc/fail2ban/filter.d/portscan.conf

echo "
[DEFAULT]
ignoreip = 

[sshd]
port    = 65222
maxretry = 1
action = shorewall
bantime = -1

[portscan]
enabled  = true
filter   = portscan
logpath  = /var/log/kern.log
action   = shorewall
maxretry = 7
findtime = 13600
bantime  = 72H
ignoreip = 
" > /etc/fail2ban/jail.local
```

closed 2021-09-06 by github-actions[bot]