Ysurac/openmptcprouter #1877: Starlink WAN marked as "Gateway DOWN" when it is actually up

Kurocon, 2021-06-03:
Hey all, first of all thanks for the great piece of software, it has been really helping me out with my network.

I have a small nuisance where one of my WAN connections (Starlink) has been showing a "Gateway offline" notice ever since I connected it. Everything is working fine, so no worries there, I'm just looking for a way to get rid of that notice. Is there a setting on the interface I should tweak, or some gateway address I could configure so this will show that the connection is working properly? (and also allow me to view the latency, which I presume is missing now because it thinks it is down)

## Expected Behavior
All WAN connections show a green checkmark when they are actually all working correctly.

## Current Behavior
One WAN connection (Starlink) is showing a yellow exclamation mark and the message "Gateway DOWN"

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: v0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: Virtualized (QEMU KVM on Debian Linux)

Example image
[image]

Ysurac, 2021-06-03:
If the gateway is really not pingable, no issue. You can remove warning by checking "Disable gateway ping" in System->OpenMPTCProuter, "Advanced settings" tab.

Kurocon, 2021-06-03:
I see, thanks! Is it possible to disable it only for that particular connection, or is it only a system-wide toggle?

Ysurac, 2021-06-03:
It's for all interfaces but only for status page display. It's still enabled for interfaces monitoring.

Adorfer, 2021-06-04:
I heard that ICMP-ping on ipv6 is some kind of strange/modified on Starlink. This should not matter here, since VPS are entered as ipv4, not as FQDN (which could be dual stack, but even there i would assume, OMR only goes for ipv4 for the ping test.)

Kurocon, 2021-06-04:
Thanks for the input, I disabled the checks for now. If its a general setting only for the status page I would suggest splitting it up per WAN interface, so they can still be used on the interfaces that do support it, but I can understand that it is not a very high priority issue/request. For me personally I will close the issue, as the problem is solved for my case.

Thanks for the help, good luck with the rest of the project!

closed 2021-06-04 by Kurocon