Ysurac/openmptcprouter #1880: Big Time Difference between server and router & Ping Delay

satmarin, 2021-06-04:
Hi I have 2 questions:

1) Ping Delay
Due to the mobile nature of the 4G WAN connections used we sometimes get ping delay > 1sec.  
Is it correct that the gateway is not used when ping delay is > than 1 sec. as indicated in the OpenMPTCP status page.  And if so where can this delay be extended to 2 seconds?

2) Big Time Difference
We get message in openMPTCP status that there is a big time difference between VPS and router but server is on CEST (Brussels) and router is on Europe/Brussels. What is considered 'big time difference' and does it affect the links in any way?

Thanks
Michaël

  - OpenMPTCProuter version: openmptcprouter v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: Version 0.1024 5.4.65-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-06-04:
1. It's only a warning in status page, so connection is still used. There is options to disable warning display in "Advanced settings" tab
2. Time diff is calculated on UTC, so if the message is displayed (when diff > 10s) then there is a problem and date/time must be fixed

satmarin, 2021-06-04:
Thanks you for the fast response.  
1) OK
2) does it affect the connection when time difference is > 10sec but < 2 minutes ?  Maybe message could also indicate the time on both devices which might make it easier to troubleshoot.

Ysurac, 2021-06-04:
Yes it affect the connection. I don't want to put too much data on status page ;)

satmarin, 2021-06-04:
The router updates the time according to ntpdate correctly.  I am trying to update the VPS but it doesn't seem to have ntpdate.  What would be the best way to time sync the VPS?

Ysurac, 2021-06-04:
I often use chrony to sync time on VPS.

satmarin, 2021-06-04:
Excellent, worked perfectly.  Thank you Yannick

closed 2021-09-08 by github-actions[bot]