Ysurac/openmptcprouter #1881: 0.58rc5: Warning Multipath block

JmRy, 2021-06-05:
## Expected Behavior
No alert message indicating an MPTCP block

## Current Behavior
4 WAN's (3 Adsl + 4G)
IPv6 is enabled

[image]

Despite the alert, the multipath is not blocked. The lines aggregation works
[image]

MPTCP Check
```
1: 192.168.7.1 IP::DiffServicesCP IP::CheckSum 
2: * 
3: 10.4.1.20  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
4: 10.187.162.252  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
5: 77.136.134.13  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
6: * 
7: * 
...
63: * 
64: * 

Result:
nil

```
## Specifications

  - OpenMPTCProuter version: 0.58rc5
  - OpenMPTCProuter VPS version: Linux otb 5.4.86-mptcp
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.2
 
  - Wan Hardware  
     TP-Link Archer MR600 v1 router 
     4G SFR Red with LTE profil SFR(par défaut:1) Type de PDP: IPv4 APN: websfr 
     Tunnel ipV6 disabled

Ysurac, 2021-06-05:
SFR always reported this error on all detection tests I made, even if I use another way for test in 0.58...
I will check if I can reproduce this with my Prixtel SIM on SFR network.

johncoles, 2021-07-15:
I have a feeling I am seeing this bug on two connections as well. I spent about 4 hours trying to figure out why connections that worked fine 2 weeks ago were showing the error 😄 

I'm not able to test this currently but will be on site with the device on Friday, is there a good way to test this out and provide any useful data for you?

Ysurac, 2021-07-15:
It's only a warning. For no there no option to disable it, I will add that.
Providers are using CGNAT ?

johncoles, 2021-07-15:
> Providers are using CGNAT ?

I believe they are, yes.
So it won’t affect the actual operation? OMR will still route traffic over the connections showing that warning?

Ysurac, 2021-07-15:
This only mean I can't detect if a middle box block MPTCP or not.
So, this doesn't affect the actual operation at all. It's a test only done by status page.