Ysurac/openmptcprouter #1882: Issues with throughput draws all the data via one modem instead of combining throughput

Leaptear, 2021-06-05:
## Expected Behavior

Expected behaviour is to combine both connections and bond them to achieve higher throughput

## Current Behavior

Only gets data from one of the modems and the other sits idle when they should be both combined

## Specifications

  - OpenMPTCProuter version:  Latest Version 
  - OpenMPTCProuter VPS version:  Latest Version 
  - OpenMPTCProuter platform:  RPI4
[image]

Ysurac, 2021-06-05:
Did you check in Network->MPTCP that MPTCP is supported by both connections ?
Did you test via SSH with ```omr-test-speed``` aggregated speed and with ```omr-test-speed wan2``` (and all physical interface) speed of each connection ?
What is the latency of each connection (available in status page) ?

Leaptear, 2021-06-06:
Hi @Ysurac I have seemed to fix it albeit I am not sure if it's the right way, If I turn off the proxy and make it all go down the VPN tunnel via Glorytun UDP it bonds the connections and works fine it seems like the proxy side is a bit iffy? or maybe I need to somehow make it send my video stream via the VPN instead of the proxy but there again id rather have everything bonded including the devices connected via the wifi hotspot so not sure, would be really interested to hear your suggestions, I have done the two steps you listed and they both showed only 2000kb/s but much much more when on VPN only thanks.

Leaptear, 2021-06-07:
bit of an update @Ysurac  It's bonding when doing a speedtest but when livestreaming it only goes via one modem and not bonding both? Weird.... any ideas?

Ysurac, 2021-06-07:
livestreaming in upload or download side ? You use UDP or TCP ?

Leaptear, 2021-06-07:
> livestreaming in upload or download side ? You use UDP or TCP ?

Im using Glorytun UDP And It's not doing much on the download side when livestreaming, but on the upload side It's only going via the one modem, Im streaming to an RTMP server. for example if im streaming it goes via one modem but if I instantly start a speed test on the phone i can see it bonding the speed test between both modems but livestreaming is still going via one which is producing connection issues cause it can't get enough bandwidth

Ysurac, 2021-06-07:
I ask about the protocol stream you use. RTMP should be over TCP I think.
Where is the RTMP server ? on the VPS or elsewhere ?

Leaptear, 2021-06-07:
> I ask about the protocol stream you use. RTMP should be over TCP I think.
> Where is the RTMP server ? on the VPS or elsewhere ?

Hi It's hosted on a local machine on the network the VPS is also hosted on the same network, If i enable TCP the bonding ceases to operate at all, It just defaults to the main modem while the other modem floats at 30-40 kbps and doesn't bond even with speedtests

Leaptear, 2021-06-07:
Is there a way to allow me to access the local network that the VPN is tunnelling to? so I can just stream to the local IP on the network? to see if that might fix it? @Ysurac

closed 2021-09-10 by github-actions[bot]