Ysurac/openmptcprouter #1883: Activation IPV6 sur les box ?

JmRy, 2021-06-05:
Bonjour,

Sachant qu'IPv6 est activé sur mon VPS et que ce protocole est fonctionnel sur mon réseau local, y-a-t-il un intérêt a activer l'IPv6 sur les Box ?

Merci de vos réponses

Cordialement

Ysurac, 2021-06-05:
La 0.57 ne supporte l'IPv6 sur les WANs.
La 0.58 supporte IPv6 sur les WANs et VPS, mais il n'y a pas forcément d'intérêt à passer en full IPv6 si il n'y a pas un gain de vitesse.

JmRy, 2021-06-06:
> La 0.57 ne supporte l'IPv6 sur les WANs. >> _**que ou pas** ?_

Mes box me fournissant une IPv6, j'ai donc fait un essai guère concluant malheureusement. Bon faut dire aussi que l'IPv6 n'est pas spécialité ;)

Donc voici ce que me fourni ma box SFR
[image]

Sachant que l'adresse Mac de la box est e0:a1:d7:4a:d5:40 et celle de l'interface WAN 7A:20:A6:43:AB:C5, si j'indique comme 
adresse IP xxxx:xxxx:xxxx:3e00:7820:a6ff:fe43:abc5 et passerelle xxx:xxxx:xxxx:3e00:e2a1:d7ff:fe4a:d540 j'obtiens ceci
[image]
[image]
Résultat plus d'IPv6 sur le routeur et le VPS. le WAN est accessible. Sous SSH du routeur l'IP ping mais pas la passerelle.

Si j'indique comme adresse IP fe80::230:7820:a6ff:fe43:abc5 et passerelle fe80::230:88ff:fe04:349a j'obtiens
[image]
Résultat l'IPv6 est fonctionnel sur le routeur et le VPS, mais le WAN est inaccessible.

Est-ce moi qui bug dans ce je saisis comme paramètres ou il y a un autre problème ?

Ysurac, 2021-06-06:
@JmRy la 0.57 ne supporte pas l'IPv6 sur les WANs (ni que le VPS soit en IPv6). Sur la  0.58 les WANs peuvent avoir une IPv6 et le VPS doit dans ce cas être configuré avec l'IPv6 (et l'IPv4 si il y a des WANs en IPv4)

closed 2021-09-05 by JmRy