Ysurac/openmptcprouter #1885: Problem with between VPS and Raspi

Bug-Storm, 2021-06-06:
## Expected Behavior

## Current Behavior

Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13982]: listening at 0.0.0.0:1100
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13981]: tcp port reuse enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13981]: UDP relay enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13982]: tcp port reuse enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13982]: UDP relay enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13981]: udp port reuse enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13982]: udp port reuse enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13981]: running from root user
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13982]: running from root user
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13984]: tcp port reuse enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13984]: UDP relay enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13984]: udp port reuse enabled
Tue Dec 15 19:45:32 2020 daemon.info /usr/bin/ss-redir[13984]: running from root user
Tue Dec 15 19:45:33 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (wan2)
Tue Dec 15 19:45:34 2020 user.notice unbound: default memory configuration
Tue Dec 15 19:45:34 2020 daemon.notice unbound: [14630:0] notice: init module 0: iterator
Tue Dec 15 19:45:34 2020 daemon.info unbound: [14630:0] info: start of service (unbound 1.13.0).
Tue Dec 15 19:45:34 2020 user.notice mptcp: Reloading mptcp config due to ifup of loopback (lo)
Tue Dec 15 19:45:34 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 198.27.92.1)
Tue Dec 15 19:45:34 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Tue Dec 15 19:45:35 2020 daemon.notice ttyd[15527]: ttyd 1.6.1 (libwebsockets 3.1.0)
Tue Dec 15 19:45:35 2020 daemon.notice ttyd[15527]: tty configuration:
Tue Dec 15 19:45:35 2020 daemon.notice ttyd[15527]:   start command: /usr/libexec/login.sh
Tue Dec 15 19:45:35 2020 daemon.notice ttyd[15527]:   close signal: SIGHUP (1)
Tue Dec 15 19:45:35 2020 daemon.notice ttyd[15527]:   terminal type: xterm-256color
Tue Dec 15 19:45:35 2020 daemon.notice ttyd[15527]:  Using foreign event loop...
Tue Dec 15 19:45:35 2020 daemon.notice ttyd[15527]: Creating Vhost 'default' port 7681, 2 protocols, IPv6 off
Tue Dec 15 19:45:35 2020 daemon.notice ttyd[15527]:  Listening on port: 7681
Tue Dec 15 19:45:35 2020 daemon.notice procd: /etc/rc.d/S99v2ray: Flush terminated
Tue Dec 15 19:45:36 2020 daemon.info procd: - init complete -
Sun Jun  6 16:13:04 2021 daemon.notice ttyd[15527]: time discontiguity: at old time 1608061537s, new time 1622995984s: +14934447s
Sun Jun  6 16:13:10 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jun  6 16:13:10 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jun  6 16:13:11 2021 daemon.err omr-tracker-ss[10193]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sun Jun  6 16:13:11 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sun Jun  6 16:13:12 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jun  6 16:13:12 2021 daemon.info dnsmasq[13914]: read /etc/hosts - 4 addresses
Sun Jun  6 16:13:12 2021 daemon.info dnsmasq[13914]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Jun  6 16:13:12 2021 daemon.info dnsmasq-dhcp[13914]: read /etc/ethers - 0 addresses
Sun Jun  6 16:13:12 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jun  6 16:13:12 2021 daemon.warn dnsmasq[13914]: failed to create listening socket for 10.255.254.2: Address not available
Sun Jun  6 16:13:12 2021 daemon.debug dnsmasq[13914]: listening on tun0(#19): 10.255.254.2 port 53
Sun Jun  6 16:13:12 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Jun  6 16:13:13 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jun  6 16:13:13 2021 daemon.info dnsmasq[13914]: read /etc/hosts - 4 addresses
Sun Jun  6 16:13:13 2021 daemon.info dnsmasq[13914]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Jun  6 16:13:13 2021 daemon.info dnsmasq-dhcp[13914]: read /etc/ethers - 0 addresses
Sun Jun  6 16:13:13 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jun  6 16:13:25 2021 cron.err crond[3409]: time disparity of 248908 minutes detected
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info: service stopped (unbound 1.13.0).
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info: server stats for thread 0: 51 queries, 1 answers from cache, 50 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info: server stats for thread 0: requestlist max 15 avg 4.08 exceeded 0 jostled 0
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info: average recursion processing time 0.423361 sec
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info: histogram of recursion processing times
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info: [25%]=0.120149 median[50%]=0.241979 [75%]=0.643216
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info: lower(secs) upper(secs) recursions
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info:    0.008192    0.016384 1
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info:    0.032768    0.065536 4
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info:    0.065536    0.131072 9
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info:    0.131072    0.262144 13
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info:    0.262144    0.524288 8
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info:    0.524288    1.000000 10
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info:    1.000000    2.000000 4
Sun Jun  6 16:13:36 2021 daemon.info unbound: [14630:0] info:    2.000000    4.000000 1
Sun Jun  6 16:13:36 2021 daemon.notice unbound: [20019:0] notice: init module 0: iterator
Sun Jun  6 16:13:36 2021 daemon.info unbound: [20019:0] info: start of service (unbound 1.13.0).
[... 568 lines cut ...]
ou
"Impossible de contacter le script d'administration du Serveur"

je ne sais plus quoi faire ..

J'ai bien une 4G stable car j'ai l'antenne relais a 500m d'ici puis mon ADSL est aussi stable .

I have a little problem with the OpenMPTCP 

I have one WAN ADSL and The other in 4G.

I can not have a stable connection between the router and the VPS

I have already tried to have several things but still the same concern 

Even when I have both connection enabled, the speed of download / Upload is optimal, it is as if I had a wan in backup or desactivated  

Whether in 4G or ADSL I always have a loss of connection to the VPS
"No response from the Server"
Yes
"Unable to contact the server administration script"

I don't know what to do anymore ..

I have a stable 4G because I have the relay antenna 500m from here and then my ADSL is also stable.

Bug-Storm, 2021-06-06:
Un test de vitesse  j'ai ça 
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 56
host: speedtest.tokyo2.linode.com - ping: 289
host: speedtest.singapore.linode.com - ping: 215
host: speedtest.newark.linode.com - ping: 139
host: speedtest.atlanta.linode.com - ping: 144
host: speedtest.dallas.linode.com - ping: 173
host: speedtest.fremont.linode.com - ping: 177
host: speedtest.tele2.net - ping: 66
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 56
host: ipv4.bouygues.testdebit.info - ping: 54
host: par.download.datapacket.com - ping: 54
host: nyc.download.datapacket.com - ping: 106
host: ams.download.datapacket.com - ping: 80
host: fra.download.datapacket.com - ping: 60
host: lon.download.datapacket.com - ping: 73
host: mad.download.datapacket.com - ping: 190
host: prg.download.datapacket.com - ping: 70
host: sto.download.datapacket.com - ping: 99
host: vie.download.datapacket.com - ping: 75
host: war.download.datapacket.com - ping: 80
host: atl.download.datapacket.com - ping: 176
host: chi.download.datapacket.com - ping: 139
host: lax.download.datapacket.com - ping: 227
host: mia.download.datapacket.com - ping: 235
host: nyc.download.datapacket.com - ping: 128
Best server is http://ipv4.bouygues.testdebit.info/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 69.1M    0     0  1069k      0  2:32:10  0:01:06  2:31:04  937k^X

wan1

Select best test server...
host: speedtest.frankfurt.linode.com - ping: 44
host: speedtest.tokyo2.linode.com - ping: 290
host: speedtest.singapore.linode.com - ping: 287
host: speedtest.newark.linode.com - ping: 128
host: speedtest.atlanta.linode.com - ping: 140
host: speedtest.dallas.linode.com - ping: 144
host: speedtest.fremont.linode.com - ping: 173
host: speedtest.tele2.net - ping: 40
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 41
host: ipv4.bouygues.testdebit.info - ping: 26
host: par.download.datapacket.com - ping: 33
host: nyc.download.datapacket.com - ping: 108
host: ams.download.datapacket.com - ping: 37
host: fra.download.datapacket.com - ping: 40
host: lon.download.datapacket.com - ping: 35
host: mad.download.datapacket.com - ping: 38
host: prg.download.datapacket.com - ping: 51
host: sto.download.datapacket.com - ping: 57
host: vie.download.datapacket.com - ping: 43
host: war.download.datapacket.com - ping: 60
host: atl.download.datapacket.com - ping: 116
host: chi.download.datapacket.com - ping: 120
host: lax.download.datapacket.com - ping: 165
host: mia.download.datapacket.com - ping: 129
host: nyc.download.datapacket.com - ping:
Best server is http://ipv4.bouygues.testdebit.info/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 2663k    0     0  22333      0   5d 04h  0:02:02   5d 04h  1890^

wan2

Select best test server...
host: speedtest.frankfurt.linode.com - ping: 62
host: speedtest.tokyo2.linode.com - ping: 267
host: speedtest.singapore.linode.com - ping: 315
host: speedtest.newark.linode.com - ping: 149
host: speedtest.atlanta.linode.com - ping: 154
host: speedtest.dallas.linode.com - ping: 160
host: speedtest.fremont.linode.com - ping: 178
host: speedtest.tele2.net - ping: 55
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 80
host: ipv4.bouygues.testdebit.info - ping: 39
host: par.download.datapacket.com - ping: 56
host: nyc.download.datapacket.com - ping: 140
host: ams.download.datapacket.com - ping: 61
host: fra.download.datapacket.com - ping: 64
host: lon.download.datapacket.com - ping: 61
host: mad.download.datapacket.com - ping: 62
host: prg.download.datapacket.com - ping: 74
host: sto.download.datapacket.com - ping: 93
host: vie.download.datapacket.com - ping: 66
host: war.download.datapacket.com - ping: 83
host: atl.download.datapacket.com - ping: 151
host: chi.download.datapacket.com - ping: 133
host: lax.download.datapacket.com - ping: 185
host: mia.download.datapacket.com - ping: 152
host: nyc.download.datapacket.com - ping: 122
Best server is http://ipv4.bouygues.testdebit.info/10G.iso, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 9536M    1 98.9M    0     0  1009k      0  2:41:14  0:01:40  2:39:34  801k^X

Ysurac, 2021-06-06:
You should use the template, I need OpenMPTCProuter version and arch used (RPI2,3 or 4) ?
I see 2711, so this seems to be a RPI4 but with how much memory ? I see that you use V2Ray, v2ray use more CPU and memory and this seems to be a problem in your case. You can also disable tracebox check, in advanced parameters tab, that seems to use too much memory.

Adorfer, 2021-06-06:
basically: the RPI is running out of RAM. Either due to beeing a 1MO model or because the config is done "to consume a lot of RAM".

Bug-Storm, 2021-06-06:
> You should use the template, I need OpenMPTCProuter version and arch used (RPI2,3 or 4) ?
> I see 2711, so this seems to be a RPI4 but with how much memory ? I see that you use V2Ray, v2ray use more [...]

OHH sorry it's bug because i paste the log 

v0.57.3 r0+15225-bfc433efd4(i try x64 and same problem) Raspi4B 2Gb
VPS OVH vps-f96ddb78 5.4.86-mptcp

Bug-Storm, 2021-06-06:
> basically: the RPI is running out of RAM. Either due to beeing a 1MO model or because the config is done "to consume a lot of RAM". From "BCM2711" it might be a RPI4. So perhaps there is something in the config hogging RAM. other installed packages?

Now i've the opentmptcp clean image 

I try to set basic config, and i've the same problem

Ysurac, 2021-06-06:
Check what is using so much memory using for example htop via SSH.
Do you install any kernel module or app ?
Do you do any other configuration than wan ?

Bug-Storm, 2021-06-06:
[image]

I've the basic config , with wan 1 master and wan 2 on 

Donc j'ai viens de re flasher l'image , j'ai donc une config basique , rien qu'un wan 1 et wan 2 configuré voilà 

rien de plus installer

Ysurac, 2021-06-06:
Here there is no memory issue, so you shouldn't have same problems in log.

Bug-Storm, 2021-06-06:
[image]
It's normal ?? Max time ??

i try test my Connexion and i've this on log 

Sun Jun  6 19:33:19 2021 user.notice Shadowsocks: Rules DOWN
Sun Jun  6 19:33:20 2021 user.notice Shadowsocks: Rules UP
Sun Jun  6 19:33:20 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Jun  6 19:33:20 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jun  6 19:33:22 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jun  6 19:33:22 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[22706]: exiting on receipt of SIGTERM
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: started, version 2.82 cachesize 150
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: DNS service limited to local subnets
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq-dhcp[11247]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using only locally-known addresses for domain test
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using only locally-known addresses for domain onion
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using only locally-known addresses for domain localhost
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using only locally-known addresses for domain local
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using only locally-known addresses for domain invalid
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using only locally-known addresses for domain bind
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using only locally-known addresses for domain use-application-dns.net
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using only locally-known addresses for domain lan
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using nameserver 127.0.0.1#5353
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: using 1 more local addresses
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: read /etc/hosts - 4 addresses
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq[11247]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Jun  6 19:33:22 2021 daemon.info dnsmasq-dhcp[11247]: read /etc/ethers - 0 addresses
Sun Jun  6 19:33:22 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-local[11344]: using tcp fast open
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-local[11344]: enable TCP no-delay
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-local[11344]: initializing ciphers... chacha20-ietf-poly1305
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-local[11344]: listening at 127.0.0.1:1111
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-local[11344]: tcp port reuse enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-local[11344]: udprelay enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-local[11344]: udp port reuse enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-local[11344]: running from root user
Sun Jun  6 19:33:22 2021 user.info : using tcp fast open
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11348]: enable TCP no-delay
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11348]: initializing ciphers... chacha20-ietf-poly1305
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11348]: listening at 0.0.0.0:1101
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11348]: tcp port reuse enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11348]: UDP relay enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11348]: udp port reuse enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11348]: running from root user
Sun Jun  6 19:33:22 2021 user.info : using tcp fast open
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11346]: enable TCP no-delay
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11346]: resolving hostname to IPv6 address first
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11346]: initializing ciphers... chacha20-ietf-poly1305
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11346]: listening at 0.0.0.0:1100
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11346]: tcp port reuse enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11346]: UDP relay enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11346]: udp port reuse enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11346]: running from root user
Sun Jun  6 19:33:22 2021 user.info : using tcp fast open
Sun Jun  6 19:33:22 2021 user.info : using tcp fast open
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11347]: enable TCP no-delay
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11347]: initializing ciphers... chacha20-ietf-poly1305
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11347]: listening at 0.0.0.0:1101
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11347]: tcp port reuse enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11347]: UDP relay enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11347]: udp port reuse enabled
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11347]: running from root user
Sun Jun  6 19:33:22 2021 daemon.info /usr/bin/ss-redir[11345]: enable TCP no-delay
[... cut ...]
Sun Jun  6 19:33:55 2021 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Jun  6 19:33:55 2021 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Jun  6 19:33:55 2021 daemon.err glorytun-udp[25007]: running on device tun0 as pid 25007
Sun Jun  6 19:33:56 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Jun  6 19:34:00 2021 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.1.2 dev wan1 weight 2 nexthop via 192.168.1.1 dev wan2 weight 1

Ysurac, 2021-06-06:
VPN is not stable, but why do you use Glorytun UDP ? You have issues with default configuration ?

Bug-Storm, 2021-06-06:
> VPN is not stable, but why do you use Glorytun UDP ? You have issues with default configuration ?

Because i use with the ShadowSocks 
I changed after do you have speak  :  v2ray use more CPU and memory

Ysurac, 2021-06-06:
Default is Glorytun TCP.
As I asked, did you have issues with default configuration or only with your changes ?

Bug-Storm, 2021-06-06:
> Default is Glorytun TCP.
> As I asked, did you have issues with default configuration or only with your changes ?

with default config 

i just try fast.com or speed test or nperf and vpn or vps is down

Ysurac, 2021-06-06:
ok. Did you activate SQM ?

Bug-Storm, 2021-06-06:
> ok. Did you activate SQM ?

not , i''ve basic config 
set wan 1 and wan 2 and serveur IP 

No port redirection , no set bypass , nothing

Ysurac, 2021-06-06:
Activate SQM for each wan in the wizard and set download & upload speed.

Bug-Storm, 2021-06-06:
> Activez SQM pour chaque wan dans l'assistant et définissez la vitesse de téléchargement et de téléchargement.

Same problem , VPS is Down afte download test :/ 

I think my problem is with my VPS 
maybe it not stable ?

Ysurac, 2021-06-06:
Maybe, you can check on the VPS ```/var/log/dameon.log``` and ```dmesg``` if you have errors.
But in most case SQM should work, you may need to lower defined speeds.

Bug-Storm, 2021-06-06:
On daemon.log

Jun  6 18:15:01 vps-f96ddb78 ss-server[17188]: remote recv: Connection reset by                                                                                                                                                              peer
Jun  6 18:15:28 vps-f96ddb78 ss-server[17188]: server_recv_send: Connection rese                                                                                                                                                             t by peer
Jun  6 18:15:42 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:16:24 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:17:05 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:17:46 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:18:28 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:19:10 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:19:51 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:20:18 vps-f96ddb78 ss-server[17188]: remote recv: Connection reset by                                                                                                                                                              peer
Jun  6 18:20:19 vps-f96ddb78 ss-server[17188]: remote recv: Connection reset by                                                                                                                                                              peer
Jun  6 18:20:22 vps-f96ddb78 ss-server[17188]: remote recv: Connection reset by                                                                                                                                                              peer
Jun  6 18:20:32 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:21:14 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i                                                                                                                                                             s not connected
Jun  6 18:21:38 vps-f96ddb78 systemd[1]: Started Session 357 of user debian.
Jun  6 18:21:55 vps-f96ddb78 ss-server[17188]: getpeername: Transport endpoint i

Bug-Storm, 2021-06-06:
Not errors on dmesg

[1156803.363966] Shorewall:net-fw:DROP:IN=eth0 OUT= MAC=fa:16:3e:dc:1a:b5xxxxxxx:08:00 SRC=51.91.223.35 DST=193.7xxx LEN=52 TOS=0x00 PREC=0x00 TTL=56 ID=33482 DF PROTO=TCP SPT=49569 DPT=25565 WINDOW=64240 RES=0x00 SYN URGP=0

Bug-Storm, 2021-06-07:
Hey "donc" i flash the version v0.58rc5 r0+16336-b36068d35d and SQM work !!! 

I've problem with upload speed with my 4G , but with SQM it's work 

For my ADSLi i've problem  with latence , my ADSL is very sensible , it's not a problem because i set this like backup , and same ,SQM work !!

and i like the new option to restart interface if is down :p 

I'll looking if it's work or not , for the moment it's fine ,

Adorfer, 2021-06-07:
@Bug-Storm 
1) Is it fixed (for you): then please closed it as solved. 
2) if it's fixed: please make a short explanation what was -in your view- the issue/point which solved it. 
(your statement "For my ADSLi i've problem with latence , my ADSL is very sensible , it's not a problem because i set this like backup , and same ,SQM work !!": i can not parse in relation to the initial problem. This sounds more like 'bandwidth lower than expected' but not 'problem between VPS and Raspi/Raspi dumping kernel-errors in log'
3) if it's not solved, please make a recap of a)expected behavious and b)observed behaviour.

Bug-Storm, 2021-06-10:
> @Bug-Storm
>
> 1. Is it fixed (for you): then please closed it as solved.
> 2. if it's fixed: please make a short explanation what was -in your view- the issue/point which solved it.
>    (your statement [...]

It 's work , but today my vpn is down, it's possible set crontab if vpn or serveur is down , reboot openmptcp ? thanks

closed 2021-09-13 by github-actions[bot]