Ysurac/openmptcprouter #1886: Glorytun keeps "DUMP"-ing stuff to error log

milankragujevic, 2021-06-07:
## Expected Behavior
Glorytun doesn't spew "DUMP" err messages in syslog

## Current Behavior
Glorytun keeps spewing "DUMP" err messages

```
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.454974] mmc0: SDHCI controller on fe340000.emmc2 [fe340000.emmc2] using ADMA
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.463159] of_cfs_init
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.465666] of_cfs_init: OK
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.469257] Waiting for root device /dev/mmcblk0p2...
Mon Jun  7 08:46:45 2021 kern.warn kernel: [    1.476739] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
Mon Jun  7 08:46:45 2021 kern.warn kernel: [    1.483699] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.580433] mmc0: new ultra high speed DDR50 SDHC card at address 59b4
Mon Jun  7 08:46:45 2021 kern.debug kernel: [    1.587025] mmcblk mmc0:59b4: no of_node; not parsing pinctrl DT
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.587398] mmcblk0: mmc0:59b4 SD16G 14.6 GiB
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.592770]  mmcblk0: p1 p2
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.596504] mmc1: new high speed SDIO card at address 0001
Mon Jun  7 08:46:45 2021 kern.debug kernel: [    1.602021] mmcblk mmc1:0001: no of_node; not parsing pinctrl DT
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.612045] EXT4-fs (mmcblk0p2): mounted filesystem without journal. Opts: (null)
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.619562] VFS: Mounted root (ext4 filesystem) on device 179:2.
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.626307] devtmpfs: mounted
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.629598] Freeing unused kernel memory: 512K
Mon Jun  7 08:46:45 2021 kern.info kernel: [    1.634143] Run /sbin/init as init process
Mon Jun  7 08:46:45 2021 user.info kernel: [    1.692345] init: Console is alive
Mon Jun  7 08:46:45 2021 user.info kernel: [    1.695861] init: - watchdog -
Mon Jun  7 08:46:45 2021 user.info kernel: [    2.176448] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Mon Jun  7 08:46:45 2021 kern.info kernel: [    2.191123] pps_core: LinuxPPS API ver. 1 registered
Mon Jun  7 08:46:45 2021 kern.info kernel: [    2.196106] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
Mon Jun  7 08:46:45 2021 kern.info kernel: [    2.207211] PTP clock support registered
Mon Jun  7 08:46:45 2021 kern.info kernel: [    2.218046] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Mon Jun  7 08:46:45 2021 kern.info kernel: [    2.225571] ehci-fsl: Freescale EHCI Host controller driver
Mon Jun  7 08:46:45 2021 kern.info kernel: [    2.231559] ehci-platform: EHCI generic platform driver
Mon Jun  7 08:46:45 2021 kern.debug kernel: [    2.243448] libata version 3.00 loaded.
Mon Jun  7 08:46:45 2021 user.info kernel: [    2.246289] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Mon Jun  7 08:46:45 2021 user.info kernel: [    2.263771] init: - preinit -
Mon Jun  7 08:46:45 2021 kern.warn kernel: [    2.352170] bcmgenet: Skipping UMAC reset
Mon Jun  7 08:46:45 2021 kern.info kernel: [    2.357649] bcmgenet fd580000.ethernet: configuring instance for external RGMII
Mon Jun  7 08:46:45 2021 kern.info kernel: [    2.365216] bcmgenet fd580000.ethernet eth0: Link is Down
Mon Jun  7 08:46:45 2021 user.notice kernel: [    6.414141] mount_root: mounting /dev/root
Mon Jun  7 08:46:45 2021 kern.info kernel: [    6.421949] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Mon Jun  7 08:46:45 2021 user.info kernel: [    6.427558] mount_root: loading kmods from internal overlay
Mon Jun  7 08:46:45 2021 user.info kernel: [    6.454778] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Mon Jun  7 08:46:45 2021 user.info kernel: [    6.462418] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Mon Jun  7 08:46:45 2021 user.info kernel: [    6.531184] block: attempting to load /etc/config/fstab
Mon Jun  7 08:46:45 2021 user.info kernel: [    6.537521] block: extroot: not configured
Mon Jun  7 08:46:45 2021 user.warn kernel: [    6.562362] urandom-seed: Seeding with /etc/urandom.seed
Mon Jun  7 08:46:45 2021 user.info kernel: [    6.602479] procd: - early -
Mon Jun  7 08:46:45 2021 user.info kernel: [    6.605458] procd: - watchdog -
Mon Jun  7 08:46:45 2021 user.info kernel: [    7.132372] procd: - watchdog -
Mon Jun  7 08:46:45 2021 user.info kernel: [    7.136164] procd: - ubus -
Mon Jun  7 08:46:45 2021 user.info kernel: [    7.190272] procd: - init -
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.195008] uart-pl011 fe201000.serial: no DMA platform data
Mon Jun  7 08:46:45 2021 user.info kernel: [    7.371392] kmodloader: loading kernel modules from /etc/modules.d/*
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.392661] natsemi dp8381x driver, version 2.1, Sept 11, 2006
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.392661]   originally by Donald Becker <becker@scyld.com>
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.392661]   2.4.x kernel port by Jeff Garzik, Tjeerd Mulder
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.412127] zram: Added device: zram0
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.418692] NET: Registered protocol family 8
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.423074] NET: Registered protocol family 20
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.431576] NET: Registered protocol family 15
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.437203] Initializing XFRM netlink socket
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.444555] Modular ISDN core version 1.1.29
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.449087] NET: Registered protocol family 34
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.454424] mISDN: Layer-1-over-IP driver Rev. 2.00
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.459961] 0 virtual devices registered
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.465528] DSP module 2.0
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.468241] mISDN_dsp: DSP clocks every 64 samples. This equals 2 jiffies.
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.489312] tun: Universal TUN/TAP device driver, 1.6
Mon Jun  7 08:46:45 2021 kern.info kernel: [    7.497482] mISDN: HFC-multi driver 2.03
[... 2486 lines cut ...]
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'wan1' 'wan2' 'wan3'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.network='wifi'
```

## Specifications

  - OpenMPTCProuter version: v0.58rc5
  - OpenMPTCProuter VPS version: Version 0.1025-test 5.4.100-mptcp (*note: v0.58rc5 script installed*)
  - OpenMPTCProuter platform: RPI 4B with 4GB RAM

Ysurac, 2021-06-07:
All is green in status page ? The FS is not read only ?
No problem on load average of both VPS and router ?

milankragujevic, 2021-06-07:
@Ysurac 

Yes, all green, filesystem is mounted rw on RPI.

Load on RPI: 
` 09:49:35 up  1:02,  load average: 0.86, 0.69, 0.67`
Load on Server:
` 07:49:26 up  1:02,  2 users,  load average: 0.37, 0.26, 0.23`

I rebooted them as you can see, to check if there is a problem.

Around 3.2-3.5GB out of 4 GB free on both Server and RPI.

I will reinstall OMR on RPI on another SDCard and report back results.

milankragujevic, 2021-09-05:
Seems to happen due to some corruption/inconsistencies in config possibly? Can be resolved by reinstalling the image, appears rarely (on two times ever for me). Closing.

closed 2021-09-05 by milankragujevic