Ysurac/openmptcprouter #1887: wrt3200acm

phenix14000, 2021-06-07:
Bonjour Yannick,

J'ai essayé d'installer l'image "openmptcprouter-v0.57.3-r0+15225-bfc433efd4-mvebu-cortexa9-linksys_wrt3200acm-squashfs-factory.img" mais celle -ci ne fonctionne pas. Lors du reboot du routeur a la fin de l'installation, il est impossible de se connecter via n'importe quelle urls que ce soit celle de linksys ou celle de OMPTCPR.
Je suis obligé de faire un factory reset.

Cordialement,
Xavier

Ysurac, 2021-06-07:
Après l'installation il faut rebooter en appuyant sur le bouton I/O, puis la connexion se fait via le port Internet après un premier boot assez long.

phenix14000, 2021-06-07:
Très bien je fais ça demain merci pour ton retour super rapide ;)

closed 2021-09-10 by github-actions[bot]